roleoroleo/yi-hack-Allwinner-v2 #69: How to stop Hack from creating Events

r4zi3lrrr, 2021-01-17:
Hi All,

Apologies if this has been asked before but is there any way to stop this hack from creating events i.e. mp4 files.

It's just that my SD card seems to fill up extremely quickly 16 GB.

If there isn't a way of stopping these events from occurring.  What's the easiest way to delete them (don't really want to go through the web-interface as it take ages to delete)

Thank you.

roleoroleo, 2021-01-17:
The hack is not responsible for this behavior. It's the main configuration made with the app.
But you could:
- Disable che switch "Recording without Cloud".
- Set a percentage in the "SD free space (%)" box.
- Create a script and schedule with cron where you delete the files.

r4zi3lrrr, 2021-01-17:
Thank you for the info.

closed 2021-01-17 by r4zi3lrrr