roleoroleo/yi-hack-Allwinner-v2 #73: sendmail or curl capabilities

Mrivilla, 2021-01-20:
Hi Dears,
I wanted to send an email when movement is detected, but unfortunately sendmail does not work, very likely because firmware does not support ssl.
Another alternative is using POST API with some email providers, but we do not have CURL.
I tried HTTP string with nc(netcat) but string is too long to use it.
I see some advantages avoiding extra mqtt, ftp servers for simple advise just using the camara firmware.
I have a YI Home 1080p 9FSU.

Mrivilla, 2021-04-07:
Hi Dears, I have a camera with y20ga installed.

Yi 1080p Home 9FUS | 8.2.0* | y20ga

I have some services that works with curl, could you please help me on the process to include curl?
I found this process, but before apply I would like to know this will work for Allwinner 

cd /home/yi-hack-v3/sbin/
wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/curl
chmod +x /home/yi-hack-v3/sbin/curl
cd /home/yi-hack-v3/lib
wget https://github.com/utya1988/yi-hack/raw/master/sd/test/curl/libusr/libz.so.1.2.8 -O libz.so.1 

thanks for the support.
Marcos.

roleoroleo, 2021-04-07:
This curl will not work on this platform.
Try this repo: https://github.com/roleoroleo/yi-hack-Allwinner.gdrive

Mrivilla, 2021-04-08:
Thanks, I followed the procedure and gdrive got files, but wanted to use curl for other calls, and I got that message. 
[image]

To use individually curl do I need to configure something else? in the scripts it seems they are calling directly the file.
thanks.
Marcos.

roleoroleo, 2021-04-08:
Se LD_LIBRARY_PATH environment variable before running curl:
```
LD_LIBRARY_PATH=something ./bin/curl
```

Mrivilla, 2021-04-09:
[image]
This is perfect.
thanks you very much.
Marcos.

closed 2021-04-09 by Mrivilla