roleoroleo/yi-hack-Allwinner-v2 #74: RTSP is not working

kkawett, 2021-01-20:
Hello @roleoroleo 
First of all, thank you for the work you've done here.

I've a problem with my Yi Home.
The RTSP feed is not working.

I've tried both 0.1.3 and 0.1.2, it is the same.

I've turned on the log for rtsp, here is what I have:
`2021-01-20 20:10:41 - Detected possible locked process (1)
2021-01-20 20:10:42 - Detected possible locked process (2)
2021-01-20 20:10:43 - Detected possible locked process (3)
2021-01-20 20:10:45 - Detected possible locked process (4)
2021-01-20 20:10:46 - Detected possible locked process (5)
2021-01-20 20:10:47 - Detected possible locked process (6)
2021-01-20 20:11:00 - Detected possible locked process (1)
`

here is a snapshot of the output of top
`Mem: 59816K used, 1096K free, 1872K shrd, 5648K buff, 11440K cached
CPU: 41.6% usr  8.3% sys  0.0% nic 50.0% idle  0.0% io  0.0% irq  0.0% sirq
Load average: 0.89 0.36 0.13 1/115 2284
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1111     1 root     S    40536 66.3   0 37.5 ./rmm
 1307     1 root     S     4212  6.8   0  6.2 rRTSPServer
    3     2 root     SW       0  0.0   0  6.2 [ksoftirqd/0]
 1195     1 root     S     3812  6.2   0  0.0 ./oss
 1194     1 root     S     3476  5.6   0  0.0 ./p2p_tnp
 1192     1 root     S     2704  4.4   0  0.0 ./mp4record
 1369     1 root     S     2420  3.9   0  0.0 framefinder y21ga
 1189     1 root     S     1400  2.2   0  0.0 /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1193     1 root     S     1148  1.8   0  0.0 ./cloud
  640     1 root     S     1092  1.7   0  0.0 ./dispatch
 2284  1569 root     R      996  1.6   0  0.0 top
 1334     1 root     S      988  1.6   0  0.0 {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 1569  1563 root     S      984  1.6   0  0.0 -ash
    1     0 root     S      980  1.6   0  0.0 /sbin/init
 1266     1 root     S      980  1.6   0  0.0 udhcpc -i wlan0 -b -s /home/app/script/default.script
 2204  1334 root     S      980  1.6   0  0.0 sleep 10
  549     1 root     S      980  1.6   0  0.0 -/bin/sh
 1563  1229 root     S      948  1.5   0  0.0 dropbear -R -B
 1229     1 root     S      876  1.4   0  0.0 dropbear -R -B
 1208     1 root     S      860  1.4   0  0.0 httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1197     1 root     S      676  1.1   0  0.0 ./watch_process
 1230     1 root     S      668  1.0   0  0.0 ipc_multiplexer
  466     2 root     SW       0  0.0   0  0.0 [kworker/u2:2]
  599     2 root     SW       0  0.0   0  0.0 [RTW_CMD_THREAD]
`

Thanks a lot for your help

roleoroleo, 2021-01-20:
The log is not so strange.
What client are you using?
Are server and client in the same network?
Are you using udp transport or tcp transport?

kkawett, 2021-01-20:
I'm trying to connect with VLC.
Are server and client in the same network? Yes.
Are you using udp transport or tcp transport? I'm not sure how to check this... In the settings nothing seems to be set to UDP or TCP..

kkawett, 2021-01-21:
I remember that the first time I set up the hack I've turned off "Save video when a motion is detected".
Since it was recording all the time, maybe it didn't help the rtsp to work.

izemkung, 2021-01-23:
@kkawett Yi Home 2k or 1296p?

closed 2022-06-11 by roleoroleo