roleoroleo/yi-hack-Allwinner-v2 #75: No MQTT motion_start message if ai_human_detection activated

marcospg75, 2021-01-25:
Hello,
After multiple test, I identify that if I activate ai_human_detection, the camera only send motion_stop messages.
After desactivating ai_human_detection and rebooting the camera, I receive both start and stop motion messages.
Best Regards

roleoroleo, 2021-01-26:
It's a know problem.
Check for other issues, also in this project: https://github.com/roleoroleo/yi-hack-Allwinner

marcospg75, 2021-01-26:
> It's a know problem.
> Check for other issues, also in this project: https://github.com/roleoroleo/yi-hack-Allwinner

I was jumping issue by issue but can't find the solution... can you post the right link to solve this?
BR

roleoroleo, 2021-01-27:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/231
https://github.com/roleoroleo/yi-hack-Allwinner/issues/89
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/32

I think it's a bug related to yi program, because it doesn't work even without hack.

roleoroleo, 2021-10-30:
The mqtt message after an ai_human_detection will be removed in 0.2.0.
It's not possible to make it work.

kamkilt, 2022-05-21:
Hello, I have also find some issues with motion detection and ai human detection.

Since I have a dog, I'am receiving a lot of motion notifications from Home Assistant, I have turned on the person detection option and stopped receiving "motion_start" in hass through MQTT, after disabling person detection motion_start came back.
BTW. I have spend a lot of time till I found this issue.

But I have done a little testing with human detection and I've found that my camera, when person detection is on sends following messages:

```
/home/yi-hack # ipc_read -n 2 -d
debug on
Clear message in queue... done.
Clear message in queue... done.
IPC message. Len: 11. Status: Success!
Parsing message
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
MOTION_STOP
IPC message. Len: 18. Status: Success!
Parsing message
41 49 5f 48 55 4d 41 4e 5f 44 45 54 45 43 54 49 4f 4e 
AI_HUMAN_DETECTION
IPC message. Len: 11. Status: Success!
Parsing message
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
MOTION_STOP
IPC message. Len: 18. Status: Success!
Parsing message
41 49 5f 48 55 4d 41 4e 5f 44 45 54 45 43 54 49 4f 4e 
AI_HUMAN_DETECTION
IPC message. Len: 11. Status: Success!
Parsing message
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
MOTION_STOP
IPC message. Len: 18. Status: Success!
Parsing message
41 49 5f 48 55 4d 41 4e 5f 44 45 54 45 43 54 49 4f 4e 
AI_HUMAN_DETECTION
IPC message. Len: 11. Status: Success!
Parsing message
4d 4f 54 49 4f 4e 5f 53 54 4f 50 
MOTION_STOP
```

Those messages are looking pretty accurate, like camera my camera is sending at the start of human motion AI_HUMAN_DETECTION and MOTION_STOP at the end, when person detection is on, in ipc_read there are no timestamps, so I cannot tell how accurate it is, this behaviour is consistent with the app, because app also is sending only person detection notification.

Unfortunately I have a problem, that AI_HUMAN_DETECTION is not being send over MQTT, I'm only receiving MOTION_STOP messages.

I have tried setting the same topic for human detection and motion detection:
```
TOPIC_AI_HUMAN_DETECTION=motion_detection
TOPIC_MOTION=motion_detection

AI_HUMAN_DETECTION_START_MSG=human_start
AI_HUMAN_DETECTION_STOP_MSG=human_stop
MOTION_START_MSG=motion_start
MOTION_STOP_MSG=motion_stop
```

And following sensor in the HASS:

```
  - platform: mqtt
    name: "Living room human motion sensor"
    state_topic: "yicam/motion_detection"
    payload_on: "human_start"
    payload_off: "motion_stop"
    availability_topic: "yicam/status"
    payload_available: "online"
    payload_not_available: "offline"
    device_class: motion
```

But human_start is not being sand, I'm monitoring this in HASS MQTT settings, listening to yicam/#

I've done little digging in the code and found than in [mqttv4](https://github.com/roleoroleo/mqttv4/tree/6e12e1644e4f13a2fe0a8683ccdf290739525a1d)/[include](https://github.com/roleoroleo/mqttv4/tree/6e12e1644e4f13a2fe0a8683ccdf290739525a1d/include)/ipc.h

Human detection start command is different (longer) than that my camera is sending:
```
#define IPC_AI_HUMAN_DETECTION_START    "\x41\x49\x5F\x48\x55\x4D\x41\x4E\x5F\x44\x45\x54\x45\x43\x54\x49\x4F\x4E\x5F\x53\x54\x41\x52\x54" 
```
VS
```
41 49 5f 48 55 4d 41 4e 5f 44 45 54 45 43 54 49 4f 4e 
```

Is there any way I can investigate why human_start is not being send over MQTT?

My camera is:
YI Dome X on MStar
Firmware Version | 0.4.7
Base Version | 4.0.0.0F_201909251646
Model Suffix | y30

roleoroleo, 2022-05-21:
Probably, different models have a different behavior.
I could build a beta version for you, if you want to test it.

roleoroleo, 2022-05-21:
[ipc_sniff.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8747834/ipc_sniff.tar.gz)

kamkilt, 2022-05-21:
What is the difference between different queues?

AI_HUMAN_DETECTION message is not showing in queue 1 but it is showing on queue 2 in ipc_read.
That could be reason that it is not being forwarded by mqttv4?

I will test sniffer tomorrow.

roleoroleo, 2022-05-21:
There is no difference.
Probably you can't see the message in ipc_dispatch_1 because mqttv4 reads the queue and remove the message, before you can read it.

Anyway, mqttv4 doesn't check for message "AI_HUMAN_DETECTION" and this is not correct.
I could build a beta version to try.

roleoroleo, 2022-05-21:
Check this interesting issue:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/333

kamkilt, 2022-05-22:
I tried this, but same AI_HUMAN_DETECTION messages are not being send over MQTT.
Please build a beta version if you can.

kamkilt, 2022-05-22:
I have played a little bit with pic_sniffer and found following commands regarding motion and human detection.
When person detection is on (filtered with being of 01 00 00 00 02 00 00 00):
```
01 00 00 00 02 00 00 00 f5 00 f5 00 00 00 00 00 
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00 
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00 
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
01 00 00 00 02 00 00 00 7d 00 7d 00 00 00 00 00 
01 00 00 00 02 00 00 00 f6 00 f6 00 00 00 00 00 
01 00 00 00 02 00 00 00 f5 00 f5 00 00 00 00 00 
01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00 
```

There are both messages being send:
IPC_AI_HUMAN_DETECTION_START (01 00 00 00 02 00 00 00 f5 00 f5 00 00 00 00 00)
IPC_AI_HUMAN_DETECTION_STOP (01 00 00 00 02 00 00 00 ed 00 ed 00 00 00 00 00)

but ipc_read shows only
```
ipc_read -n 2
AI_HUMAN_DETECTION
MOTION_JPG
MOTION_STOP
```

on MQTT without any change, no messages for AI_HUMAN_DETECTION.

I also tried following commands from [yi-hack-Allwinner-v2](https://github.com/roleoroleo/yi-hack-Allwinner-v2): 
IPC_AI_HUMAN_DETECTION_OFF
IPC_AI_HUMAN_DETECTION_ON
IPC_SOUND_DETECTION_ON
IPC_SOUND_DETECTION_OFF

They are working correctly and they are missing in ipc_cmd for [yi-hack-MStar](https://github.com/roleoroleo/yi-hack-MStar).

roleoroleo, 2022-05-23:
> but ipc_read shows only

You can't read other messages because ipc_multiplexer doesn't forward them.
I will  send a beta.

roleoroleo, 2022-05-24:
[mqttv4.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8760622/mqttv4.gz)
[ipc_multiplexer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8760624/ipc_multiplexer.gz)

Overwrite these files.
You have to kill the processes before overwriting.

kamkilt, 2022-05-30:
I have tested those modules for couple of days and it's working!
When person detection is ON, motion_start and motion_end are being send like in official app, only when human is detected.

roleoroleo, 2022-05-30:
I will share this solution in this thread:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/333
and I will check if I can release it as a standard solution.

closed 2023-01-08 by stale[bot]