roleoroleo/yi-hack-Allwinner-v2 #76: [Feature Request] Support for all Yi-Hack-v4 supported models

NdR91, 2021-01-26:
Hello,

As per the object: would be great if we could get the support on AllWinner-v2 for all the devices previously supported by Yi-Hack-v4.

Unfortunatelly, the Yi-Hack-v4 FW is not supported anymore. The fw doesn't get an update since May 2019. A lot of bugs are still there and some features on development never came (Ex. ONVIF support).

Is there a possibility to receive this support from AllWinner FW?

Thank you!

roleoroleo, 2021-01-26:
Sirry but it's not possible.It's a different hw platform
-------- Messaggio originale --------Da: NdR91 <notifications@github.com> Data: 26/01/21  17:16  (GMT+01:00) A: roleoroleo/yi-hack-Allwinner-v2 <yi-hack-Allwinner-v2@noreply.github.com> Cc: Subscribed <subscribed@noreply.github.com> Oggetto: [roleoroleo/yi-hack-Allwinner-v2] [Feature Request] Support for all
  Yi-Hack-v4 supported models (#76) 
Hello,
As per the object: would be great if we could get the support on AllWinner-v2 for all the devices previously supported by Yi-Hack-v4.
Unfortunatelly, the Yi-Hack-v4 FW is not supported anymore. The fw doesn't get an update since May 2019. A lot of bugs are still there and some features on development never came (Ex. ONVIF support).
Is there a possibility to receive this support from AllWinner FW?
Thank you!

NdR91, 2021-01-26:
Ahhhh, bad news. 
Anyway... thank you for the answer!

closed 2022-06-11 by roleoroleo