roleoroleo/yi-hack-Allwinner-v2 #77: Crontab is working?

tjuanma, 2021-01-27:
I would like to make a reboot every midnight by cron.

To do that I'm trying with:

0 0 * * * /tmp/sd/yi-hack/www/reboot.sh
0 0 * * * /sbin/reboot

But no one of those are working.

Doing that straight from SSH makes my camera reboot properly so i think that cron is not working as expected, is that possible?

Running "ps" i can't find the cron process working.

the "root" file where crons saves, updates properly after changing the config and reboot manually the camera, but cron never get executed.

Help please.

Another way to schedule a reboot?

roleoroleo, 2021-01-28:
It should work.
I'm using it.
If you run manually the command, does it work?
`/home/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/`

mcanteri, 2021-01-29:
> It should work.
> I'm using it.
> If you run manually the command, does it work?
> `/home/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/`

Hi @roleoroleo,
In the 0.1.3 release there isn't that executable.
I was trying to use that feature in the lasts days but I found too that isn't working.
I thought about a daily reboot too since I keep finding a core file in the record folder and via rstp HA from time to time trace errors. 
After putting the executable in usr/sbin folder and tried your command I can see the crond running:
20803 root       860 S    crond -c /var/spool/cron/crontabs/
So it is just the executable missing.

roleoroleo, 2021-01-29:
Sorry.
The change is committed but not released.
I have to arrange a new release.

roleoroleo, 2021-01-29:
0.1.4 released.
Pay attention to the breaking change.

WalkingDot, 2021-05-22:
If I'm using the web interface to add a cronjob, only the minute value get saved.

I used this example:
`0 2 * * * echo 'This is a test' > /tmp/sd/record/crontest.txt`

Is saved in the text field:
`0`

roleoroleo, 2021-05-22:
Sorry, I added a new bug in the last commit.
Please overwrite the file /home/yi-hack/www/cgi-bin/set_configs.sh
with this:
https://github.com/roleoroleo/yi-hack-MStar/blob/master/src/www/httpd/cgi-bin/set_configs.sh

WalkingDot, 2021-05-22:
Thanks, here is the diff from the old one and the fixed set_configs.sh:
```
< YI_HACK_PREFIX="/home/yi-hack"
---
> YI_HACK_PREFIX="/tmp/sd/yi-hack"

< IFS=$(echo -en "\n\b")

<     elif [ "$KEY" == "TIMEZONE" ] ; then
<         echo $VALUE > $YI_HACK_PREFIX/etc/TZ
```
Should I change the "YI_HACK_PREFIX", too?

WalkingDot, 2021-05-26:
i wanted to add a second cronjob, when I save only the first entry is added.

```
0 1 * * * echo 'test1' > /tmp/sd/record/crontest.txt
0 2 * * * echo 'test2' > /tmp/sd/record/crontest.txt
```

roleoroleo, 2021-05-27:
You are right.
I need to check it.

roleoroleo, 2021-05-28:
Here it is:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/39d7924371e6499e0881bbfd0d5abb2c95820122

WalkingDot, 2021-06-21:
Is it maybe possible to update the crontab version? The @reboot function isn't working:
`@reboot echo 'test' > /tmp/sd/record/crontest.txt`

This would allow to run scripts/binaries on boot, without changing anything of your code.

roleoroleo, 2021-06-22:
This cron is part of the busybox.
If the syntax you used doesn't work I can't update it, sorry.
But if you want to run somehing at boot you could use startup.sh.
Check the last lines of system.sh: https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/static/static/yi-hack/script/system.sh

WalkingDot, 2021-06-23:
Thanks, your suggestion to the startup.sh helps me a lot :-)

closed 2023-01-08 by stale[bot]