roleoroleo/yi-hack-Allwinner-v2 #84: Random reboot with 0.1.4

sro2000, 2021-01-31:
Hi,
Facing random reboots with 0.1.4/r30gb as soon as I access my camera with any ONVIF compatible apps (Android or Windows). No problem through RTSP however.
Thanks

roleoroleo, 2021-01-31:
Check if the problem is the snapshot, for example through the web interface.

sro2000, 2021-01-31:
Yes, seems to be. When I open the Snapshot tab, I can see a fresh snapshot and then the camera reboots itself. No problem however on the PTZ tab.

roleoroleo, 2021-01-31:
It's a know issue: low memory.
Enable swap space.

sro2000, 2021-01-31:
Thank you, it works. My bad, I desactivated the swap space in 0.1.4 and didn't think it could be the reason. Sorry for wasting your time !

roleoroleo, 2021-01-31:
No problem.

huntz, 2021-02-08:
It would be nice to indicate this configuration change in README.md, so new users would be aware of it

roleoroleo, 2021-02-09:
This isn't a configuration change.
Swap space problem is here from the beginning.
But I could add a note in the readme.

closed 2022-06-11 by roleoroleo