roleoroleo/yi-hack-Allwinner-v2 #89: RTSP python OpenCV Freeze

albinpopote, 2021-02-07:
RTSP is activated on my YI Home 1090P (BFUS) under 0.1.4 hack firmware. It is protected by a username and password.

When I try to see camera frames over RTSP protocol under VLC I haven't any error, video is fluid and works fine.
But I try to use the camera to make motion detection with python code and OpenCV binding for python.

When I launched the program (without motion detection, only show the RSTP frames on computer), the video is very slow and after a short delay (lower than 10s), the frame freezes and the program exits with the followwing errors:

Traceback (most recent call last):
  File "./test-2.py", line 41, in <module>
    cv2.imshow( 'Result', img)
cv2.error: OpenCV(4.5.1) /tmp/pip-req-build-ddpkm6fn/opencv/modules/highgui/src/window.cpp:376: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

Sometime I got X264 logs like that:
[h264 @ 0x1889d00] left block unavailable for requested intra4x4 mode -1
[h264 @ 0x1889d00] error while decoding MB 0 42, bytestream 479

Same code works perfectly with others RTSP flows from internet.

Anyone can test with his camera to see if the same error occurred.
Is it possible to troubleshot rRTSPServer (I didn't find any log file). I will search on openCV python bindings to get more details of really occurred.

For information:
- rRTSPserver don't crash, PID is the same before and after python program crashing
- error occurred on read function from VideoCapture object in openCV
- VideoCapture object is still opened after read error
- I tried to force FFMPEG program for VideoCapture and TCP protocol for RTSP Transport

uklatt, 2021-02-20:
Does it havven with "low-res" stream?
I have simmilar problems, but only with the "high-res" stream.

Uwe

albinpopote, 2021-03-14:
Same with low-res stream... 

I made some tests inspired by the following python library
https://pypi.org/project/rtsp/

Works with a thread to update (read) the frame from camera. I don't understand why it doesn't work in the same process...

I got some errors from h264 stream on live streaming (without freeze):
```
[h264 @ 0x95af600] cabac decode of qscale diff failed at 99 0
[h264 @ 0x95af600] error while decoding MB 99 0, bytestream 512
[h264 @ 0x95e9100] cabac decode of qscale diff failed at 53 0
[h264 @ 0x95e9100] error while decoding MB 53 0, bytestream 693
[h264 @ 0x8d49500] error while decoding MB 20 41, bytestream -7
[h264 @ 0x9622c00] cabac decode of qscale diff failed at 71 0
[h264 @ 0x9622c00] error while decoding MB 71 0, bytestream 803
[h264 @ 0x8d49500] error while decoding MB 25 2, bytestream -10
[h264 @ 0x9605e80] error while decoding MB 40 52, bytestream -8
[h264 @ 0x8d49500] cabac decode of qscale diff failed at 48 0
```

closed 2022-06-11 by roleoroleo