roleoroleo/yi-hack-Allwinner-v2 #90: Will the yi home 1080p IFUSY44GBN compatible.?

chippa2k, 2021-02-08:
Hello, Will the yi home 1080p IFUSY44GBN be able to have these extra features. Thank you.

roleoroleo, 2021-02-08:
I don't know.
What platform is?
What firmware version is?

chippa2k, 2021-02-09:
Here is a backup
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5952309/backup.zip)

roleoroleo, 2021-02-09:
Your cam is a y21ga.
You can use this hack.

chippa2k, 2021-02-10:
Not working, just yellow light stays on, thank you for looking into it for me.

roleoroleo, 2021-02-10:
Remove all saved mp4 from the sd card (record folder) and try again.

mcanteri, 2021-02-10:
> Not working, just yellow light stays on, thank you for looking into it for me.

And use the provided cable and wall adaptor, if it is weak the cam doesn't work correctly.

chippa2k, 2021-02-12:
Still no good, can i edit any files for this to work, i am using original power and micro usb. Thank you for your time and thoughts.

roleoroleo, 2021-02-12:
I checked again your dump and it's ok.
What program are you using to decompress the archive?

chippa2k, 2021-02-12:
7Zip, is that no good. thank you.

roleoroleo, 2021-02-12:
7zip is ok.
Another question: does backup procedure work?
I mean the procedure explained in the wiki.

chippa2k, 2021-02-12:
Yes it does.

roleoroleo, 2021-02-12:
The hack uses the same procedure.
It must work.

roleoroleo, 2021-02-12:
Did you check the web interface?

chippa2k, 2021-02-13:
Yes, I now think its bricked, will open it up to repair. thank you.

chippa2k, 2021-02-18:
When i come to open it up for repair, is there any commands or file i can use or write to the loader, to make this work. thank you.

chippa2k, 2021-02-18:
Ive unbricked, and watched the boot process, but its not finding a file for some reason, here is the end of the boot,

checkdisk.c(do_fat_check-121)[00:00:07.388]:retBuf(fsck.fat 3.0.26 (2014-03-07)
step 0 5
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
step 1 5
step 2 5
step 3 7
step 4 7
step 5 7
step 6 7
/dev/mmcblk0p1: 559 files, 2088/998043 clusters
)

checkdisk.c(do_fat_check-140)[00:00:07.391]:Mount sd ok.

checkdisk.c(check_partion_vfat-156)[00:00:07.391]:TF Card check finished.

checkdisk, main() ret_status(10000)
checkdisk, main() ret_status(10000),g_sysconf_ptr->tf_status.stat(10000)
---update file(home_y21gam) Not exist---
/backup/init.sh: line 157: syntax error: unexpected end of file (expecting "fi")

root@(none):/# checkdisk.c(do_fat_check-121)[00:00:07.388]:retBuf(fsck.fat 3.0.2
6 (2014-03-07)
-/bin/sh: syntax error: unexpected word (expecting ")")
-/bin/sh: syntax error: unexpected ")"
root@(none):/# step 0 5
-/bin/sh: step: not found

roleoroleo, 2021-02-18:
init.sh is wrong.
I don't know why.
You have to restore mtdblock4 with the original one.
Read the wiki: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam

chippa2k, 2021-02-19:
Unbricked camera, put your files back on, and its bricked again, so could be the size of the sd card, im using 32gb. So all i put on the sd card (root) is the release files, the 3 folders and init file.,Or the files within the zip, I have tried both scenerios, but still no good. Thank you, for your time, could i not just write a bin file to the start of memory address.

roleoroleo, 2021-02-19:
Maybe the init.sh file is in dos format?
Extract the sd card and check if the newline is dos format or linux format.
https://github.com/roleoroleo/yi-hack-Allwinner/issues/132#issuecomment-682039627

roleoroleo, 2021-02-19:
Your SD sould look like this:
[image]

roleoroleo, 2021-02-19:
Another question: are you able to login with telnet?

chippa2k, 2021-02-19:
Yes i can login with telnet, im just gonna test your newline theory.

chippa2k, 2021-02-19:
LF is at the end of my init.sh files.

roleoroleo, 2021-02-19:
Please, cat the file and post it.

chippa2k, 2021-02-19:
Sorry cant telnet through wifi, only while its connected to ttl, the blue light comes on, then dies..

roleoroleo, 2021-02-19:
If your current status is different than this post https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/90#issuecomment-776024681 make another backup and send it to me.
I will check it.

roleoroleo, 2021-02-19:
Or maybe a power problem? Defective adapter?
Did you try to use another power adapter and another cable?

roleoroleo, 2021-02-19:
If you are connected through ttl, starts an ftp server and send me init.sh.

chippa2k, 2021-02-20:
Thank you, i know im getting closer, i saw your script execute, and now it reads on the webpage.
ERROR
The requested URL could not be retrieved
The following error was encountered while trying to retrieve the URL: http://ip-cam:8080/

Unable to determine IP address from host name ip-cam

The DNS server returned:

Name Error: The domain name does not exist.
This means that the cache was not able to resolve the hostname presented in the URL. Check if the address is correct.

Your cache administrator is webmaster.

I changed the sdcard, so it was a card problem, thank you for your help.

roleoroleo, 2021-02-20:
Use the ip addess and not the name.
Example:
http://192.168.1.100:8080

chippa2k, 2021-02-21:
Yes I did, but no good, camera just keeps crashing, we are close, but not there yet, I've tried 4 different sd cards, and 3 different power supplies, it only draws 0.10amps, and the unit does get hot. I will keep trying, thank you for your support. Keep up the excellent work, wished I had your knowledge.

roleoroleo, 2021-02-21:
I think your init.sh is corrupted.
Send me a copy of it.

chippa2k, 2021-02-23:
Can you make any sense of this, the hack runs, then stop, with a write rror.

--------------------------insmod sensor--------------------------
[    3.915485] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
[    5.902053] sunxi-gmac 5020000.eth eth0: Initialize hardware error
============================================= home low_half_init.sh... =========                                                          ================================
============================================= begin to start app... ============                                                          =============================
############## Starting Hack ##############
############## Starting Hack ##############
############## Starting Hack ##############
############## Starting Hack ##############
### Hacking
Applying hack
### Updating /backup/init.sh
cp: write error: No space left on device
chmod: /backup/init.sh: No space left on device
### Updating /backup/tools/wifidhcp.sh
cp: can't open '/backup/tools/wifidhcp.sh': No space left on device
chmod: /backup/tools/wifidhcp.sh: No such file or directory
### Disabling hack for next reboot
starting pid 649, tty '': '/etc/init.d/rcK'
mount: mounting /dev/mtdblock3 on /home failed: Resource busy
mount: mounting /dev/mtdblock4 on /backup failed: Resource busy
sh: you need to specify whom to kill
mount: mounting /dev/mtdblock3 on /home failed: Resource busy
mount: mounting /dev/mtdblock4 on /backup failed: Resource busy
sh: you need to specify whom to kill
killall: udevd: no process killed
starting pid 672, tty '': '/sbin/swapoff -a'
starting pid 673, tty '': '/bin/umount -a -r'
umount: devtmpfs busy - remounted read-only
The system is going down NOW!
Sent SIGTERM to all processes

chippa2k, 2021-02-23:
Some files are in newbackup, and not backup.

roleoroleo, 2021-02-23:
backup partition is full.
It's a jffs2 filesystem with very little free space.
The only way is to restore the mtd4 partition.

chippa2k, 2021-02-23:
That's how I'm unbricking it aswel, is there a way to extract the files or bins from firmware file.

Thank you.

roleoroleo, 2021-02-24:
If you have a backup you can extract files from images.
For squashfs (mtd2 and mtd3) use unsquashfs command.
For jffs2 (mtd4) use this script with a 64k erase size:
```
#!/bin/bash

## Script to mount jffs2 filesystem using mtd kernel modules.
## EMAC, Inc. 2009

if [[ $# -lt 2 ]]
then
    echo "Usage: $0 FSNAME.JFFS2 MOUNTPOINT [ERASEBLOCK_SIZE]"
    exit 1
fi

if [ "$(whoami)" != "root" ]
then
    echo "$0 must be run as root!"
    exit 1
fi

if [[ ! -e $1 ]]
then
    echo "$1 does not exist"
    exit 1
fi

if [[ ! -d $2 ]]
then
    echo "$2 is not a valid mount point"
    exit 1
fi

if [[ "$3" == "" ]]
then
<------>esize="128"
else
<------>esize="$3"
fi

# cleanup if necessary
umount /dev/mtdblock0 &>/dev/null
modprobe -r mtdram &>/dev/null
modprobe -r mtdblock &>/dev/null

modprobe mtdram total_size=32768 erase_size=$esize || exit 1
modprobe mtdblock || exit 1
dd if="$1" of=/dev/mtdblock0 || exit 1
mount -t jffs2 /dev/mtdblock0 $2 || exit 1

echo "Successfully mounted $1 on $2"
exit 0
```

chippa2k, 2021-02-25:
Thats a bit over my head, how can i format or erase this mtd4 partition or get a copy of the original and replace with mine, my backup may be corrupted, but just enough to get me out of trouble, your unbricking code works a charm, I have now got the camera back to normal operation without your hack. 

Thank you for your time.

chippa2k, 2021-02-26:
Is it possible you can send a blank or non corrupt mtd4 partition please. Thank you.

roleoroleo, 2021-02-27:
[mtd4.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6055299/mtd4.zip)

chippa2k, 2021-02-28:
You are the man, that got i working (in a fashion), i now get webpage up, ive changed to lowest settings, and i can stream, i enabled swap file, but then the camera crashes, but i will try try again. thank you so much for your help, how do i buy a coffee...

chippa2k, 2021-02-28:
This can be closed.

closed 2021-02-28 by chippa2k

roleoroleo, 2021-02-28:
> You are the man, that got i working (in a fashion), i now get webpage up, ive changed to lowest settings, and i can stream, i enabled swap file, but then the camera crashes

With swap space enabled the cam shouldn't restart

> how do i buy a coffee...

At the bottom of the page: https://github.com/roleoroleo/yi-hack-Allwinner-v2