roleoroleo/yi-hack-Allwinner-v2 #91: Hack for Yi Camera Dome U?

Gizmo-Ger, 2021-02-08:
I just got a new camera called Yi Dome U. I thought I would give it a try with the lastest hack. Unfortunately the hack doesn´t work on this model, although the backup worked fine. Is it possible to use the hack with this model as well? I can provide the backup files for investigation. Thanks in advance
Firmware is 9.0.22.14_202011061019

roleoroleo, 2021-02-08:
I will check it.
Share the backup.

Gizmo-Ger, 2021-02-08:
https://drive.google.com/file/d/1fPYTzNl3fE-pNGLFzcvKcdGBEkT5KNMc/view?usp=sharing

roleoroleo, 2021-02-09:
Your cam is an h52ga.
Very similar to a y21ga.
I will try to add a beta support fot it.

Gizmo-Ger, 2021-02-09:
Awesome. Thanks for your effort and your time. I highly appreciate.

roleoroleo, 2021-02-10:
Try this hack.
You should be able to install it and reach the web interface.
I don't know if rtsp and snapshot are working.
Try it (low and high res).

[h52ga_0.1.4.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5958667/h52ga_0.1.4.tar.gz)

Gizmo-Ger, 2021-02-10:
Thank you. I tried the version but the camera keeps rebooting and turning 360 and rebooting again. Can I provide any logs?

roleoroleo, 2021-02-10:
If you remove the sd card, does it work?

Gizmo-Ger, 2021-02-10:
Yes, once the SD card is removed the camera works normally.

roleoroleo, 2021-02-10:
Dump the SD and share with me.
A compressed file is enough.

Gizmo-Ger, 2021-02-10:
https://drive.google.com/file/d/1qYmrTb-99_YMYBc8Y1rxg35CZIwDYBvf/view?usp=sharing

roleoroleo, 2021-02-10:
Is there a file "lower_half_init.sh" in the root of your sd card?

Gizmo-Ger, 2021-02-10:
yes, it is

roleoroleo, 2021-02-10:
No, it doesn't matter.

The SD seems to be ok.
Try to edit etc/system.conf with the following content:
```
HTTPD=yes
TELNETD=yes
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=no
RTSP_STREAM=high
RTSP_AUDIO=yes
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=high
ONVIF_NETIF=wlan0
ONVIF_WM_SNAPSHOT=yes
NTPD=no
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
```

and try again the cam.

roleoroleo, 2021-02-10:
What program did you use to decompress the archive?

Gizmo-Ger, 2021-02-10:
I´ll try in a minute. I am using 7zip

Gizmo-Ger, 2021-02-10:
Ok, checked with your modifications. I can access the web interface and no more reboots. Streams and snapshots don´t work.

roleoroleo, 2021-02-10:
Now you should enable the services one at a time and check which service causes the reboot.

Gizmo-Ger, 2021-02-11:
I tested again with RTSP, ONVIF, SSH, NTPD, HTTPD, swap enabled. FTP and MTTQ are disabled. Camera works so far without any reboots.
PTZ works, but without preview. Snapshots and streams don´t work.

roleoroleo, 2021-02-11:
Ok.
Probably the swap is mandatory.
Without it the cam reboots.

To fix snapshot and stream I need 2 copies of fshare_frame_buf file.
You can find it in /etc.

Copy it, wait 5 secs and copy it again.

Gizmo-Ger, 2021-02-12:
Unfortunately there is no such file.

Gizmo-Ger, 2021-02-12:
[image]

roleoroleo, 2021-02-12:
Sorry, my mistake.
It is in /dev.

Gizmo-Ger, 2021-02-12:
Looks like something is really off on my side. Because there is no /dev directory
[image]

roleoroleo, 2021-02-12:
/dev in the root.
Not inside /tmp/sd/yi-hack

Gizmo-Ger, 2021-02-13:
got it figured out
https://drive.google.com/open?id=1-317PmdFygSqH2MtX0Oa9tkWO7Ogk-GB&authuser=shstieglitz%40gmail.com&usp=drive_fs
https://drive.google.com/open?id=17XOu1iYtnAIQoK-fK9vNfFxx0k-jG1je&authuser=shstieglitz%40gmail.com&usp=drive_fs

roleoroleo, 2021-02-13:
Try this beta:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5975381/rRTSPServer.gz)

Stop the watchdog and the old rtsp server and run this new binary:
```
. /tmp/sd/yi-hack/script/env.sh
killall wd_rtsp.sh
killall rRTSPServer
RRTSP_RES=both RRTSP_AUDIO=no RRTSP_MODEL=h52ga rRTSPServer
```
If you want to enable debug add RRTP_DEBUG=1
RRTSP_AUDIO could be: no, pcm, alaw, ulaw and aac.

Gizmo-Ger, 2021-02-13:
Awesome! Thank you so much. Now I get a stream

roleoroleo, 2021-02-13:
Does it works?
High res and low res?

This should fix snapshots:
[snap.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5975865/snap.tar.gz)

Gizmo-Ger, 2021-02-13:
Both streams when freshly rebooted. After a cpl of hours it won´t work anymore. With the your new snap.tar.gz snapshots work flawlessly. Thanks again

roleoroleo, 2021-02-13:
Run the server with debug option and redirect the stderr on file.
Then send me the file.
I need to check the log when the server stops working.

roleoroleo, 2021-02-13:
Also check if ptz works.
Both in the web page and through onvif device manager.

Gizmo-Ger, 2021-02-13:
ptz works, both in web page and onvif device manager. Although up and down seem to be mixed up.

roleoroleo, 2021-02-13:
Up and down only?
Mmmmh
Check this issue and let me know if your cam is different:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/68

roleoroleo, 2021-05-20:
I think h51ga.

closed 2023-01-08 by stale[bot]