roleoroleo/yi-hack-Allwinner-v2 #92: yi dome guard KFUS firmware 9.0.05.32_202004101552

moriwaky, 2021-02-08:
I tried the r30gb version 0.1.4 and 0.1.3 but it´s not stable for my camera. I know is a beta. But I can use my two camera as betatester if you want. RSPT just works on low quality and the PTZ doesn´t works on synology survillance, If I use ONVIF I get PTZ control but the camera restarts every minute. here you have the original firmaware of the camera if you want to take a look:

https://1drv.ms/u/s!Aq2FEDvef10rjh1eHb62QIeXFl4D?e=LxFgM7

moriwaky, 2021-02-08:
Update I have read that "swap file" fixs randoms reboot on ONVIF.

roleoroleo, 2021-02-08:
Does rtsp work after enabling swap?

moriwaky, 2021-02-08:
yes but in low res and no PTZ control.  With swap file on, I don´t have randoms reboot on ONVIF but I have no live view on synology survillance.

moriwaky, 2021-02-08:
Sorry I my last answer was wrong. I don´t have any video on rtsp with swap file "on" And with Onvif I don´t have live view from the synology survillance (maybe codec related issue because I can´t check the telemetry of the live video)

roleoroleo, 2021-02-08:
Please check if video works with vlc and check if onvif works with onvif device manager.

moriwaky, 2021-02-08:
Video works on vlc  high quality and low. And in onvif device manager i can use the ptz and I have video. but I synology I just get low quality video on rtsp. and no live video and no record video on onvif. Is anything I can do to make it work with the synology survillance? thanks for your time by the way

rt400, 2021-02-09:
i have also yi dome guard, in the camera web i dont have option in PTZ to save presnt . the onvif work bad, the rtsp work good only in low res

roleoroleo, 2021-02-09:
@moriwaky 
The high res stream is h265 coded while the low res stream is h264 coded.
Maybe Synology doesn't support h265.
Set onvif profile low and check if you are able to see low res stream in onvif.

@rt400
What's your model suffix? r30gb?

rt400, 2021-02-09:
@roleoroleo yes.
KFUS with 9.0.05.33_202009251856

rt400, 2021-02-09:
how can i stream video and audio together ?

roleoroleo, 2021-02-09:
Enable it in configuration web page.

rt400, 2021-02-09:
its enable but in live stream like vlc there is no sound. its looks that disparate..

roleoroleo, 2021-02-09:
If you check the menu Tools->Codec Information, could you see the audio section?
Or is there only video section?

rt400, 2021-02-10:
yes, i see the audio section.
but there is no sound. if i play the mp4 that the camera record i can hear the sound

roleoroleo, 2021-02-10:
Strange behavior.
Sorry, I don't know how to check it because I don't have this model.
I will try to send you a beta version.

rt400, 2021-02-10:
strange, i check it on other model YI Home 1080p with y20ga perfix and also didn't have sound

rt400, 2021-02-10:
@roleoroleo What about the PTZ features ? Yi Doom Guard

roleoroleo, 2021-02-10:
Try this beta:
[r30gb_0.1.4.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5960110/r30gb_0.1.4.tar.gz)

About PTZ it should work.
Please, explain me better your problem.
Did you try it with onvif device manager?

rt400, 2021-02-11:
> Try this beta:
> [r30gb_0.1.4.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/5960110/r30gb_0.1.4.tar.gz)
> 
> About PTZ it should work.
> Please, explain me better your problem.
> Did you try it with onvif device manager?
here is what i have in the camera web control
[image]
as tou can see i dont have save present and other option

there is other way to update the firmware without copy it to sd card ?

roleoroleo, 2021-02-11:
Other options?
The web interface allows you to go to preset.
It doesn't allow to set preset.

> there is other way to update the firmware without copy it to sd card ?

If you have a previous version, you can update it online.

rt400, 2021-02-11:
OK. i ask cause i have the sonoff camera also and there i have more option

How can i update it online ?

roleoroleo, 2021-02-11:
> OK. i ask cause i have the sonoff camera also and there i have more option

Ok. The yi cam already has a system for setting presets.
So I didn't want to make a new one like Sonoff.
You can configure the presets using the app.

> How can i update it online ?

Menu Maintenance -> Upgrade firmware
But I recently solved a prblem in the ota procedure, so it is better if you do the installation from scratch.

rt400, 2021-02-11:
so i need the yi app to do it ? there is no other way ?

roleoroleo, 2021-02-11:
I didn't reverse the yi messages about presets (if there are).
Maybe it's possible sending ipc messages.

rt400, 2021-02-14:
@roleoroleo first the update firmware that you send me fix something that I want to tell you, audio ... so thanks.
now about the ptz (ipc messages) can i help you with this to add it ? its really necessary for me.
many thanks for your job!!!

josebc83, 2021-06-30:
Is it necessary to have the camera activated with the yi software? or can I apply the hack with the camera taken out of the box?

WalkingDot, 2021-06-30:
@josebc83 The hack is working out of the box. If you want to use wifi for the first setup, you need to add the wpa_supplicant (https://github.com/roleoroleo/yi-hack-Allwinner/issues/289).

closed 2023-01-08 by stale[bot]