roleoroleo/yi-hack-Allwinner-v2 #93: Change wifi network to yi home 1080p

albertopalotes35, 2021-02-11:
Hello good, I have installed the yi hack Allwinner v2 in the yi home 1080p and I would like to know how I can change the wifi network to which the camera connects

roleoroleo, 2021-02-12:
See the wiki.

albertopalotes35, 2021-02-12:
I have read the wiki and I have a question, after putting the files with the new wifi configuration on the sd card that has the yi hack and waiting for the restart, do you have to delete the wifi configuration files that I copied to the sd card?

roleoroleo, 2021-02-12:
Yes.

albertopalotes35, 2021-02-12:
thank you very much

GqSteve, 2021-02-26:
I have an YI 1080p Home BFUS with 9.0.19* firmware. I installed the y21ga 0.1.4 firmware and worked perfectly 👍 

I tried however to change the WiFi ssid and password with your description and this doesn't work. I know the procedure since I used it many times with the "yi-hack-Allwinner". After step "6. The cam will reboot once.", the cam doesn't change the WiFi connection. It is still connected to the old WiFi ssid.

closed 2022-06-11 by roleoroleo