roleoroleo/yi-hack-Allwinner-v2 #96: r30gb with fw 0.1.4

roleoroleo, 2021-02-17:
I need some feedback on how the r30gb model works with fw 0.1.4.
1. Does RTSP work in high res and low res with a vlc client?
2. Does snapshot work in high res and low res?
3. Does PTZ work with onvif or from web page?
4. Is PTZ inverted left/right up/down?

Thank you.

rt400, 2021-02-17:
I can help you with this :
1. Rtsp with high res not work only low res
2. snapshot work ok with high and low res
3. Ptz from web page work also from onvif manager , but there is no save present or run present on boot
4. the Ptz is not inverted (my camera is on table so no need the rotation)

Hope this will help...

roleoroleo, 2021-02-18:
About rtsp high res, could you run the process with debug option and send me the log?
- Login with ssh
- killall wd_rtsp.sh
- killall rRTSPServer
- . /tmp/sd/yi-hack/script/env.sh
- RRTSP_RES=both RRTSP_AUDIO=no RRTSP_DEBUG=1 rRTSPServer
- Then run a RTSP client and connect to the cam

rt400, 2021-02-19:
```1613716464559: frame detected - frame_len: -1454005340 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464559: frame detected - frame_len: -164420411 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464559: frame detected - frame_len: -954861830 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: 1607852953 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: -167421191 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: -159247458 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: -930732721 - frame_counter: 4096 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: 676100017 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: 1922434160 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464562: frame detected - frame_len: -1368340446 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464563: frame detected - frame_len: 1452680282 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464563: frame detected - frame_len: 222145505 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464563: frame detected - frame_len: 1805721331 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464563: frame detected - frame_len: -1127568674 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464563: frame detected - frame_len: -610429418 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464564: frame detected - frame_len: -333357970 - frame_counter: 4096 - frame_counter_last_valid: -1 - resolution: 0
1613716464564: frame detected - frame_len: -1373361170 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464564: frame detected - frame_len: 1533346832 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464564: frame detected - frame_len: -474773114 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464564: frame detected - frame_len: -874228468 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464565: frame detected - frame_len: 1457123407 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464565: frame detected - frame_len: -856669482 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464565: frame detected - frame_len: -760251360 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464565: frame detected - frame_len: -294177078 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464565: frame detected - frame_len: -1942040745 - frame_counter: 4096 - frame_counter_last_valid: -1 - resolution: 0
1613716464567: frame detected - frame_len: 2050080607 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464569: frame detected - frame_len: 1176022084 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464569: frame detected - frame_len: -1411763502 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464569: frame detected - frame_len: -680948136 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464569: frame detected - frame_len: -441593828 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464571: frame detected - frame_len: -1697964740 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464571: frame detected - frame_len: -1136144334 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464573: frame detected - frame_len: -1218884642 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464573: frame detected - frame_len: 865166238 - frame_counter: 4096 - frame_counter_last_valid: -1 - resolution: 0
1613716464573: frame detected - frame_len: -1978975029 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464574: frame detected - frame_len: 1423422057 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464574: frame detected - frame_len: 1924131414 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464575: frame detected - frame_len: 1676886521 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464575: frame detected - frame_len: -709266715 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716464575: frame detected - frame_len: 1192481700 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716464575: SPS   detected - frame_len: 265281002 - frame_counter: 2082 - frame_counter_last_valid: -1 - resolution: 0
1613716464575: frame detected - frame_len: 16857035 - frame_counter: 2052 - frame_counter_last_valid: -1 - resolution: 0
1613716464575: frame detected - frame_len: -295173888 - frame_counter: 2049 - frame_counter_last_valid: -1 - resolution: 0
1613716464576: frame detected - frame_len: 1804005224 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
[... cut ...]
1613716542059: frame detected - frame_len: 432125980 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716542360: frame detected - frame_len: -1777346678 - frame_counter: 1088 - frame_counter_last_valid: -1 - resolution: 0
1613716542360: frame detected - frame_len: 404647145 - frame_counter: 4096 - frame_counter_last_valid: -1 - resolution: 0
1613716542561: frame detected - frame_len: 732212809 - frame_counter: 2048 - frame_counter_last_valid: -1 - resolution: 0
1613716542561: frame detected - frame_len: -1735042210 - frame_counter: 1088 - fram

roleoroleo, 2021-02-19:
The log is completely wrong.
I need a copy of your fshare_frame_buf file.

rt400, 2021-02-19:
https://wtools.io/paste-code/b3KD
@roleoroleo this ??

roleoroleo, 2021-02-19:
I need it in binary form.
Download it using ftp or scp.
It's in /dev/shm

rt400, 2021-02-19:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6009230/fshare_frame_buf.zip)
@roleoroleo hope now its OK

roleoroleo, 2021-02-19:
Sorry, my mistake.
I forgot a parmeter in the cmdline:
- Login with ssh
- killall wd_rtsp.sh
- killall rRTSPServer
- . /tmp/sd/yi-hack/script/env.sh
- RRTSP_RES=both RRTSP_AUDIO=no RRTSP_MODEL=r30gb RRTSP_DEBUG=1 rRTSPServer
- Then run a RTSP client and connect to the cam

rt400, 2021-02-19:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6010331/fshare_frame_buf.zip)

roleoroleo, 2021-02-19:
fshare_frame_buf is ok.
Send me the log using the command line of the last message.

rt400, 2021-02-20:
```1613738007283: frame_len: 1371 - cb_current->size: 262144
1613738007283: frame_len: 1371 - frame_counter: 36535 - resolution: 1080
1613738007283: frame detected - frame_len: 877 - frame_counter: 44040 - frame_co                                                                             unter_last_valid: 36535 - resolution: 0
1613738007309: frame detected - frame_len: 871 - frame_counter: 35542 - frame_co                                                                             unter_last_valid: 35542 - resolution: 360
1613738007334: frame_len: 871 - cb_current->size: 49152
1613738007334: frame_len: 871 - frame_counter: 35542 - resolution: 360
1613738007334: frame detected - frame_len: 1311 - frame_counter: 36536 - frame_c                                                                             ounter_last_valid: 36536 - resolution: 1080
1 - fFrameSize 871 - fMaxSize 60147 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 871 - fMaxSize 59276 - fLimitNumBytesToStream 0 - fPreferredFrame                                                                             Size 0
1613738007359: frame_len: 1311 - cb_current->size: 262144
1613738007359: frame_len: 1311 - frame_counter: 36536 - resolution: 1080
1613738007359: frame detected - frame_len: 876 - frame_counter: 35543 - frame_co                                                                             unter_last_valid: 35543 - resolution: 360
1613738007385: frame_len: 876 - cb_current->size: 49152
1613738007385: frame_len: 876 - frame_counter: 35543 - resolution: 360
1613738007385: frame detected - frame_len: 1936 - frame_counter: 36537 - frame_c                                                                             ounter_last_valid: 36537 - resolution: 1080
1 - fFrameSize 876 - fMaxSize 59276 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 876 - fMaxSize 58400 - fLimitNumBytesToStream 0 - fPreferredFrame                                                                             Size 0
1613738007410: frame_len: 1936 - cb_current->size: 262144
1613738007410: frame_len: 1936 - frame_counter: 36537 - resolution: 1080
1613738007410: frame detected - frame_len: 854 - frame_counter: 35544 - frame_co                                                                             unter_last_valid: 35544 - resolution: 360
1613738007435: frame_len: 854 - cb_current->size: 49152
1613738007435: frame_len: 854 - frame_counter: 35544 - resolution: 360
1613738007435: frame detected - frame_len: 1435 - frame_counter: 36538 - frame_c                                                                             ounter_last_valid: 36538 - resolution: 1080
1 - fFrameSize 854 - fMaxSize 58400 - fLimitNumBytesToStream 0
1 - Completed
0 - fFrameSize 854 - fMaxSize 57546 - fLimitNumBytesToStream 0 - fPreferredFrame                                                                             Size 0
```

roleoroleo, 2021-02-20:
I need a longer log.
Activate the putty logging and let it run for a few minutes.

GizmoLibre, 2021-02-21:
Hi, it's me from the other thread. Did the log with my r30gb Yi Dome Guard. Did your parameters in Putty, and the weird thing is, high res stream seems stable while doing this test in VLC. 

[putty.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6018207/putty.log)

After rebooting the camera and trying again in VLC, this time with sound over rtsp stream (i see your parameter is audio=no for testing), high res stream freezes after two minutes, both video and sound. 

Dunno what the logs says to you, but while doing the parameters over ssh you provided high res stream seems ok, how can we make that permanent? :D

roleoroleo, 2021-02-21:
The log shows how the frames are detected and captured from the buffer.
This is the critical part of the process.
And the log is ok, no lost frames, no errors.

The streaming part is out of my control/knowledge because is responsibility of the live555 library.
If you noticed errors only when the audio is active maybe one of the following problems:
- getting audio frames
- synchronization
- performance (cpu or memory are low)

I will check the 1st and maybe the 2nd but I can't do anything about 3.

roleoroleo, 2021-02-21:
Remember, when you start a test, to wait a minute or two after the reboot before start the player.

GizmoLibre, 2021-02-21:
Ok, i am gonna do some further more in depth testing, letting it on high res for longer periods of time, turning off sound over rtsp, and such, and i will report tomorrow what i found out. Hopefully we will see if the high res can be fixed or not.

GizmoLibre, 2021-02-22:
Here are my observations. Firstly, i was too quick to judge the high res stream as stable while testing with Putty commands, it is not stable at all. There is no fixed time after which it freezes in VLC, first time i was testing it was doing ok for 7,8 minutes, so i ended it there thinking it was stable. After that i did about 10 more tests to see how it goes. So there is no rule at all, once it froze 30 seconds into the test, other time 2 and half minutes, once around 4 minutes in, rarely it keeps working for 7+ minutes. The thing is, looking at the Putty logs, i see nothing suggesting the freeze happening, frame counter goes on in the log, even if it is frozen in VLC.

Rebooting the camera and testing it the "normal" way without running Putty, gives me the same results. So, often times the VLC will just crash instead of freezing the image. Launching it again, and opening the high res stream varies, sometimes it will just crash again, other times it will say it can't open the stream for a few times, then it will decide it does want to play, freezing again after X amount of time. One thing to note, i hoped disabling audio over RTSP changes things in terms of stability - i found out it has zero effect, so the audio is not to blame here. Main thing bothering me here, is how do i confirm the stream freezing with some actual data, as i said Putty logs show us that everything is okay, and it is far from it.

I tried to put RTSP stream on Both instead of High. As expected, this just makes things worse. Having both low and high RTSP stream enabled, further decreases stability of High stream, giving random green artifacts, sometimes rainbow colored ones, and every so often a completely gray image with some outlines for a few seconds. To check if the VLC is not at fault, i even tried couple of other RTSP supported players, like PMplayer, PotPlayer, Perfect IP camera viewer and such, all of them show freezing or instability of high res stream.

As for Home Assistant integration, i am baffled why the high stream won't load at all, because low res one works great. I tried to integrate it with the [Generic component](https://www.home-assistant.io/integrations/generic), same one works with low res, but just changing that ch0_1.h264 (low) to ch0_0.h264 (high) it does not want to load. Tried it with rtsp2webrtc add on, the same one works with low res, but not with high res. Also i mentioned Onvif, that too works on low res, but not with high res, it just won't load the stream. So basically, high res stream only works through VLC and such programs, being not stable at all, so all in all, in this state is not usable, so i guess i will stick to low for now. 

Here is another much larger Putty log that i paused right after the stream froze in VLC, but i see nothing at the end of the log suggesting that it did. 

[putty2.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6021895/putty2.log)

If you can have some suggestions what i can test next, i would be glad to help.

roleoroleo, 2021-02-22:
The log is perfect, so the problem isn't related to the frame extraction procedure.
There are 3 different possibilities:
- a problem during internal buffer reading/writing (the extractor fill a circular buffer and the live555 library read it).
- a live555 internal problem.
- a performance issue (network, memory or cpu).

If you have time, please run these tests:
- Try to use ethernet (if I'm not mistaken r30gb has ethernet) to exclude a wifi issue.
- Run h264grabber instead of rRTSPServer and redirect to a file, then try to read the h265 stream with another app.
- If the file at previuous step is ok, try to stream it with the live555 test app:
[testOnDemandRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6022642/testOnDemandRTSPServer.gz)

closed 2022-06-11 by roleoroleo