roleoroleo/yi-hack-Allwinner-v2 #97: Yi Outdoor 1080p (IFUS) working with Allwinner-v2

do1osc, 2021-02-18:
Short Info:

Allwinnwer-v2 works on Yi **Outdoor** 1080p (IFUS) FW: 9.0.19

Maybe some other persons can test it, too.

If ir works reliable, please add it to Support-List

roleoroleo, 2021-02-18:
What suffix did you use?

do1osc, 2021-02-18:
prefix: y21ga
what did you mean with suffix ?

do1osc, 2021-02-18:
it also works with firmware: 9.0.26.05_202011201756

roleoroleo, 2021-02-18:
I mean y21ga.
Are you able to make a backup copy to share with me?
I would like to check the difference between home and outdoor and confirm that the hack is ok for this model.

do1osc, 2021-02-18:
What i have to do for getting a backup ?

roleoroleo, 2021-02-18:
Read the wiki:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)

DennisUsesRPA, 2021-03-03:
Hi, 

I also have the IFUS yi outdoor but I can't seem to manage to hack the entity. Also this firmware: 9.0.26.05_202011201756

Franck69570, 2021-03-27:
Hi.

I have the IFUS yi outdoor with firmware 9.0.26.05_202011201756 and I successfully hacked the cam with y21ga_0.1.5.tgz

Thanks !

roleoroleo, 2021-03-27:
@Franck69570 
Yes, thanks.
I would like to check if there is any problem.

Franck69570, 2021-03-27:
Here you go : [backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6216379/backup.zip)

roleoroleo, 2021-03-28:
The backup is ok, thanks.

WInetu007, 2021-04-10:
> Hi.
> 
> I have the IFUS yi outdoor with firmware 9.0.26.05_202011201756 and I successfully hacked the cam with y21ga_0.1.5.tgz
> 
> Model YHS.3020
> FCC ID : 2AFIB-YHS3020
> IC ID : 20436-YHS3020
> 
> IFUSY595NJ
> 
> Thanks !
> 
> @roleoroleo, did you get a bakcup ? Do you want mine ?

works with the same cofnfiguration

roleoroleo, 2021-04-12:
Your model is h30ga.
I will add the support for this model in the next release.

Max1963, 2021-04-16:
First of all many thanks to roleoroleo for the hard work on those camera; also my two outdoor cameras are IFUSY595RTP with firmware 9.0.26.05_202011201756 and are both successfully hacked with y21ga_0.1.5.tgz; must change to h30ga?

roleoroleo, 2021-04-16:
It would be better.
For future updates.

Max1963, 2021-04-16:
Ok many thanks, will do soon!

closed 2021-05-16 by roleoroleo

radoprajitno, 2022-10-06:
Hi.
Thanks for you hard work Roleoroleo.
I have  Yi Outdoor Camera 1080 (IFUSY59), Firmware 9.0.28.08
I've did hack (with factory.done visible on the sd card)
However, Im unable to open the config web on my ip port:8080
the camera still works fine with the Yi Apps

Any suggestion?

roleoroleo, 2022-10-06:
9.0.28 means a new model, at the moment not supported.
Please, send me a backup of the flash.

radoprajitno, 2022-10-06:
Just updates::
I succeedly hack the camera XD
It was SD card issues

It was 9.0.26.08...
I did miss type it

All i did is dig up my sd card reader somewhere from my storage boxes.

I was doing ftp file transfer (SD card from my phones)
It did works for my y25ga cameras.

Somehow it did not work for h30ga.

After it plugged directly to sd card reader,
Re-did the hacking and it just works.

Http server, rstp, and everything works just as expected.

Thanks roleo!
Much appreciate it