roleoroleo/yi-hack-Allwinner-v2 #98: Unstable highreshighres rtsp sgream

uklatt, 2021-02-19:
Hello,

first of all: Thank you for this hack!

I modified my new Yi 1080p home camera with this hack and connect it to my pvr (Shinoby).
With low res the stream works perfect.
With high res the stream is interrupted by errors very often.
Is this because of missing processing power of cam?
What can I do to change this?
Is it possible to improve performace?

My wifi connection is perfect and Shinobi can handle much more complex streams (I have another working 5k res camera)

Uwe

roleoroleo, 2021-02-19:
Test the stream with vlc and check if you get errors.
Then, try to run the process manually and send me the log.

- Login with ssh with putty
- killall wd_rtsp.sh
- killall rRTSPServer
- . /tmp/sd/yi-hack/script/env.sh
- Start the logging in putty
- RRTSP_RES=both RRTSP_AUDIO=no RRTSP_DEBUG=1 rRTSPServer
- Then run a RTSP client and connect to the cam
- Wait for the error.
- Stop logging

uklatt, 2021-02-20:
I have tried to catch two logs when VLC shows scrambled images.
I can see nothing in the logs 

Interesting:
When I check both strams (high an low) with VLC at the same time, I can see problems in "high" stream but at the same time the "low" stream works without visible distortions.

[log1.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6014558/log1.txt)
[log2.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6014559/log2.txt)

Uwe

roleoroleo, 2021-02-20:
The log is ok.
I think it's a performance issue.
A mixed combination of network and cpu/memory usage.
The low stream is very light compared to the high stream and probably works better when the cam is overloaded.

uklatt, 2021-02-20:
OK, I logged the following "errors" at client side:
```
[h264 @ 0x563e01113ec0] error while decoding MB 101 4, bytestream -6
[h264 @ 0x563e0119c980] top block unavailable for requested intra mode [h264 @ 0x563e0119c980] error while decoding MB 70 0, bytestream 581
[h264 @ 0x563e0124cbc0] cabac decode of qscale diff failed at 30 59 [h264 @ 0x563e0124cbc0] error while decoding MB 30 59, bytestream 1207
[h264 @ 0x563e0119c980] error while decoding MB 68 14, bytestream -5
[h264 @ 0x563e0124cbc0] top block unavailable for requested intra mode [h264 @ 0x563e0124cbc0] error while decoding MB 71 0, bytestream 660
[h264 @ 0x563e0124cbc0] top block unavailable for requested intra mode -1 [h264 @ 0x563e0124cbc0] error while decoding MB 90 0, bytestream 626 
```

Is there something I can do to stabilize the stream?
Or is it possible to increase the performance of a "high" stream?

Is it correct that rRTS Server is responsible for streaming the data? 
Can we optimize this code further?

Uwe

roleoroleo, 2021-02-20:
> Is there something I can do to stabilize the stream?

Did you try to use tcp transport?
ffmpeg -rtsp_transport tcp .....

> Or is it possible to increase the performance of a "high" stream?

Are you using swap file? You could try to add it.

> Is it correct that rRTSServer is responsible for streaming the data?
> Can we optimize this code further?

Yes, rRTSPServer is the RTSP server. The code is composed by two parts:
- A function that reads the h264 stream from a buffer.
- The library (live555) that creates the rtsp stream.

The 1st part is my code and could be optimized.
The 2nd part, I don't know.

roleoroleo, 2021-02-20:
If you want to separate the two parts for testing, try h264grabber.
It reads h264 stream from the buffer and sends it to output.
The code is not used, but it's available in the hack. It should work.

uklatt, 2021-02-20:
I use TCP for transport protocol.
I tested with and without SWAP. (Memory seems to be no problem, as can be seen in "top"). With SWAP makes no differnce.
Would it help to "disable" all low-res code in rRTSPServer?
It looks like some low-res code is still executed when I enable high-res only.
I will try to get familar with h265grabber ;-)

Uwe

uklatt, 2021-02-20:
I used the following command while vieweing "high-res" stream in VLC (with visible errors).
`h264grabber_h -d > /dev/null`
The log doesn't show any errors. I would say your code is good ;-)

Coud be a performance problem with live555 lib.
Are there options to finetune this lib (disable unneeded features, use lighter transport protocol)?

Uwe

roleoroleo, 2021-02-20:
h264grabber is complicated.
It needs some explanation.

Here is what I know regarding the /dev/shm/fshare_frame_buf file.
The file is present, even if slightly different, also in the other cam versions.
It is a circular buffer of about 1.7 MB which contains the video streams (without audio): both high and low res.
Access to the file is regulated by a series of semaphores and notification files which are always found in /dev/shm
In particular, a semaphore for reading and one for writing:
-rw-r--r--    1 root     root       1786156 Apr 15 19:13 fshare_frame_buf
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_lock
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_0
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_1
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_10
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_11
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_12
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_13
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_14
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_15
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_16
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_2
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_3
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_4
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_5
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_6
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_7
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_8
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_read_notify_9
-rw-r--r--    1 root     root            16 Apr 15 19:05 fshare_write_lock

The file has a 300 (or 368, it depends on the model) byte header that is difficult to understand if you don't disassemble rmm.
However at address 0x12 and address 0x16 it seems that there are circular buffer read/write indexes.

These indexes point to a buffer address where a frame is present.
Each frame consists of a 22 (or 28) byte header and a payload.
The first 4 bytes define the size of the frame (without header), an absolute frame counter follows.
Bytes 19 and 20 (or 23 and 24) appear to be a frame counter differentiated for high res and low res.
Byte 18 is a resolution identifier: 8 for low res and 4 for high res.

Some other info.
The SPS frame is an exception to the others because it does not start immediately with the NALU but contains another 6 byte header.
The other frames all begin with the NALU 0x00000001.
SPS-PPS-IDR are always in sequence all together and after IDR there is always a frame that seems to have nothing to do with the video (it does not start with a NALU) but has a header with a frame counter.

The program reads the address in the header at 0x16 and takes it as the upper limit.
From there he starts looking for NALUs and waits for the first SPS.
Once this is done, it looops looking for all subsequent NALU checking the progressive counter.
The algorithm can certainly be improved, if someone wants to go further.

r30gb uses h265 for high res and h264 for low res.
The other models use only h264.

roleoroleo, 2021-02-20:
> Could be a performance problem with live555 lib.
> Are there options to finetune this lib (disable unneeded features, use lighter transport protocol)?

I don't know...
C++ is not my best friend.
I searched for a long time for another rtsp server but couldn't find anything light enough.
If you build the repo, you can try live555 test programs.
For example testOnDemandRTSPServer is able to stream from a video file and it's useful to test the streaming part.

uklatt, 2021-02-20:
I am trying to compile my own version for tests but are stuck at building rRTSPServer.
(Other components up to this are created.)

I got the following error:

```
arm-openwrt-linux-g++ -c -IUsageEnvironment/include -Igroupsock/include -IliveMedia/include -IBasicUsageEnvironment/include -I. -I./include -O1 -DSOCKLEN_T=socklen_t -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -DNO_OPENSSL=1 -Wall -DBSD=1   -o src/WAVAudioFifoServerMediaSubsession.o src/WAVAudioFifoServerMediaSubsession.cpp
arm-openwrt-linux-g++ -c -IUsageEnvironment/include -Igroupsock/include -IliveMedia/include -IBasicUsageEnvironment/include -I. -I./include -O1 -DSOCKLEN_T=socklen_t -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -DNO_OPENSSL=1 -Wall -DBSD=1   -o src/ADTSFromWAVAudioFifoSource.o src/ADTSFromWAVAudioFifoSource.cpp
src/ADTSFromWAVAudioFifoSource.cpp: In static member function 'static ADTSFromWAVAudioFifoSource* ADTSFromWAVAudioFifoSource::createNew(UsageEnvironment&, const char*)':
src/ADTSFromWAVAudioFifoSource.cpp:43:9: warning: unused variable 'ErrorStatus' [-Wunused-variable]
     int ErrorStatus;
         ^~~~~~~~~~~
src/ADTSFromWAVAudioFifoSource.cpp: In member function 'virtual void ADTSFromWAVAudioFifoSource::doGetNextFrame()':
src/ADTSFromWAVAudioFifoSource.cpp:179:9: warning: variable 'ErrorStatus' set but not used [-Wunused-but-set-variable]
     int ErrorStatus;
         ^~~~~~~~~~~
arm-openwrt-linux-g++ -c -IUsageEnvironment/include -Igroupsock/include -IliveMedia/include -IBasicUsageEnvironment/include -I. -I./include -O1 -DSOCKLEN_T=socklen_t -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -DNO_OPENSSL=1 -Wall -DBSD=1   -o src/ADTSFromWAVAudioFifoServerMediaSubsession.o src/ADTSFromWAVAudioFifoServerMediaSubsession.cpp
arm-openwrt-linux-g++ -c -IUsageEnvironment/include -Igroupsock/include -IliveMedia/include -IBasicUsageEnvironment/include -I. -I./include -O1 -DSOCKLEN_T=socklen_t -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -DNO_OPENSSL=1 -Wall -DBSD=1   -o src/aLawAudioFilter.o src/aLawAudioFilter.cpp
arm-openwrt-linux-g++ -c -IUsageEnvironment/include -Igroupsock/include -IliveMedia/include -IBasicUsageEnvironment/include -I. -I./include -O1 -DSOCKLEN_T=socklen_t -D_LARGEFILE_SOURCE=1 -D_FILE_OFFSET_BITS=64 -DNO_OPENSSL=1 -Wall -DBSD=1   -o src/DummySink.o src/DummySink.cpp
arm-openwrt-linux-g++ -orRTSPServer -L.  src/rRTSPServer.o src/ByteStreamCBMemorySource.o src/H264VideoCBMemoryServerMediaSubsession.o
src/H265VideoCBMemoryServerMediaSubsession.o src/CBMemoryServerMediaSubsession.o src/WAVAudioFifoSource.o src/WAVAudioFifoServerMediaSubsession.o
src/ADTSFromWAVAudioFifoSource.o src/ADTSFromWAVAudioFifoServerMediaSubsession.o src/aLawAudioFilter.o src/DummySink.o liveMedia/libliveMedia.a
groupsock/libgroupsock.a BasicUsageEnvironment/libBasicUsageEnvironment.a UsageEnvironment/libUsageEnvironment.a  ./lib/libfdk-aac.a -lpthread
./lib/libfdk-aac.a: error adding symbols: Archive has no index; run ranlib to add one
collect2: error: ld returned 1 exit status
make: *** [Makefile:83: rRTSPServer] Error 1
```
I can see the compiled libfdk-acc.a in libfdk subfolder, but there is no library in live subfolder.

Uwe

roleoroleo, 2021-02-20:
Try to pull the repo now.
I fixed a bug in the compilation makefile.

uklatt, 2021-02-20:
OK, now I was able to build anything 8-)
There was only one problem left: The gsoap-2.8.92 lib must be downloaded from a different location!
I have done it manually.

Now I can dig deeper...
Uwe

roleoroleo, 2021-02-20:
844KB is ok.
Recently I added aac audio and the new binary is bigger.

uklatt, 2021-02-21:
I've tested with better compiler optimizations for rRTSPServer, live andd fdk-aac  (-O3 instead of -O1), but that doesn't help (just grows the binary) ;-)
What do you think of using ffmpeg to deliver streams from h264grabber? Have you tried that already?

Uwe

roleoroleo, 2021-02-21:
I didn't tried.

uklatt, 2021-02-21:
I used your newest **h264grabber** and push the output with a pipe to **live555MediaServer** (live555MediaServer produces the rtsp stream). This creates simmilar glitches in the video stream.

I think, it's a synchronisation problem (finding the start of the next frame).
The data from **/dev/shm/fshare_frame_buf** delivers both high and low quality streams and we need a better sync for detection of the next frame start.

I fiddled arround with the delay parameter **USLEEP** an now set it to 24mS (because both streams deliver 20 frames a second  -> one frame every 25 mS). This seems to work better but not perfect.

Now we should optimize the frame start detection...

Uwe

roleoroleo, 2021-02-21:
Glad you are working on this issue. Many users have reported problems but I have never been able to find and fix them.
Keep me updated.

GizmoLibre, 2021-02-21:
Same issue with 0.14 and 0.13 firmware, tested both on my Yi Dome Guard (r30gb). High res rtsp is very unstable, it works for a two or three minutes, and just freezes. Also, Onvif on high res does not work at all, it won't even start streaming in Home Assistant, just gray video, and also in Onvif Device Manager testing software for Windows saying "No Signal". 

Having it on high res and fiddling with Onvif, also gives random reboots of the camera, number of times also just going through the config page causes reboots. Switching to low res fixes rtsp and onvif, on low res it works flawlessly, never had a reboot or other issues. 

How can i help giving some actual logs or something? I am a begginer and don't understand what is going on "behind the scenes". Cheers for the hack, just having rtsp and all the options means a lot.

uklatt, 2021-02-21:
You can download a SDK for the SoC (Allwinner V831) from [http://www.sochip.com.cn/v831/index.php?title=Downloading_the_Source](http://www.sochip.com.cn/v831/index.php?title=Downloading_the_Source)

Here are precompiled packages for V831 [http://mirror.sipeed.com/maix_ii/base/](http://mirror.sipeed.com/maix_ii/base/).

I downloaded "camerademo" and it worked! (First you need to stop rmm process and watchdog)
This program grabs images from /dev/video0 and writes it to destinatipn path.
The pictures are to bright, but viewable.

```
root@Yicam:/tmp# camerademo debug
[CAMERA]**********************************************************
[CAMERA]*                                                        *
[CAMERA]*              this is camera test.                      *
[CAMERA]*                                                        *
[CAMERA]**********************************************************
[CAMERA]**********************************************************
[CAMERA] open /dev/video0!
[CAMERA]**********************************************************
[CAMERA_DEBUG] Querey device capabilities succeed
[CAMERA_DEBUG] cap.driver=sunxi-vin
[CAMERA_DEBUG] cap.card=sunxi-vin
[CAMERA_DEBUG] cap.bus_info=
[CAMERA_DEBUG] cap.version=0x00010000
[CAMERA_DEBUG] cap.capabilities=0x85201000
[CAMERA]**********************************************************
[CAMERA] The path to data saving is /tmp.
[CAMERA] The number of captured photos is 5.
[CAMERA] save bmp format
[CAMERA] do not use watermarks
[CAMERA_DEBUG]**********************************************************
[CAMERA_DEBUG] enumerate image formats
[CAMERA_DEBUG] format index = 0, name = YUV422P
[CAMERA_DEBUG] format index = 1, name = NV16
[CAMERA_DEBUG] format index = 2, name = NV61
[CAMERA_DEBUG] format index = 3, name = YUV420
[CAMERA_DEBUG] format index = 4, name = YVU420
[CAMERA_DEBUG] format index = 5, name = NV12
[CAMERA_DEBUG] format index = 6, name = NV21
[CAMERA_DEBUG] format index = 7, name = YUYV
[CAMERA_DEBUG] format index = 8, name = UYVY
[CAMERA_DEBUG] format index = 9, name = VYUY
[CAMERA_DEBUG] format index = 10, name = YVYU
[CAMERA_DEBUG] format index = 11, name = YUYV
[CAMERA_DEBUG] format index = 12, name = UYVY
[CAMERA_DEBUG] format index = 13, name = VYUY
[CAMERA_DEBUG] format index = 14, name = YVYU
[CAMERA_DEBUG]*********************************************************
[CAMERA_DEBUG] The sensor supports the following formats :
[CAMERA_DEBUG] Index 0 : YUV422P.
[CAMERA_DEBUG] Index 1 : NV16.
[CAMERA_DEBUG] Index 2 : NV61.
[CAMERA_DEBUG] Index 3 : YUV420.
[CAMERA_DEBUG] Index 4 : YVU420.
[CAMERA_DEBUG] Index 5 : NV12.
[CAMERA_DEBUG] Index 6 : NV21.
[CAMERA_DEBUG] Index 7 : YUYV.
[CAMERA_DEBUG] Index 8 : UYVY.
[CAMERA_DEBUG] Index 9 : VYUY.
[CAMERA_DEBUG] Index 10 : YVYU.
[CAMERA_DEBUG] Index 11 : YUYV.
[CAMERA_DEBUG] Index 12 : UYVY.
[CAMERA_DEBUG] Index 13 : VYUY.
[CAMERA_DEBUG] Index 14 : YVYU.
[CAMERA_DEBUG]**********************************************************
[CAMERA_DEBUG] The YUV422P supports the following resolutions:
[CAMERA_DEBUG] Index 0 : 1936 * 1096
[CAMERA_DEBUG]**********************************************************
[CAMERA_DEBUG] The NV16 supports the following resolutions:
[CAMERA_DEBUG] Index 0 : 1936 * 1096
[CAMERA_DEBUG]**********************************************************
[... 66 lines cut ...]
[CAMERA_DEBUG]************QBUF[1] FINISH**************
[CAMERA] capture num is [2]
[CAMERA_DEBUG]*****DQBUF[2] FINISH*****
[CAMERA_DEBUG] the interval of two frames is 452 ms
[CAMERA_DEBUG]************QBUF[2] FINISH**************
[CAMERA] capture num is [3]
[CAMERA_DEBUG]*****DQBUF[0] FINISH*****
[CAMERA_DEBUG] the interval of two frames is 600 ms
[CAMERA_DEBUG]************QBUF[0] FINISH**************
[CAMERA] capture num is [4]
[CAMERA_DEBUG]*****DQBUF[1] FINISH*****
[CAMERA_DEBUG] the interval of two frames is 400 ms
[CAMERA_DEBUG]************QBUF[1] FINISH**************
[CAMERA] Capture thread finish
[CAMERA_DEBUG]***************************************************************
[CAMERA_DEBUG] Query the actual frame rate.
[CAMERA_DEBUG] camera fps = 20.4.
[CAMERA_DEBUG]***************************************************************
[CAMERA] close /dev/video0
[CAMERA_DEBUG]***************************************************************
[CAMERA_DEBUG] Performance Testing---format:NV21 size:1936 * 1096
[CAMERA_DEBUG] The interval from open to streaming is 183 ms.
[CAMERA_DEBUG]***************************************************************

```

roleoroleo, 2021-02-21:
@GizmoLibre 
I think your problem is different.
r30gb is in beta testing and it's a know problem.
Please post your situation in this issue: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/96
Note: onvif uses rtsp to stream video, so if rtsp doesn't work, onvif doesn't work.
About reboots, try to activate swap file.

roleoroleo, 2021-02-21:
@uklatt 
Interesting! I will try it.
A few months ago i tried to build some demo from this repo:
https://github.com/lindenis-org
https://github.com/lindenis-org/lindenis-v536-softwinner
but I had no luck.

roleoroleo, 2021-02-21:
@uklatt 
To get sources from sochip, did you follow the registration procedure?

uklatt, 2021-02-21:
Yes, I have registered...

uklatt, 2021-02-24:
Hello roleoroleo,

I was able to compile the **camerademo**. I have attached the source (plus binary)...
You can add the source folder in your repo and it will compile with your existing scripts.
The code is from [https://github.com/lindenis-org/lindenis-v536-package](https://github.com/lindenis-org/lindenis-v536-package)
This program runs without **rmm** and can catch pictures directly from **/dev/video0**.
You need to kill rmm bevor running it. Missing is control of brightness and contrast, but this should be possible with v4l2 ioctrl commands.

This could be a start for reading h264 streams directly from **/dev/videox** and use **live555** for RTSP streaming.

Uwe
[camerademo.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6035653/camerademo.zip)

roleoroleo, 2021-02-24:
Thank you, I will try it asap.
What about audio? Is there an eample in the demo code?

uklatt, 2021-02-24:
I have not checked audio. I need only a stable h264 high RTSP strem ;-)

roleoroleo, 2021-02-24:
I killed watch_process but it's not enough.
When I kill rmm the cam restarts.
Which process do you kill to stop watchdog?

roleoroleo, 2021-02-24:
I found 2 interesting projects:
https://github.com/ykst/librtsp
https://github.com/txgcwm/Linux-C-Examples

uklatt, 2021-02-24:
I have killed wd_rtsp.sh, rRTSPServer, onvif_srvd, dispatch, rmm, ipc_multiplexer, framefinder and the camera is still running.

uklatt, 2021-03-07:
I have tried to access /dev/video0 directly with v4l2 API.
It looks like there is only an incomplete driver implementation.
I have attached the output of "v4l2_compliance".

[compliance-test.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6097649/compliance-test.txt)

Uwe

uklatt, 2021-03-08:
I managed to do the following:
- Use v4l2 API to stream raw yuv420 or nv12 data to stdout.
- Use ffmpeg to convert this stream to h264/mp4 file (software encoder).

Without hardware acceleration the conversion is too slow.
We need code for using cedrus/cedarc hardware h264 encoder. (This is used from original app **rmm**)

Uwe

roleoroleo, 2021-03-09:
I agree with you.

roleoroleo, 2021-03-10:
According to your tests, the problem lies in the extraction of the frames from the buffer or in the rtsp server?

uklatt, 2021-03-10:
Thr rtsp server seems to work without problems.
I would say, the extraction out of the ring buffer sometimes fails.
I will try your newest version...

roleoroleo, 2021-03-10:
We could try to use this: https://github.com/Necromix/yi-vencrtsp_v2

roleoroleo, 2021-03-10:
Did you try rRTSPServer also?
It contains timing info and other improvements.

uklatt, 2021-03-10:
Yes, but newest rRTSPServer still shows artefacts and stream dropouts.
The combination of h264grabber + live555Mediaserver works better.

roleoroleo, 2021-03-12:
I will investigate.

roleoroleo, 2021-03-13:
> I checked your last version (with time infos).
> When I use the new h264grabber together with live555Streamer I get a stable highres stream!
> 
> How to test:
> 
>     * load environment: **. /tmp/sd/yi-hack/script/env.sh**
> 
>     * kill a few processes: **killall wd_rtsp.sh; killall framefinder; killall onvif_srvd; killall rRTSPServer**
> 
>     * create a fifo: **mknod /tmp/file.h264 p**
> 
>     * start live555Mediaserver from /tmp: **cd /tmp; live555MediaServer &**
> 
>     * start h264grabber: **h264grabber > /tmp/file.264**
> 
> 
> Then connect VLC to camera: rtsp://camera-ip/file.264
> 
> Uwe

I tested this scenario and it really works very well.

uklatt, 2021-03-13:
I compiled your newest version of rRTSPServer and compared it with my combination of h264grabber + live555MediaServer.
Both versions have artefacts sometimes. rRTSPServer more often than the combination of h264grabber + live555MediaServer.
With h264grabber + live555MediaServer I have a scambled bottom of of the stream every 5 Minutes for 1-2 seconds.

roleoroleo, 2021-03-13:
The log is ok.

> We could try to use this: https://github.com/Necromix/yi-vencrtsp_v2

I adapted and compiled it, if you want to try.
[rtsp_server.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6135427/rtsp_server.gz)

uklatt, 2021-03-13:
How to use rtsp_server?
Does it read h264 data from /tmp/view or /dev/shm/fshare_frame_buf?

roleoroleo, 2021-03-13:
Simply stop rRTSPServer and run this one.
It reads directly from /devshm/fshare_frame_buf.

uklatt, 2021-03-13:
When I stop rRTSPServer and start rtsp_server the vlc program doesn't show anything (rtsp://ip-of-camera/ch0_0.h264)
With rRTSPServer I can view the stream.

roleoroleo, 2021-03-13:
Try without ch0_0.h264

uklatt, 2021-03-13:
Doesn't work. (I tried UDP/TCP Port 554)

uklatt, 2021-03-14:
> 
> 
> The log is ok.
> 
> > We could try to use this: https://github.com/Necromix/yi-vencrtsp_v2
> 
> I adapted and compiled it, if you want to try.
> [rtsp_server.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6135427/rtsp_server.gz)
Isn't this only for High Silicon SoCs?

roleoroleo, 2021-03-14:
I adapted to y21ga.

roleoroleo, 2021-03-14:
I made a new report for this source.

https://github.com/roleoroleo/rtsp_server

I tried again and it works with my y21ga.

uklatt, 2021-03-14:
Thank you for the repo.
Now it runs after disabling audio.

I will check it...

roleoroleo, 2021-03-14:
Mi wifi is medium with this cam.
But I have a prevoius y20ga with a weak signal and I have a lot of problems with RTSP.

uklatt, 2021-05-05:
Hello,

I tried again the actual version but I have still no stabel video (stuttering, image artefacts, ...).
Here is an actual logfile from rRTSPServer:
[log_0.1.7.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6427951/log_0.1.7.txt)

Uwe

roleoroleo, 2021-05-05:
Better or worse than 0.1.6?

uklatt, 2021-05-05:
I would say, older versions are better tban actual version, 0.1.6 is the same as 0.1.7.
When the log is ok, I don't understand why the video stutters (it looks like missing frames).
I use VLC for testing.

Uwe

Maxym77, 2022-01-03:
> When I stop rRTSPServer and start rtsp_server the vlc program doesn't show anything (rtsp://ip-of-camera/ch0_0.h264) With rRTSPServer I can view the stream. [rtsp_server.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6135678/rtsp_server.log)

Hi, i have tested this binary on my IFUS camera and work very better compared to last release version.

But in your binary not is audio and username and password setting... how i can have an complete version of this ?

Maxym77, 2022-01-03:
> I made a new report for this source.
> 
> https://github.com/roleoroleo/rtsp_server
> 
> I tried again and it works with my y21ga.

how i can make binary? you can compile for me ?

roleoroleo, 2022-01-07:
Which binary?
https://github.com/roleoroleo/rtsp_server
This one?

I found this piece of code on the web but it's just a proof of concept to test the stream: it's not ready for production.
As you already noticed, there is no audio, no security, etc...

Maxym77, 2022-01-08:
Yes , the default rtsp binary in the last release package.

Maxym77, 2022-01-08:
yes.. is the same in the release package 0.1.9 ?

[33 more replies]

closed 2023-01-08 by stale[bot]