roleoroleo/yi-hack-Allwinner-v2 #99: Possible to make two-way audio work?

andreaswatch, 2021-02-19:
Hi, 
first of all, thanks a lot for this project, it works great and the wiki helps a lot with the first steps :-)
This is more of a question than a problem, so please feel free to close it and let me know where else I should post my questions.

I just wanted to know your opinion if it is possible to make two-way audio work.
I am very new to this project, so it may be that I misunderstood something, so please bear with me.

As far as I understand, the rmm process blocks the speaker, so my first attempt was to replace rmm with a dummy program (so the watchdog thinks it's still running), but that also breaks the video stream...so that doesn't seem like the best approach ;)

This raises a few questions for me:
- Do you think it's possible to read /dev/video0 directly with ffmpeg? (So that no rmm is needed).
- Has anyone tried reconfiguring alsa so that rmm blocks a virtual/dummy device?
- Has anyone tried compiling and running aplay?

In another issue it was described that rmm can be used to play audio with:
`/home/app/rmm "path_to_audio"`
or
`/home/app/rmm "path_to_audio" 1`
On my IFUSY44DAM with 9.0.19.02_202007141512, this replays the last recorded sound, but not the given soundfile...should it really be possible to play a custom audio file with this command?

Thank you guys!

roleoroleo, 2021-02-20:
Hi andreaswatch.

> Do you think it's possible to read /dev/video0 directly with ffmpeg? (So that no rmm is needed).

Afaik rmm is responsible to start something kernel side, so if you don't start rmm, /dev/video* are not available/usable.
But I have not been able to do many tests.

> Has anyone tried reconfiguring alsa so that rmm blocks a virtual/dummy device?

I didn't try.

> Has anyone tried compiling and running aplay?

Try attached file:
[utils.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6014514/utils.tar.gz)

andreaswatch, 2021-02-20:
Hi roleoroleo,
thank you for the alsa utils. I just wanted to let you know what I found.
aplay seems to work fine...but it is a bit tricky, my steps:
- create a copy of the original rmm
- replace rmm with a dummy
- start copy of rmm and use yi-app to access the stream, enable and disable microphone in app. rmm shows: rmm.c(send_do_speaking_msg-1718)[10:40:05.406]:msg snd success
- ctrl+c to kill rmm
- aplay audiofile.wav -> sound works.

It looks like rmm does some initialization work for the speaker and blocks the device, not sure how to come around this issue.
I will have a look with IDA, but I never did that before so there is a big chance to fail..

andreaswatch, 2021-02-26:
I hope the above info can help someone, however I am not a low level guy and fail with IDA (as expected).
Thank  you roleoroleo and feel free to close this issue.

roleoroleo, 2021-02-27:
I have tried several times in the past and have always failed.

roleoroleo, 2021-04-13:
Try the last version.
How it works:
1 - Create a raw pcm (wav without header): 16 KHz, 16 bit, mono (S16LE).
2 - Copy it to /tmp/audio_in_pcm

If you want to use a tts engine download nanotts from here:
https://github.com/roleoroleo/yi-hack-utils
and use with this webservice:
- http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
- POST
- Text message in the payload

andreaswatch, 2021-04-13:
@roleoroleo awesome, thank you!
`import requests
URL = "http://yidoor:8080/cgi-bin/speak.sh?lang=de-DE"
PARAMS = "Test 1 2 3 4"
r = requests.post(url = URL, data = PARAMS)
print(str(r.json()))`

RedKage, 2021-04-24:
Hello guys,

I am not sure, but I think since 0.1.6 I seem to have lost the audio.
Like, before I could hear what was going on near the camera.
Now the RTSP stream seems to carry only the video, there's no audio stream.
Strange

I am posting here as this looks related

9.0.19.09_202010291747 / y21ga / IFUS

roleoroleo, 2021-04-24:
Are you using rtsp with authentication?

RedKage, 2021-04-24:
> Are you using rtsp with authentication?

I didn't know authenticated RSTP was a thing.
I'm using the link in the webinterface

The high/res stream have only video (maybe that's actually normal... I dont recall if before 0.1.6 there was another stream inside)

The only audio stream (the ch0_2.h264) cannot be opened.
Tried VLC and MPC they both report can't open.

Maybe I need to revert to 0.1.5 to rule out any other user issue.

roleoroleo, 2021-04-24:
I have the same model: y21ga.
Tested now, it works.
Please check the following:
- Check if audio is enabled in your configuration.
- Check if /tmp/audio_fifo exists

RedKage, 2021-04-24:
> I have the same model: y21ga.
> Tested now, it works.
> Please check the following:
> 
> * Check if audio is enabled in your configuration.
> * Check if /tmp/audio_fifo exists

Yes audio is enabled with ulaw.
/tmp/audio_fifo doesn't exist

However I did try to make the sound output work
I did create at some point `/tmp/audio_in_fifo` to test a raw audio. But that didn't do anything.
The file disappeared though.

I can see there are the audio_fifo.requested and audio_in_fifo.requested files.

roleoroleo, 2021-04-25:
To make rtsp with audio working (using microphone), you need /tmp/audio_fifo
The file /tmp/audio_in_fifo is used to output sound to the speaker and it's not related to th rtsp stream.

The problem now is why you don't have /tmp/audio_fifo
I have found that some users have had problems during updat. Try reinstalling the firmware from scratch.

RedKage, 2021-04-25:
> To make rtsp with audio working (using microphone), you need /tmp/audio_fifo
> The file /tmp/audio_in_fifo is used to output sound to the speaker and it's not related to the rtsp stream.
> 
> The problem now is why you don't have /tmp/audio_fifo
> I have found that some users have had problems during updat. Try reinstalling the firmware from scratch.

Alright, so it turns out I do not see these files from FTP.
Which is strange, I kinda remember seeing them in FTP at some point ... hmmm
I even did upload the `audio_in_fifo` file using FTP to try out the new feature.

But using SSH both `audio_fifo`  and `audio_in_fifo` are present with rights `prw-r--r--`

roleoroleo, 2021-04-25:
Size 0 is normal because these are fifos, not normal files.
But with my cam works, so it's not possible a problem in the process binary.
Try to reinstall from scratch.

RedKage, 2021-04-25:
> Size 0 is normal because these are fifos, not normal files.
> But with my cam works, so it's not possible a problem in the process binary.
> Try to reinstall from scratch.

Thanks will do.
That means I need to use the "unhack" method 1st, and then install the hack again?

roleoroleo, 2021-04-25:
It's not mandatory to use unhack.
Simply clean your sd card and copy again the files.

RedKage, 2021-04-25:
> It's not mandatory to use unhack.
> Simply clean your sd card and copy again the files.

- Formatted the sd card
- Copied back the content of the hack_y21ga_0.1.6.tar to the sd card
- Powered on
- Restored my config
- Rebooted
- Still no mic sound

Just in case it was my config:
- Reset to yi-hack default
- Restarted
- Still no mic sound

I dont get it

roleoroleo, 2021-04-25:
I will check again with my cam.

RedKage, 2021-04-25:
Just in case that's useful
```
root@yi-853e:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  211 root         0 SW   [oom_reaper]
  212 root         0 SW<  [writeback]
  214 root         0 SW   [kcompactd0]
  215 root         0 SW<  [crypto]
  216 root         0 SW<  [bioset]
  218 root         0 SW<  [kblockd]
  259 root         0 SW   [sys_user]
  266 root         0 SW   [kworker/0:1]
  267 root         0 SW<  [cfg80211]
  274 root         0 SW<  [watchdogd]
  288 root         0 SW<  [spi0]
  298 root         0 SW   [kswapd0]
  382 root         0 SW<  [bioset]
  387 root         0 SW<  [bioset]
  392 root         0 SW<  [bioset]
  397 root         0 SW<  [bioset]
  402 root         0 SW<  [bioset]
  407 root         0 SW<  [bioset]
  412 root         0 SW<  [bioset]
  417 root         0 SW<  [bioset]
  422 root         0 SW<  [bioset]
  454 root         0 SW   [irq/302-sunxi-m]
  456 root         0 SW   [irq/166-sdc0 cd]
  457 root         0 SW   [irq/303-sunxi-m]
  486 root         0 SW   [kworker/u2:2]
  515 root         0 SW<  [bioset]
  516 root         0 SW   [mmcqd/0]
  535 root         0 SW<  [kworker/0:1H]
  548 root         0 SWN  [jffs2_gcd_mtd4]
  554 root       980 S    -/bin/sh
  596 root         0 SW   [ksdioirqd/mmc1]
  602 root         0 SW   [RTW_XMIT_THREAD]
  603 root         0 SW   [RTW_CMD_THREAD]
  604 root         0 SW   [RTWHALXT]
  645 root      1096 S    ./dispatch
 1145 root     40932 S    ./rmm
 1227 root      1400 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1302 root       980 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -x hostname:yi-853e
 1507 root      2944 S    ./mp4record
 1520 root       860 S    httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.conf
 1533 root       844 S    pure-ftpd (SERVER)
 1548 root       876 S    dropbear -R -B
 1550 root       668 S    ipc_multiplexer
 1581 root       860 S    ntpd -p pool.ntp.org
 1599 root      5940 S<   rRTSPServer
 1625 root       988 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.sh
 1665 root      8108 S    onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.6 --hardware_id --serial_num IFUSXXXXXXXXXXXXXXXXXXXXXX --ifs wlan0
 1676 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --type tdn:NetworkVideoTransmitter --xaddr http://%s --scope onvif://www.onvif.org/name/Unknown onvif://www.onv
 1677 root      2420 S    framefinder y21ga
 1686 root       864 S    /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
 7930 root       900 R    dropbear -R -B
 7939 root       984 S    -ash
 9053 root         0 SW   [kworker/0:2]
 9054 root         0 SW   [kworker/u2:0]
 9055 root         0 SW   [kworker/u2:3]
 9096 root       980 S    sleep 10
 9101 root       980 R    ps
```

RedKage, 2021-04-25:
Alright something new: when I use `AAC`, it works
Then the channel 2 stream becomes availlable, and I can open it in VLC. Sound is recorded I do hear stuff
What I know is that in 0.1.5 I used `pcm`, `alaw` and `ulaw` and they were all working

ivanfor, 2021-05-01:
> Alright something new: when I use `AAC`, it works
> Then the channel 2 stream becomes availlable, and I can open it in VLC. Sound is recorded I do hear stuff
> What I know is that in 0.1.5 I used `pcm`, `alaw` and `ulaw` and they were all working

Maybe you can try #147 rtsp binary

ivanfor, 2021-05-01:
> Try the last version.
> How it works:
> 1 - Create a raw pcm (wav without header): 16 KHz, 16 bit, mono (S16LE).
> 2 - Copy it to /tmp/audio_in_fifo
> 
> If you want to use a tts engine download nanotts from here:
> https://github.com/roleoroleo/yi-hack-utils
> and use with this webservice:
> 
>     * http://IP-CAM:8080/cgi-bin/speak.sh?lang=en-US
> 
>     * POST
> 
>     * Text message in the payload

@roleoroleo , this is really awesome. It's working flawlessly!

BTW, I think the speak.sh should have the following change, so the nanotts.tgz file can be just extracted under yi-hack folder:
```
ivan@sinchan:/tmp$ diff -u speak.sh speak.sh.orig 
--- speak.sh	2021-05-01 11:51:57.321764139 +0200
+++ speak.sh.orig	2021-05-01 11:52:02.265772012 +0200
@@ -36,7 +36,7 @@
     TMP_FILE="/tmp/sd/speak.pcm"
     if [ ! -f $TMP_FILE ]; then
         speaker on > /dev/null
-        echo "$POST_DATA" | /tmp/sd/yi-hack/bin/nanotts -l /tmp/sd/yi-hack/usr/share/pico/lang -v $LANG -c > $TMP_FILE
+        echo "$POST_DATA" | /tmp/sd/yi-hack/bin/nanotts -l /tmp/sd/yi-hack/share/pico/lang -v $LANG -c > $TMP_FILE
         dd if=$TMP_FILE of=/tmp/audio_in_fifo
         sleep 1
         speaker off > /dev/null
```

PS.: this would be the "curl" version:
`ivan@sinchan:~$ curl -v "http://root:xxxxxx@ctest.casa:8080/cgi-bin/speak.sh?lang=es-ES" -d "Hola"`

roleoroleo, 2021-05-01:
About the folder, I fixed the path both in speak.sh and in nanotts.tgz.
The right folder is /tmp/sd/yi-hack/usr/share/pico/lang
Try to download nanotts.tgz again.

I'm working to a new version that use ha tts engine.
So nanotts it's not necessary, if you use ha.

roleoroleo, 2021-05-01:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/147#issuecomment-830604831

ivanfor, 2021-05-01:
> I'm working to a new version that use ha tts engine.
> So nanotts it's not necessary, if you use ha.

That would be even better! Great job!

Mrivilla, 2021-05-12:
Hi Dears,
I installed nanotts and launch the speak.sh service with Curl. It seems all went fine:
{
    "error": "false",
    "description": "HOLA"
}

but I didn't hear anything. do I need to enable something else?
Thanks.

roleoroleo, 2021-05-12:
Did you install the libraries in the correct folder?
/tmp/sd/yi-hack/usr/share/pico/lang

Mrivilla, 2021-05-12:
> Did you install the libraries in the correct folder?
> /tmp/sd/yi-hack/usr/share/pico/lang

yes I did.
[image]
Also enabled Audio PCM
[image]

thanks.

Mrivilla, 2021-05-28:
I fixed, it was an issue in the speak.sh I had the wrong path version.
Thanks.

WalkingDot, 2021-06-09:
I also find these two functions very interesting.

For testing I created a simple page, with text-to-speech and wav upload:
[image]

https://github.com/WalkingDot/test/raw/main/speak.html.gz

Just place the "speak.html.gz" into the folder "/yi-hack/www/pages/"
And use it with:  "http://your-cam-ip:8080/index.html?page=speak"

roleoroleo, 2021-06-09:
Thanks for this example page.
If you agree, I could add it to the hack.

WalkingDot, 2021-06-09:
Sure ;-)
But it's just very basic html, without using jQuery.

WalkingDot, 2021-06-10:
I think I got it...
Here is the code: [speak.html](https://github.com/WalkingDot/test/blob/main/speak.html)

ToDo:
- volume and speed options
- save preferred options
- intercom mode / live recording
(I've found an example on mdn: [Dictaphone](https://developer.mozilla.org/en-US/docs/Web/API/MediaStream_Recording_API/Using_the_MediaStream_Recording_API) )

roleoroleo, 2021-08-18:
I added the page to the tree.

closed 2023-01-08 by stale[bot]