roleoroleo/yi-hack-Allwinner-v2 #101: ftp upload not working

albertopalotes35, 2021-02-22:
I have a problem with the ftp upload of the videos recorded by the camera, I have configured an ftp server in an asus router to which I have a pendrive connected, in the motion events settings, I have ftp upload enabled, in ftp host I have the router address 192.168.1.1, but it doesn't work, we don't upload the videos to the ftp server

roleoroleo, 2021-02-22:
Try to run the script ftppush.sh manually and check the logs.

albertopalotes35, 2021-02-22:
How I run the script, I don't know how to do it

roleoroleo, 2021-02-22:
Login with ssh and run the folling commands:
```
root@yi-hack-allwinner-v2:~# . /tmp/sd/yi-hack/script/env.sh
root@yi-hack-allwinner-v2:~# /tmp/sd/yi-hack/script/ftppush.sh start
```

albertopalotes35, 2021-02-22:
This is what I get when I run it manually by ssh, the time and date are not current because I don't have internet on the router where the camera is connected and I don't know how to manually set the date and time on the camera.

root@yi-96dd:~# . /tmp/sd/yi-hack/script/env.sh
root@yi-96dd:~# /tmp/sd/yi-hack/script/ftppush.sh start
2020-04-17 [22-29-25] [INFO] === SERVICE START ===
2020-04-17 [22-29-25] [INFO] checkFiles
ftpput: unexpected server response to STOR: 550 Permission denied (restricted operation).
2020-04-17 [22-29-26] [ERROR] uploadToFtp: ftpput FAILED.
2020-04-17 [22-29-26] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y04M17D22H/16M35S25.mp4].
ftpput: unexpected server response to STOR: 550 Permission denied (restricted operation).
2020-04-17 [22-29-26] [ERROR] uploadToFtp: ftpput FAILED.
2020-04-17 [22-29-26] [ERROR] checkFiles: uploadToFtp FAILED - [/tmp/sd/record/2020Y04M17D22H/17M00S60.mp4].
ftpput: unexpected server response to STOR: 550 Permission denied (restricted operation).

roleoroleo, 2021-02-22:
Line 5 says that the user does not have correct permissions.

albertopalotes35, 2021-02-22:
The ftp server user is admin, in the asus router's ftp server configuration I have created a video folder which has R / W permissions activated for the admin user and in the motion events configuration, in ftp directory I have selected the video folder that I have created on the ftp server

roleoroleo, 2021-02-22:
Try to login manually (eg Filezilla) to the ftp and try to upload a file.

albertopalotes35, 2021-02-23:
It already works for me, the problem was that the path of the videos folder of the ftp server was incorrectly configured in the camera configuration, the path that I had set was / Videos and the correct path is / sda1 / Videos

Thank you very much.

mik999, 2021-04-20:
Same here. Anon user gets cant' t change directory to /home/ftp and normal user seems not providing the correct password to the server.

What can I do?

laneyweb, 2021-04-28:
I get the same here too but my error is:

ftpput: unexpected server response to STOR: 553 12M00S60.mp4: Permission denied.

I have tested the ftpuser in the FTP server with WinSCP and FileZilla and the user / pass works and can create directories/files.

I have disabled unwanted services like ONVIF and Telnet and tried to enable the swap option., Nothing works. 

Any tips please?

roleoroleo, 2021-04-29:
Run it manually and post the full log:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/101#issuecomment-783599076

mik999, 2021-04-29:
In my case I'm an idiot)) I used camera's IP as ftp server address. Damn stupid)) Now it all works like a charm. Cam sends vids to a folder on the ftp server. That folder is synced to OneDrive and set as online only. So the vids don't take up any space on the ftp server and always available online.

Thanks Roleo!

laneyweb, 2021-04-29:
Thanks so much for the replies. I have captured the output showing the error after running the ftppush start. It's [here ](https://pastebin.com/dtqz4ZqH) in pastebin. Is this what you meant or is there a log file I need to grab too?

laneyweb, 2021-04-29:
> In my case I'm an idiot)) I used camera's IP as ftp server address. Damn stupid)) Now it all works like a charm. Cam sends vids to a folder on the ftp server. That folder is synced to OneDrive and [...]

I'm certainly one of those :) - I have checked the IP, user/pass many times.... so hope that's not it. Do you use a port number  in the FTP Host field? I am leaving mine as just IP as the FTP sever is on default port 21.

I wonder if it can be the FTP server. I may spin up a test FTP server in a docker but I'd need to use another port... @roleoroleo does the FTP Host field support IP:PORT?

laneyweb, 2021-04-29:
Sorry to spam the posts. I have used an admin account (instead of my ftpuser account) and it all works. I must have a permission error at my FTP Server end. Sorry for the messing around.
Thank you Roleo for the great work.

laneyweb, 2021-04-29:
Last post, I promise! :)
Is it possible that there is a limit on the password length? I had used a password of 14 characters, no extended. Now I have a 6 char password, the ftpuser works fine. Just posted in case it helps others.

roleoroleo, 2021-04-29:
I will check it.

soundylinz, 2021-07-21:
I think the problem is, that the created directory has only chmod 644.

If i chance the permissions to 755 at the ftp server, the uploads runs perfectly.

Maybe there is a error in the scripts, which do the upload?

roleoroleo, 2021-07-22:
The script doesn't set permission.
Simply it runs the mkd command to create the directory.
It depends on the user permissions or on the ftp server configuration, if the new directory is 644 or 755.
But you could try to change line 167
```
echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR}"
```
with this
```
echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}\r\nsite chmod 755 ${FTP_DIR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR}"
```
and 174
```
echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}/${FTP_DIR_HOUR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR_HOUR}"
```
with this
```
echo -e "USER ${FTP_USERNAME}\r\nPASS ${FTP_PASSWORD}\r\nmkd ${FTP_DIR}/${FTP_DIR_HOUR}\r\nsite chmod 755 ${FTP_DIR}\r\nquit\r\n" | nc -w 5 ${FTP_HOST} 21 | grep "${FTP_DIR_HOUR}"
```

soundylinz, 2021-07-22:
Sorry ...

That doesn't change anything:

[image]

And: when you restart, the folder with the date 2020 (etc.) is always created. Is that normal?

roleoroleo, 2021-07-23:
Check if the user you are using is able to change permissions.

soundylinz, 2021-07-23:
This is the FTP from my web hosting provider, I don't have access to SSH here.

With the same access data, I can use everything in my FTP software, including CHMOD, etc.

roleoroleo, 2021-08-04:
Could you try to run this command manually?
- log in to you ftp server with a command line client.
- run "mkd" to create a new directory
- check the permissions
- run "site chmod" to change the permissions
- check again the permissions

albertopalotes35, 2021-08-04:
I have a problem with the ftp upload, the upload to the ftp server does not work and if I run the command: /tmp/sd/yi-hack/script/ftppush.sh start by ssh I get this in the registry:

root@porche:~# /tmp/sd/yi-hack/script/ftppush.sh start
2021-08-04 [19-19-45] [INFO] === SERVICE START ===
2021-08-04 [19-19-45] [INFO] checkFiles
2021-08-04 [19-20-30] [INFO] checkFiles
2021-08-04 [19-21-15] [INFO] checkFiles
ash: 080421E2: bad number
2021-08-04 [19-21-15] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y08M04D21H/E220M00S60.mp4] - already sent.
2021-08-04 [19-22-00] [INFO] checkFiles
ash: 080421E2: bad number
2021-08-04 [19-22-00] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y08M04D21H/E220M00S60.mp4] - already sent.
ash: 080421E2: bad number
2021-08-04 [19-22-00] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y08M04D21H/E221M00S60.mp4] - already sent.
2021-08-04 [19-22-46] [INFO] checkFiles
ash: 080421E2: bad number

soundylinz, 2021-08-04:
Sorry, to many Problems...

I've returned the camera.

roleoroleo, 2021-08-05:
> 
> 
> I have a problem with the ftp upload in the new camera, the camera is yi home ifus version the Firmware: 9.0.36.02 and I use the y211ga files version: 0.1.8, the upload to the ftp server does not work and if I run the command: / tmp / sd / yi-hack / script /ftppush.sh start by ssh I get this in the registry:
> 
> root@porche:~# /tmp/sd/yi-hack/script/ftppush.sh start
> 2021-08-04 [19-19-45] [INFO] === SERVICE START ===
> 2021-08-04 [19-19-45] [INFO] checkFiles
> 2021-08-04 [19-20-30] [INFO] checkFiles
> 2021-08-04 [19-21-15] [INFO] checkFiles
> ash: 080421E2: bad number
> 2021-08-04 [19-21-15] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y08M04D21H/E220M00S60.mp4] - already sent.
> 2021-08-04 [19-22-00] [INFO] checkFiles
> ash: 080421E2: bad number
> 2021-08-04 [19-22-00] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y08M04D21H/E220M00S60.mp4] - already sent.
> ash: 080421E2: bad number
> 2021-08-04 [19-22-00] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y08M04D21H/E221M00S60.mp4] - already sent.
> 2021-08-04 [19-22-46] [INFO] checkFiles
> ash: 080421E2: bad number
> 
> If I go into the motion event section, the folder of what he has recorded appears, but no file of what he has recorded appears, however if I connect by ftp to the camera if the recorded files appear. you can see it in the following images:
> https://drive.google.com/file/d/19Geg0jOIoL_93v00QSzTr1TTJI6ei3F7/view?usp=sharing
> https://drive.google.com/file/d/1mavQ6scVDpLs1foJUuoVKucunEw0xAir/view?usp=sharing
> https://drive.google.com/file/d/1bATWcqh63VTzi89SxCu_Yc8T9-yLT53k/view?usp=sharing
> 
> I need to use the ftp upload function for my video surveillance system.

Probably the script must be updated with the new naming convention.

albertopalotes35, 2021-08-05:
By when do you think you can have it corrected?

roleoroleo, 2021-08-05:
I need a few days.

roleoroleo, 2021-08-10:
Try this commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/6e084f08227871573c05f067f0ee45f23507d2f8

albertopalotes35, 2021-08-10:
> Try this commit:
> [6e084f0](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/6e084f08227871573c05f067f0ee45f23507d2f8)

How is it downloaded? If I click on the link, I don't see the download button.

albertopalotes35, 2021-08-11:
> Try this commit:
> [6e084f0](https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/6e084f08227871573c05f067f0ee45f23507d2f8)

I have replaced the files of the micro sd with the new ones and it still does not work, the files are not seen in the web interface and if I run the script by ssh it does not work.
[image]
[image]
[image]

albertopalotes35, 2021-08-11:
I have created the files, I have copied the new content in each file, I have removed the micro sd from the camera and I have replaced the previous files, if I copy the old files again and run the script it works giving the previous error.

roleoroleo, 2021-08-11:
Check the permissions.
Both files must be 755.

albertopalotes35, 2021-08-11:
Both files have permissions 775
[image]
[image]

roleoroleo, 2021-08-16:
I tested it again and I can't find the problem.
Please check if the format of the files is ok. The size of your files doesn't match the right size.
eventsfile.sh is 1001 bytes long.
ftppush.sh is 7566 bytes long.

Are they in windows format?
CR-LF vs LF?

albertopalotes35, 2021-08-16:
I do not know what format it is in, create the file with the Windows notepad and change the extension to .sh, my problem may be that I do not create the file correctly, I have the camera in my country house and even the weekend I can not go to test it, I will try to create the files again with visual studio code.

roleoroleo, 2021-08-17:
I'm pretty sure that your problem is the format.
If you created the files with notepad the files are in windows format (CR+LF at the end of each line) and linux doesn't run it.
Check this post to fix the problem when you use filezilla:
https://forum.filezilla-project.org/viewtopic.php?t=49059

albertopalotes35, 2021-08-21:
I have created the files again in ubuntu and the eventsfile.sh already works, but the ftppush.sh still does not work, it gives the same error.

roleoroleo, 2021-08-23:
Ok. Wait for a new image to install.

albertopalotes35, 2021-09-03:
I have upgraded to version 0.1.9 and I keep getting the same error when running the ftppush.sh script

roleoroleo, 2021-09-03:
Let's start over.
Are you able to see the file list in the "Events" page?

albertopalotes35, 2021-09-03:
Yes, I was able to see the recorded files from the events page.

roleoroleo, 2021-09-03:
Ok.
Try to run the command manually:
```
root@yi-hack-allwinner-v2:~# . /tmp/sd/yi-hack/script/env.sh
root@yi-hack-allwinner-v2:~# /tmp/sd/yi-hack/script/ftppush.sh
```
and check the output.

albertopalotes35, 2021-09-03:
This is what I get running the script.
[image]

roleoroleo, 2021-09-03:
What's the model of your cam?

albertopalotes35, 2021-09-03:
The camera model is a yi home ifus software version 9.0.36.02.202105081647 and the files it has is y211ga.

roleoroleo, 2021-09-03:
Ok, I think that I found the problem.
y211ga has a different naming convention: I implemented it but no tested. So there is a bug.
Change "14" with "44" at line 64 of ftppush.sh:
```
--- a/src/static/static/yi-hack/script/ftppush.sh
+++ b/src/static/static/yi-hack/script/ftppush.sh
@@ -61,7 +61,7 @@ checkFiles ()
        fi
        #
        echo "${L_FILE_LIST}" | while read file; do
-               if [ "${#file}" == "14" ]; then
+               if [ "${#file}" == "44" ]; then
                        FILE_DATE=${file:15:4}-${file:20:2}-${file:23:2}T${file:26:2}:${file:32:2}
                else
                        FILE_DATE=${file:15:4}-${file:20:2}-${file:23:2}T${file:26:2}:${file:30:2}
```

albertopalotes35, 2021-09-03:
I have changed the value 14 to 44 in line 64 and it already works, I have also changed it in another camera yi outdoor ifus software version 9.0.26.07_202104231123 with h30ga files that had the same problem and also works correctly.

mcanteri, 2021-09-09:
Hi Roleo,
I am using ftp to push video file on a server and it works for most of the file but there are some that are left on the SD card and in the log I can see that the cam already tried to push them and marked as already sent. The log is:
2021-09-09 [16-01-50] [INFO] checkFiles
2021-09-09 [16-01-50] [INFO] checkFiles: ignore file [/tmp/sd/record/2020Y04M17D22H/16M33S09.mp4] - already sent.
2021-09-09 [16-01-50] [INFO] checkFiles: ignore file [/tmp/sd/record/2021Y09M09D15H/50M54S05.mp4] - already sent.
The problem is that those files aren't on the server and if I reboot the camera they will be sent correctly.
My conf is:
FTPD=no
FTP_DIR=/mnt/Video/
FTP_DIR_TREE=yes
FTP_FILE_DELETE_AFTER_UPLOAD=yes
FTP_HOST=192.168.X.X
FTP_PASSWORD=***
FTP_UPLOAD=yes
FTP_USERNAME=pi
I use the delete file after upload.
The general problem about those file that remain on the SD of the camera is that there probably were some network problem during the upload and anyway the ftppush.sh marked them as already sent: where the information about file sent is stored ?
Thank you and have a good evening,
Michele

mcanteri, 2021-09-09:
I've added the following code in the else part:
                else
                        logAdd "[INFO] checkFiles: [${file}] - already sent, trying again."
                        if ( ! uploadToFtp -- "${file}" ); then
                                logAdd "[ERROR] checkFiles: uploadToFtp FAILED - [${file}]."
                                continue
                        fi
                        if [ "${FTP_FILE_DELETE_AFTER_UPLOAD}" == "yes" ]; then
                                rm -f "${file}"
                        fi
It seems to work for now and will see in a couple of days if there still are file that remain on the SD card.

roleoroleo, 2021-09-11:
There is a var where the script stores the last file sent (only the last).
If a file is not sent and a newer file is sent, the script will skip it on the next run.
This method is not error-free, I know.

Your code is ok but you should try again only if FTP_FILE_DELETE_AFTER_UPLOAD is "yes".
Otherwise you will send all files at every cycle (and not only newer).

mcanteri, 2021-09-13:
> Your code is ok but you should try again only if FTP_FILE_DELETE_AFTER_UPLOAD is "yes".
> Otherwise you will send all files at every cycle (and not only newer).

You are right: for me it is the same but for who doesn't use the flag "delete after upload", it is.

pannal, 2022-01-24:
Hey,

I'm running an FTPD on a non-default port. For now I've changed the ftppush.sh script with the hardcoded port for `nc` and `ftpput`, but I'd love to see configurable ports in the webinterface!

Apart from that: Would it be possible to support SSL/TLS FTPS?

pannal, 2022-01-24:
Hmm, it also doesn't seem to ever send anything. I've tried running the script manually, this is the output:
```
cat /tmp/ftppush.log

2022-01-24 [17-20-08] [INFO] === SERVICE START ===
2022-01-24 [17-20-09] [INFO] checkFiles
```

`/tmp/last_file_sent` is still default.

roleoroleo, 2022-01-25:
> Apart from that: Would it be possible to support SSL/TLS FTPS?

I can't embed in the hack but you could use openssh from here:
https://github.com/roleoroleo/yi-hack-utils/

closed 2023-01-08 by stale[bot]