roleoroleo/yi-hack-Allwinner-v2 #103: Bricked y21ga / uboot stuck in autoboot

nonexus, 2021-02-22:
Hi all,

I do have two yi cams and unfortunately one did brick itself. 
The hack was running quite smoothly until suddently the cam started to crash randomly with decreasing intervalls from crash to crash.

Then the yellow light stayed on and the cam was bricked. I followed the "unbrick" tutorial and flashed first mtd4. No help... Then I flashed each mtd to be on the safe side. 

The very strange behaviour: 

If I reset the cam from the uboot terminal everything works out fine. After power cycle however the autoboot process of uboot directly jumps to 0 and it crashes. I did increase the bootdelay interval to 10, to see what is going on.

Manual reset via console: Clean countdown -> boot - all okay ( to be fair I messed with the firmware and it is stuck on the actual boot, but that is repairable)
Power Cycle: No count down -> stuck

I will solder some wires to the RX and TX console and look with my oscilloscope if I can get further ideas of what is going on.

Anyone got a idea what is wrong with uboot?

roleoroleo, 2021-02-22:
Maybe a problem with the power adapter or the usb cable?
It has already happened to me.

nonexus, 2021-02-23:
Thanks for your fast reply!
Unfortunately the issue is persistent, even with other usb cable and power supply.
On another matter: Can you tell me the correct output of "printenv" in uboot ? Maybe these are the root cause.

Thanks :)

roleoroleo, 2021-02-23:
Extracted from flash at 0x7c0000:
```
earlyprintk=sunxi-uart,0x05000000
initcall_debug=0
console=ttyS0,115200
nand_root=/dev/nand0p4
mmc_root=/dev/mmcblk0p4
init=/pseudo_init
nor_root=/dev/mtdblock2
rdinit=/rdinit
loglevel=4
cma=0
coherent_pool=16K
mac=
wifi_mac=
bt_mac=
specialstr=
root_partition=rootfs
setargs_nor=setenv bootargs  earlyprintk=${earlyprintk} initcall_debug=${initcall_debug} console=${console} loglevel=${loglevel} root=${nor_root} rootwait init=${init} rdinit=${rdinit} partitions=${partitions} cma=${cma} coherent_pool=${coherent_pool} ion_carveout_list=${reserve_list}
setargs_nand=setenv bootargs earlyprintk=${earlyprintk} initcall_debug=${initcall_debug} console=${console} loglevel=${loglevel} root=${nand_root} rootwait init=${init} rdinit=${rdinit} partitions=${partitions} cma=${cma} mac_addr=${mac} wifi_mac=${wifi_mac} bt_mac=${bt_mac} selinux=${selinux} specialstr=${specialstr} coherent_pool=${coherent_pool} ion_carveout_list=${reserve_list}
setargs_mmc=setenv  bootargs earlyprintk=${earlyprintk} initcall_debug=${initcall_debug} console=${console} loglevel=${loglevel} root=${mmc_root}  rootwait init=${init} rdinit=${rdinit} partitions=${partitions} cma=${cma} mac_addr=${mac} wifi_mac=${wifi_mac} bt_mac=${bt_mac} selinux=${selinux} specialstr=${specialstr} coherent_pool=${coherent_pool} ion_carveout_list=${reserve_list}
boot_normal=sunxi_flash read 45000000 boot;bootm 45000000
boot_recovery=sunxi_flash read 45000000 extend;bootm 45000000
boot_fastboot=fastboot
recovery_key_value_max=0x13
recovery_key_value_min=0x10
fastboot_key_value_max=0x8
fastboot_key_value_min=0x2
bootdelay=1
bootcmd=run setargs_nand boot_normal
verify=N
```

nonexus, 2021-02-27:
Thanks! I managed to get my cam working thanks to your help :)

As you stated the USB Cable and the power source do have a great impact on this. Unfortunately only one single USB-Hub is working right now. I have tested 5 different USB adapters (original yi, Samsung, HP, No-name brand...) and all do not work. 

The cam starts and at the autoboot sequence the CPU seems to crash.

Interesstingly enough, even if I get the cam up and running the voltage of DC-IN drops frequently. 

[image]

I think there is an issue with the PCB design ....

roleoroleo, 2021-02-27:
Maybe some defect in the components of the cam near the power input.

r-science, 2021-03-25:
Same issue here. @nonexus Did you find a solution?

nonexus, 2021-04-02:
@r-science Yes, indeed this was an unlucky combination of bricked firmware and powerissues.

I always tried to boot while checking the UART via TX/RX and hence the back was not present. However, in the back there is the speaker and without the speaker, there seems to be some power issues. After putting the back lid on everything was fine.

r-science, 2021-04-04:
I recovered my camera with different firmware backup from github. But still is working only with 1 or 2 of 20-30 my power adapters... Its some powerissues as you said?

alienatedsec, 2021-04-20:
I just experienced the same with y21ga - fixed when connected to the other power source.

closed 2022-06-11 by roleoroleo