roleoroleo/yi-hack-Allwinner-v2 #106: Yi 1080p Home: IFUSY44*** FW: 9.0.19.02_202007*

cisky78, 2021-02-27:
Hello, it's seems to me that, on my camera (details in the title), this hack doesn't works (i've tried with version 0.1.4 and 0.1.3). Unless i'm doing something wrong following the installatin process.

roleoroleo, 2021-02-27:
The most common problem is an incompatible SD card. Try another one.

cisky78, 2021-02-28:
Thanks for your feedback, i've tried first with a native Kingstone 8GB class 4, then i've tried with a 64GB SD, on wich i've made an unique 4GB FAT32 partition, same behaving for bouth cards. I'm wondering if i'm doying something wrong. Please, confirm that the following steps are correct:
1 - download the "y21ga_0.1.4.tgz" file
2 - copy the following folder/file to the empty SD root:
  Factory
  newbackup
  yi-hack
  lower_half_init.sh
3 - insert the SD into the camera (power disconnected), then power up the camera and wait (should i reset the camera first?)...

thanks a lot!

roleoroleo, 2021-02-28:
Is Factory renamed to Factory.done after the boot?

cisky78, 2021-03-01:
yes, after the boot, i've got the following file/folders in the root:
Factory.done
log
newbackup
record
yi-hack
lower_half_init.sh

roleoroleo, 2021-03-01:
Did you try to connect to the web site on port 80?

cisky78, 2021-03-01:
I'm stupid! sorry! indeed it's working on port 8080... thanks a lot for your support

closed 2021-03-03 by roleoroleo