roleoroleo/yi-hack-Allwinner-v2 #109: Hostname not working for IFUS

phoenix1887, 2021-03-07:
Hostname not working on IFUS camera. I have multiple BFUS cameras that it IS working with and multiple IFUS cameras I verified hostname is NOT working as it should. Anyone else experiencing this?

(I have a simple router: https://www.amazon.com/gp/product/B073TSK26W)

roleoroleo, 2021-03-07:
You are right.
If you want to patch this bug add this line:
```
mount --bind /backup/tools/wifidhcp.sh /home/app/script/wifidhcp.sh
```
in the file /tmp/sd/lower_half_init.sh
before the following lines:
```
./dispatch &
#sleep 2
#./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./rtmp &
#./watch_process &
./dispatch &
#sleep 2
#./rmm &
#sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./rtmp &
#./watch_process &
```

roleoroleo, 2021-03-07:
I will fix it in the next release.

phoenix1887, 2021-03-07:
Thanks for your awesome work🙏🙂

closed 2022-06-11 by roleoroleo