roleoroleo/yi-hack-Allwinner-v2 #110: Real time live stream in Home Assistant

Simone1922, 2021-03-09:
Hi guys,

I'm using a YI 1080p IFUS and everything is working. The rtsp stream I get with VLC is just one second delayed while in Home Assistant I get longer delays (5 seconds or much more). I was wondering if there is a way to avoid this delay and see the "live" stream from the camera with a shorter delay? I used a smartphone with "IP webcam app" and the stream with "http" was almost in real time so I'd like to have the same with the Yi camera (if possibile).

Thanks in advance for any suggestion.

wild-run, 2021-05-14:
same problem here

roleoroleo, 2021-05-14:
Please try webrtc plugin.
https://github.com/AlexxIT/WebRTC/

wild-run, 2021-05-15:
First congratulate you for your great work, this addon that you comment works well to see the video, but now the problem is when I create an entity to detect movement and others, I have used this addon https://github.com/roleoroleo/yi-hack_ha_integration and I have also tried to add it manually but it does not work at all well

roleoroleo, 2021-05-16:
Did you enable mqtt in ha?
It's required.

closed 2023-01-08 by stale[bot]

Gifford47, 2023-06-14:
> Hi guys,
>
> I'm using a YI 1080p IFUS and everything is working. The rtsp stream I get with VLC is just one second delayed while in Home Assistant I get longer delays (5 seconds or much more). I was [...]

Does anybody have an ideo how to stream realtime? I use shinobi and the stream lag is about 7 seconds ...