roleoroleo/yi-hack-Allwinner-v2 #113: FTP Upload / Record to SD not working

colethegamer123, 2021-03-14:
Hello my friend,

I have some issues with the Allwinner v2 hack.

I bought a new Yi Hack 1080p AI+ Camera. 
[image]

The issue is I cannot record to SD card it does not save anything when the motion is detected, anyhow with MQTT the motion sensor is working.
The FTP record also not working, but I think its only because it does not find anything on the SD Card to upload.
[image]

It was freshly installed, I already reseted once. Its a new device so there should be no hardware problem as I see.

[image]

I also checked the logs of the FTP and there I cannot see anything what would try to connect from this IP Address. So its not an FTP issue, thats sure, I have YiHack Allwiner 1 version Cameras too, same type, they upload properly.

Best regards,
Thanks for your time and keep up the good work mate :)

roleoroleo, 2021-03-14:
Is recording enabled in your yi app?
If cloud is disabled, is "Recording without Cloud" enabled?

colethegamer123, 2021-03-14:
Internet Connection is disabled on the router for these Cameras.

If cloud is disabled, is "Recording without Cloud" enabled?
 - No its disabled both settings.
 
 But I have 3 more Camera where its the same like that with AllWinner 1 and they are working correctly. 
 
 I'll try to enable that like this:
 
[image]

Lets see if its working like that. Thanks for your really fast proactive answer!

colethegamer123, 2021-03-14:
Its working like that, I understand the meaning if these settings now, sorry for my user error!
```
Sun Mar 14 17:32:45 2021 [pid 2] CONNECT: Client "192.168.0.63"
Sun Mar 14 17:32:45 2021 [pid 1] [colethegamer] OK LOGIN: Client "192.168.0.63"
Sun Mar 14 17:32:45 2021 [pid 3] [colethegamer] OK UPLOAD: Client "192.168.0.63", "/share/16M40S00.mp4", 60337 bytes, 
```

Thank you, I'll close the issue after a few tests! Keep up the good work and Thanks for the really fast answers :)

colethegamer123, 2021-03-14:
One more little thing, this is just like a question, I dont want to open another issue about this because this is not connected to Yihack as I see, but you are the only one who maybe knows the answer.

Did you ever tried to use a Yi Home camera with a non-original Yi Micro USB Cable?
 - The 3.5meter long original cable is way too long for me, I have 20 cm Micro USB cables and I want to use those ones.
 
After I connected with this the Camera Turn on, but stops booting, It cannot be reached via network, and the LED is solid orange. I never seen anybody was mentioned this problem, also the provider does not want to help me. There is any way to somehow disable the Camera's cable certificate check or something like that? Looks like its only working with original cables, what is a mess, you cannot buy them separeatly. 

Thank you.

roleoroleo, 2021-03-14:
Yes, I used the cam with other cables.
It's possible but it depends on the cable.
Many people report the same problem you posted.

colethegamer123, 2021-03-14:
Interesting I tried already 7 cable what does not work, all type is different brand, Anyhow I'll look for more short ones. Thanks for your help. I close the issue.

closed 2021-03-14 by colethegamer123