roleoroleo/yi-hack-Allwinner-v2 #116: ONVIF not working anymore?

ebaschiera, 2021-03-27:
Hi,
upgrading from 0.1.3 to 0.1.5 made ONVIF stop working on may 2 cameras (y21ga)...
My Home Assistant cannot find the cameras anymore using the ONVIF integration, and also an Android app called Onvifer I downloaded for testing, cannot find any cameras.
So I made a reset, reconfigured from scratch, but I still have that problem.
Can you help me understanding what's wrong?
My configuration is as follows. 
Thanks!
[image]

roleoroleo, 2021-03-27:
Did you select the correct network interface?

Murada99, 2021-03-28:
Same problem, I upgraded from 0.1.3 to 0.1.5 and onvif stopped working.
Changed to previous version 0.1.4 and it works now.

roleoroleo, 2021-03-28:
What's your model?

ebaschiera, 2021-03-28:
> Did you select the correct network interface?

My cameras have wifi only, so I presume yes.

roleoroleo, 2021-03-28:
Check the command line of the onvif command.
Parameter --ifs

Murada99, 2021-03-28:
> What's your model?

model: y21ga

Watanakas, 2021-03-28:
I have the same problem as partner Ermanno Baschiera, with previous version it works fine for me,
thanks for your job.

roleoroleo, 2021-03-28:
What's exactly the problem?
Automatic discovery or onvif?
If you run Onvif Device Manager, does the cam appear automatically in the list?
If not, if you enter the cam manually with the ip address, does it works?

roleoroleo, 2021-03-28:
With my y21ga, onvif works properly.

mcanteri, 2021-03-28:
I have problems too recording video clip from the ffmpeg stream.
My cameras are simply configured as:

`camera:
  - platform: ffmpeg
    input: rtsp://192.168.1.132/ch0_1.h264
    name: living_room_camera
    #extra_arguments: -rtsp_transport tcp -analyzeduration 15
  - platform: ffmpeg
    input: rtsp://192.168.1.94/ch0_1.h264
    name: outdoor_camera
    #extra_arguments: -rtsp_transport tcp -analyzeduration 15
`
In the log file I can see a lot of these errors:
`2021-03-28 20:58:20 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 0.2 seconds
2021-03-28 21:00:04 DEBUG (SyncWorker_2) [homeassistant.components.stream.recorder] Starting recorder worker thread
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8754).
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8461).
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8788).
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 9050).
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8908).
2021-03-28 21:00:04 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8760).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8997).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8934).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8988).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 9087).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 9071).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8946).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8809).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8824).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8786).
2021-03-28 21:00:05 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
2021-03-28 21:00:06 ERROR (recorder_save_worker) [libav.h264] Invalid NAL unit size (0 > 8810).
2021-03-28 21:00:06 ERROR (recorder_save_worker) [libav.h264] Error splitting the input into NAL units.
`
The strange thing is that if I play with mplayer or vlc the video is perfect so probably is something related to the code implemented in Home Assistant.

roleoroleo, 2021-03-28:
I'm testing it with vlc and ffmpeg and I can't see any error.
Tomorrow, I will try to re-install from scratch.

closed 2022-06-11 by roleoroleo