roleoroleo/yi-hack-Allwinner-v2 #118: Home Assistant live streaming no longer working with new rRTSPServer v0.1.5

stickpin, 2021-03-28:
Hi @roleoroleo,

I have an issue with the new rRTSPServer that comes with release v0.1.5 on my y21ga camera.
Home Assistant live streaming no longer working with the Home Assistant app, Google Chrome, Brave but DO works with Firefox Browser.
Enough just to replace rRTSPServer from v0.1.4 and everything works correctly.

Please, let me know how can I help to resolve this bug.

Thanks a lot for your hard work!

mcanteri, 2021-03-28:
I have a problem too with RTSP in HA:

https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/116#issuecomment-808949830

stickpin, 2021-03-28:
@mcanteri ONVIF itself works fine for me.

mcanteri, 2021-03-28:
> @mcanteri ONVIF itself works fine for me.

I don't use ONVIF, what is the relevant benefit of use it since down under it uses RTSP anyway ?

mcanteri, 2021-03-28:
> I am also not using ONVIF with HA but an FFmpeg camera integration.

Me too, as you can read my configuration in the other thread, some comment above:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/118#issuecomment-808950616

Simple ffmpeg integration that let you configure and fine tune with all the option available :+1:

roleoroleo, 2021-03-29:
There are a many NVRs where you can't set rtsp input but only onvif.
Getting back to the problem, I can't reproduce it as described here: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/116#issuecomment-808957611

roleoroleo, 2021-03-29:
I move forward by attempts.
With this beta release you can remove timing info (added in the last versione) from stream.
Please try if it works:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6220074/rRTSPServer.gz)

stickpin, 2021-03-29:
Thanks @roleoroleo!
Just have tried, same result, playing in Firefox, not playing in Chrome/Brave/etc... 

Do I need to use some flags to run it?
Currently i am testing it by just dropping new beta into the bin folder and running ~/bin# ./rRTSPServer_beta1

roleoroleo, 2021-03-29:
No, default is without timing info.

roleoroleo, 2021-03-29:
Could try this binary?
I fixed a little bug.
[Uploading rRTSPServer.gz…]()

stickpin, 2021-03-29:
@roleoroleo same issue. :(

roleoroleo, 2021-03-29:
Please try again adding -s option.
And "-d 3" option to increase the log.

stickpin, 2021-03-29:
I've attached two logs. 
Same code base v0.1.5 different rRTSPServer servers (latest beta that you sent beta_2 and from v0.1.4 release).
[beta_2.log.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6223595/beta_2.log.zip)
[0.1.4.log.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6223593/0.1.4.log.zip)

roleoroleo, 2021-03-29:
The logs are ok.
I don't understand.

stickpin, 2021-03-29:
let me try to find ffmpeg options that HA running with so you will be able to reproduce it yourself.

nmenoux, 2021-03-29:
@roleoroleo (for info) I've just restarted manually the original rRTSPServer on the second cam (AllWinner-v2) and I got a segmentation fault as below : 

```
1617046444098: frame detected - frame_len: 8 - frame_counter: 58345 - frame_counter_last_valid: 58345 - resolution: 1080
1617046444098: frame_len: 8 - cb_current->size: 262144
1617046444098: frame_len: 8 - frame_counter: 58345 - resolution: 1080
1617046444098: frame_write_index: 31 - cb_current->output_frame_size 42
1617046444098: frame detected - frame_len: 16408 - frame_counter: 58346 - frame_counter_last_valid: 58346 - resolution: 1080
RTSP doGetNextFrame() whole frame - fFrameSize 855 - counter 58343 - fMaxSize 262140 - fLimitNumBytesToStream 0
RTSP doGetNextFrame() - whole frame completed
RTSP doGetNextFrame() start - fFrameSize 262140 - fMaxSize 262140 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
RTSP doGetNextFrame() read_index == write_index
1617046444124: frame_len: 16408 - cb_current->size: 262144
1617046444124: frame_len: 16408 - frame_counter: 58346 - resolution: 1080
1617046444124: frame_write_index: 32 - cb_current->output_frame_size 42
1617046444124: frame detected - frame_len: 812 - frame_counter: 58344 - frame_counter_last_valid: 58344 - resolution: 360
1617046444124: frame_len: 812 - cb_current->size: 49152
1617046444124: frame_len: 812 - frame_counter: 58344 - resolution: 360
1617046444124: frame_write_index: 29 - cb_current->output_frame_size 42
1617046444124: SPS   detected - frame_len: 19 - frame_counter: 58345 - frame_counter_last_valid: 58345 - resolution: 360
1617046444124: frame_len: 19 - cb_current->size: 49152
1617046444124: frame_len: 27 - frame_counter: 58345 - resolution: 360
1617046444124: frame_write_index: 30 - cb_current->output_frame_size 42
1617046444124: frame detected - frame_len: 8 - frame_counter: 58346 - frame_counter_last_valid: 58346 - resolution: 360
RTSP doGetNextFrame() whole frame - fFrameSize 808 - counter 58344 - fMaxSize 262140 - fLimitNumBytesToStream 0
RTSP doGetNextFrame() - whole frame completed
RTSP doGetNextFrame() start - fFrameSize 262140 - fMaxSize 262140 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
RTSP doGetNextFrame() whole frame - fFrameSize 23 - counter 58345 - fMaxSize 262140 - fLimitNumBytesToStream 0
Segmentation fault (core dumped)
```

nmenoux, 2021-03-29:
I will try to use your AllWinner-v2 patched rRTSPServer  in debug (3) mode. I keep you informed

stickpin, 2021-03-29:
Ok I've found the way to reproduce it with FFmpeg.

Download this Chrome extension, so you will be able to play HLS natively in Chrome/Brave Browser: https://chrome.google.com/webstore/detail/play-hls-m3u8/ckblfoghkjhaclegefojbgllenffajdc

Run Following FFmpeg command:
`./ffmpeg -rtsp_transport tcp -i rtsp://username:password@192.168.1.1:554/ch0_0.h264 -vcodec copy -acodec aac -movflags +faststart -master_pl_name master_playlist.m3u8 -hls_segment_type fmp4 -hls_list_size 2 -hls_init_time 1 -hls_time 1 -hls_flags delete_segments -hls_segment_filename "segments/%d.m4s" "segments/playlist.m3u8"`

Run some kind of HTTP server, I am using Mac and running the following command `python -m SimpleHTTPServer 8080` from the folder where FFmpeg creating segments.

After that try to play master_playlist.m3u8 in Chrome/Brave.
`http://localhost:8080/segments/master_playlist.m3u8`

You will see that browser will retrieve the first 2 chunks and after that will continue retrieving only `playlist.m3u8` but not chunks.
Once you will return to rRTSPServer from v0.1.4 everything will run smoothly.

I hope it helps.

stickpin, 2021-03-30:
@roleoroleo yes! works as expected now!
Thanks a lot!

roleoroleo, 2021-03-30:
It was an option of the live555 library.
Probably the Chrome extension is not so "standard".

stickpin, 2021-03-30:
@roleoroleo I don't think it's related to the extension itself, Chrome, Safari, and HA iOS app was not able to play. Maybe something related to AppleWebKit HLS implementation. Firefox not using it.

Latest MStar version using the same live555 version as Allwinner-v2? Because my 4 MStar cameras are running fine.

roleoroleo, 2021-03-30:
Is the same version of the lib but MStar doesn't use FramedSource and, in this case, the option is not available.
MStar will work without this problem.

mcanteri, 2021-03-31:
> Please try this binary:
> [rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6229017/rRTSPServer.gz)

Tested and is working for me too, thank you @roleoroleo !

nmenoux, 2021-04-01:
@roleoroleo the new rRTSPServer works perfectly on my cam too 👍  Thanks !

DaniJulia, 2021-04-14:
Home Assistant live streaming again no longer working with new update v0.1.6
(This issue was fixed before, in 0.1.5, with the binary you attached)

stickpin, 2021-04-14:
@DaniJulia working fine for me with v0.1.6.

roleoroleo, 2021-04-14:
The fix is correctly committed and compiled.

DaniJulia, 2021-04-15:
Only out of curiosity: After "on-line" upgrade, indeed, all cameras stop sending video to Home Assistant.
However, after a "clean" install of the new firmware, everything is Ok.
Thank you SO MUCH for you work!!

DaniJulia, 2021-04-15:
BTW, @roleoroleo, please forgive me because I know that this is not the place where to ask this... if you can, please delete this message as soon as you read it (I will anyway).
I want to ask you if you have any kind of profile place, with any kind of patreon, paypal, or similar, to accept contributions? I really would like to buy you a beer from time to time.

roleoroleo, 2021-04-15:
The upgrade procedure is a complex task. Sometimes maybe something goes wrong.

There is a paypal link in the main page of this repo.
A beer is greatly appreciate .

closed 2021-05-16 by roleoroleo