roleoroleo/yi-hack-Allwinner-v2 #119: dont'hacking

alteus83, 2021-03-28:
Hello,

my camera has the following version/firmware:  IFUSY44UWSIIJN..../ 9.0.19.02_202007141512

but with the version nothing happens once you put the SD(thi version use y21ga_0.1.5.tar) and wait for the reboot.
I try to connect to the IP but nothing.

Help me 
:(

roleoroleo, 2021-03-29:
Try another sd card.

alteus83, 2021-03-29:
which one do you recommend?

roleoroleo, 2021-03-29:
Eh sì...

I can't recommend a card.
Some work, some don't work.

alteus83, 2021-03-29:
Ottimo, andrò a tentativi ma almeno comincio a eliminare qualcosa

alteus83, 2021-03-30:
come non detto, ho provato con una SD kingstone che avevo a casa da 4g e oggi ho comprato una verbatim da 16 ma nulla......non parte il cambio del firmware.

sono sfortunato :( chissà che versione sarà la mia

roleoroleo, 2021-03-30:
Based on the fw version your cam is compatible with this hack.

alteus83, 2021-03-30:
Però non parte l’upgrade,

Domanda, una volta messo il file unzippato, il y21ga_0.1.5.tar e messa la
SD con il file nella cam da accesa non devo fare altro che andare sull’ip
da guida? (Ovviamente aspettando che torna blu la spia)

roleoroleo, 2021-03-30:
Did you decompress gz or tar.gz?
You have to copy in your sd card Factory folder, newhome folder, etc..

roleoroleo, 2021-03-31:
Decompress tar also.
The sd card should appear:
[image]

alteus83, 2021-03-31:
Ti ringrazio per la disponibilità, sta sera provo e ti faccio sapere.

alteus83, 2021-03-31:
Ho risolto, ho cambiato ancora SD e alla fine è andata, hai per caso una config esempio da consigliarmi? da inserire in home assistant per creare uan scheda?

Grazie Mille, se hai un link donazioni ti offro una birra :) Gentilissimo!

roleoroleo, 2021-04-01:
Thank you for you donation.
For home assistant, take a look here https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Home-Assistant-integration
or here https://github.com/EnricoFerro/yi-hack-to-mqtt

a20691, 2021-04-03:
Hi. 

I'm trying to hack a camera IFUSY65XV4, but with the firmware 9.0.36*. Can any of you provide me the 9.0.19* firmware, i will try downgrade my camera. @roleoroleo do you think the downgrade could be a solution or a new hack have to be made? With this hack the camera is always restarting.

roleoroleo, 2021-04-03:
Probably it's not a fw upgrade but a different model.
Before trying to hack it, make a backup copy (follow the wiki).

emnatsak, 2021-04-10:
Hello!
same problem 😒 
camera SN Y211... IFUSY65... FW : 9.0.36.xx 
With hack y21ga_0.1.5.tar the camera is always restarting. I made a backup - I can send it to you if necessary

roleoroleo, 2021-04-11:
fw 9.0.36 means that this cam is not supported, send me a backup.

emnatsak, 2021-04-11:
Hello!
shoud I post backup file here?
https://www.dropbox.com/sh/2oh3jsr1918qqlh/AAADf_yO58V50g9v2lIGOPDaa?dl=0

FW version 9.0.36.00_202011101611
also there is new FW 9.0.36.01_202103231716  I have not updated yet

roleoroleo, 2021-04-11:
Your model is y211ga.
Never seen.

roleoroleo, 2021-04-11:
Please send me 2 or 3 copies of the file /dev/shm/fshare_frame_buf
Make a copy, wait 5 seconds and make another copy.

a20691, 2021-04-11:
> Hello!
> shoud I post backup file here?
> https://www.dropbox.com/sh/2oh3jsr1918qqlh/AAADf_yO58V50g9v2lIGOPDaa?dl=0
> 
> FW version 9.0.36.00_202011101611
> also there is new FW 9.0.36.01_202103231716 I have not updated yet

Be careful don't upgrade to this new firmware. My brick camera after that. The good thing was that I activated the warranty and they sent a new one with firmware 9.0.19 and is already with the hack working. Now I'll try unbrick the one with 9.0.36

roleoroleo, 2021-04-11:
9.0.19 and 9.0.36 are fw for DIFFERENT models.
Don't mix them.

a20691, 2021-04-11:
> Your model is y211ga.
> Never seen.
> You can remove the link if you want.

[image]

roleoroleo, 2021-04-11:
The label on the pcb is not important.
Check the content of the file system.

emnatsak, 2021-04-11:
> Please send me 2 or 3 copies of the file /dev/shm/fshare_frame_buf
> Make a copy, wait 5 seconds and make another copy.
I do not know how to do it ;(

roleoroleo, 2021-04-11:
If you tried to hack this cam with y21ga hack, you should have ssh or telnet access.
Use ssh to login and make the copies.

a20691, 2021-04-11:
> If you tried to hack this cam with y21ga hack, you should have ssh or telnet access.
> Use ssh to login and make the copies.

I can't try because the camera is bricked, but I don't believe because the camera is always restarting, at least it happens to me before it brick.

emnatsak, 2021-04-11:
> If you tried to hack this cam with y21ga hack, you should have ssh or telnet access.
> Use ssh to login and make the copies.

thanx. got it via telnet. user root and no pass

first attachment 3 files copy 
[cp-fshare_frame_buf-1-2-3.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6292858/cp-fshare_frame_buf-1-2-3.zip)
second another way - 3 files cat

[cat-fshare_frame_buf-1-2-3.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6292861/cat-fshare_frame_buf-1-2-3.zip)

roleoroleo, 2021-04-13:
Try the last release.

emnatsak, 2021-04-13:
> Try the last release.

Hi!
not working :(
extracting the contents of y211ga_0.1.6.tgz to the root of SD card (tried with 2 SD)
 camera boots - yellow light for 3-5 sec then "click" then blinking blue light for 5-7 sec and camera goes to restart
 and again and again...
 removing SD card - camera boots normally and works as usually
 
 any ideas?

roleoroleo, 2021-04-13:
Probably the previous hack breaks something.

emnatsak, 2021-04-13:
What should I do? 
reset?
upgrade to new FW 9.0.36.01_202103231716 (now is FW version 9.0.36.00_202011101611)

roleoroleo, 2021-04-13:
You should restore the original files.

emnatsak, 2021-04-13:
> You should restore the original files.

pls tell me right way to restore backup.

roleoroleo, 2021-04-14:
Restore at least /backup/init.sh
Get it from here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/sdhack/y211ga/newbackup/init.sh

emnatsak, 2021-04-14:
something does not work
I took the init.sh file - it is identical to the same file in the y211ga_0.1.6.tgz hack
connected by telenet and cp it in the camera.
then I also checked the wifidhcp.sh file loaded into the camera, it is also identical to the same file in the y211ga_0.1.6.tgz hack
but the problem remains - the hack doesn't apply

maybe I had to write the original files and reapply the hack?
where can I get these files?
Or is it better to do a full restore from a backup - is there a step-by-step guide somewhere?

roleoroleo, 2021-04-14:
Do you have the serial connection connected?

emnatsak, 2021-04-14:
> Do you have the serial connection connected?

no/

BTW - I have second identical camera unpacked - with original software

roleoroleo, 2021-04-14:
Make a backup copy (see the wiki).

emnatsak, 2021-04-14:
> Make a backup copy (see the wiki).

done 
backup/mtd.txt
backup/mtdblock0.bin
backup/mtdblock1.bin
backup/mtdblock2.bin
backup/mtdblock3.bin
backup/mtdblock4.bin
backup/mtdblock5.bin
backup/mtdblock6.bin
backup/mtdblock7.bin

can I use "Remove the hack"  - unhack_y21ga.tar.gz?

roleoroleo, 2021-04-14:
Check the updated wiki.

emnatsak, 2021-04-14:
> Check the updated wiki.
done unhack " y21ga model: unhack_y21ga.tar.gz "
hack removed - no telnet access - connection refused
then I tried to hack with y211ga_0.1.6.tgz
and... sorry, with same result - 
camera boots - yellow light for 3-5 sec then "click" then blinking blue light for 5-7 sec and camera goes to restart

what I'm doing wrong?

roleoroleo, 2021-04-14:
You should use unhack_y211ga.tar.gz
But it doesn't matter, it doesn't change.

Sorry but it's non possible to make a diagnosis without a log.
Try to edit etc/system.conf and to disable all services:
```
HTTPD=yes
TELNETD=yes
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=no
RTSP_STREAM=both
RTSP_AUDIO=no
SPEAKER_AUDIO=no
ONVIF=no
ONVIF_WSDD=no
ONVIF_PROFILE=both
ONVIF_WM_SNAPSHOT=no
NTPD=no
NTP_SERVER=ntp1.inrim.it
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
ONVIF_NETIF=wlan0
```

roleoroleo, 2021-04-14:
Ah...
Another question.
How to you copy the file in the sd card?
Which software are you using?

emnatsak, 2021-04-14:
tell - and I send you all necessary files/logs

and - does anyone else hacked the same camera with success result?
and - as I understand full restore from original backup (files - mtd.txt, mtdblock0.bin..mtdblock7.bin) will require disassembling the camera and access via the serial port - the issue cannot be solved programmatically - write it as a new firmware

roleoroleo, 2021-04-14:
Check if the scripts files in yi-hack/script have the right format.
LF vs CR-LF at the end of each line.

roleoroleo, 2021-04-14:
Try to use 7zip.

emnatsak, 2021-04-15:
> Try to use 7zip.

Hello!
sorry - no result
tried many different variations - ways to hack - 
clean formated SD (2pcs) - unhack done (no telnet) - reset camera - hacking y211ga_0.1.6 (unpacked by 7zip) - no result
ok
did full reset
updated FW to latest  FW 9.0.36.01 (before made one more backup)
hacking y211ga_0.1.6  - no result  - 
camera boots - yellow light for 3-5 sec then "click" then blinking blue light for 5-7 sec and camera goes to restart
without SD with hack - camera boot normally.

Does anybody else with same camera did it (hack) with success result? sorry, maybe hack has a little mistake? 
I can send you all files you need

emnatsak, 2021-04-15:
> Try to edit etc/system.conf and to disable all services:
> 
tried - with the same NO result

roleoroleo, 2021-04-15:
Another SD card?

emnatsak, 2021-04-15:
> Another SD card?

tried 2 SD cards 16Gb and 32Gb

as I think - hack install script began to work (new folders and files new files appear on the memory card) but at the moment the something goes wrong

elidev12, 2021-04-15:
> Hello!
>
> sorry - no result
>
> tried many different variations - ways to hack -
>
> clean formated SD (2pcs) - unhack done (no telnet) - reset camera - hacking y211ga_0.1.6 (unpacked by 7zip) - no result [...]

Got the same results for my camera (FW 9.0.36.01). Camera constantly restarts even when disabling the services in etc/system.conf

emnatsak, 2021-04-19:
> Another SD card?

> Got the same results for my camera (FW 9.0.36.01). Camera constantly restarts even when disabling the services in etc/system.conf

any news?

roleoroleo, 2021-04-20:
To solve the problem I need a serial connection.
Are you able to make it?

emnatsak, 2021-04-20:
I'll try - pls tell what to do step by step

roleoroleo, 2021-04-20:
See the wiki about the unbrick.

a20691, 2021-04-24:
> To solve the problem I need a serial connection.
> Are you able to make it?

Yes
I unbrick my carema rigth now.
It's working and now i need the hack again, what can i do?

roleoroleo, 2021-04-25:
Install the hacking following the standard procedure and capture the serial log.
For example using putty with logging enabled.
Then send me the log.

a20691, 2021-04-25:
> Install the hacking following the standard procedure and capture the serial log.
> For example using putty with logging enabled.
> Then send me the log.

https://mega.nz/#!liB3GYZT!c6GzOzwR8xJRk5y3qkuryZY26MakxIymhRE7CRhourE

roleoroleo, 2021-04-25:
Check this post:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/143

closed 2023-01-08 by stale[bot]