roleoroleo/yi-hack-Allwinner-v2 #121: MQTT: Socket error on client <unknown>, disconnecting.

mcanteri, 2021-04-02:
Hi @roleoroleo,

after one of the last release I discovered a lot of messages like that in the messages log on the Rpi where Home Assistant run.
The cameras still work sending motion events but the log is flood by those messages every 5 seconds.
I haven't changed anything in HA as for the configuration and neither on the cameras, so I don't know which part is the cause of the problem.
I'm attaching an archive of what I find in the log.
[mqtt-errors.log.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6249053/mqtt-errors.log.gz)

roleoroleo, 2021-04-02:
Is mqtt working properly?
Do you receive updates from the cam?
If yes, the problem is the new mqtt-config service added (at the moment it's not working properly).
To avoid log flooding, simply leave empty the box "Topics Prefix for remote configuration".

mcanteri, 2021-04-02:
Yes, the two camera are working correctly as I said, only the log is flooded with those messages.

That stopped the errors, thank you!

roleoroleo, 2021-04-13:
Try the last release.

closed 2021-05-16 by roleoroleo