roleoroleo/yi-hack-Allwinner-v2 #124: New Cam Yi 9.0.07.11 ? 1296P PTZ, test with y21ga, r30gb,h52ga, onvif detected but no image, no vidéo ...

gjai, 2021-04-02:
Hi,

I've a New Cam Yi 9.0.07.11 ? 1296P PTZ, test with y21ga, r30gb,h52ga, onvif detected but no image, no vidéo ... do you have an idea ?

( Je suis français, je fais ce que je peut avec l'anglais xD )

roleoroleo, 2021-04-03:
If the model is new, we need to update this hack.
Please, make a backup copy and send it to me.
Read the wiki about backup.
Don't try to apply the hack.

gjai, 2021-04-03:
This is my backup https://www43.zippyshare.com/v/pQ9VDxJb/file.html

roleoroleo, 2021-04-03:
The suffix of your cam is q321br_lsx.
It's a new model but it's similar to y21ga.
I need more details and a picture.
- resolution (high and low)?
- home or dome?
- network (lan, wifi, or both)?
- ...

gjai, 2021-04-03:
[image]

gjai, 2021-04-03:
Wireless only. 
Résolution HD 1296p , SD unknown,...
But i've Root Access with your hack, if you want another information ? Command Line ? Log ? ...

gjai, 2021-04-03:
[image]

roleoroleo, 2021-04-03:
I need 2 or 3 copies of the file /dev/shm/fshare_frame_buf
This file contains both streams, low and high res.
I need it to update h264grabber with new parameters.

gjai, 2021-04-03:
https://we.tl/t-Inyc95Cj17

gjai, 2021-04-03:
if it might help ... 
[image]

roleoroleo, 2021-04-03:
Are you sure the file is only 524588 bytes long?
Normally it's about 1,7 MB.

gjai, 2021-04-03:
[image]
 :(

roleoroleo, 2021-04-06:
Do you want to try this beta?
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6264124/rRTSPServer.gz)

gjai, 2021-04-06:
i'v copy new rRTSPServer and try to launch

> root@ENFANTS:~/bin# ./rRTSPServer -m y21ga -r high -p 554 -d 3
debug on, level 3
1617708057729: mapping file /dev/shm/fshare_frame_buf, size 1786224, to b6bb1000
1617708057732: closing the file /dev/shm/fshare_frame_buf
1617708057733: starting capture main loop
1617708057760: frame detected - frame_len: 71008337 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708057785: frame detected - frame_len: 109306087 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708057886: frame detected - frame_len: 110601471 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708057911: frame detected - frame_len: 70381817 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708057961: frame detected - frame_len: 50706440 - frame_counter: 4096 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708057961: frame detected - frame_len: 73179055 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058037: frame detected - frame_len: 122899034 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708058087: frame detected - frame_len: 125291567 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708058137: frame detected - frame_len: 55748198 - frame_counter: 4096 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058163: frame detected - frame_len: 76298494 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058238: frame detected - frame_len: 75285409 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058238: frame detected - frame_len: 124727007 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708058314: frame detected - frame_len: 76464160 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058340: frame detected - frame_len: 57863261 - frame_counter: 4096 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058365: frame detected - frame_len: 83622079 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058366: frame detected - frame_len: 714258546 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708058442: frame detected - frame_len: 85557306 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058442: frame detected - frame_len: 118119614 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708058543: frame detected - frame_len: 133464115 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
1617708058568: frame detected - frame_len: 55587948 - frame_counter: 4096 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058594: frame detected - frame_len: 84364834 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708058619: frame detected - frame_len: 129732662 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
[... cut ...]
1617708059703: frame detected - frame_len: 61913212 - frame_counter: 2048 - fram                                                                                                             e_counter_last_valid: -1 - resolution: 0
1617708059703: frame detected - frame_len: 137907929 - frame_counter: 1024 - fra                                                                                                             me_counter_last_valid: -1 - resolution: 0
Failed to create RTSP server: unable to create stream socket: Address family not                                                                                                              supported by protocol

>

gjai, 2021-04-06:
with argument y21ga, r30gb, h52ga  i've "Bus error (core dumped)"
with q321br ... audio stream it's ok, but no vidéo

roleoroleo, 2021-04-06:
The correct model is q321br_lsx
Please run it with the following parameters:
```
rRTSPServer -m q321br_lsx -r high -a no -d 3
rRTSPServer -m q321br_lsx -r low -a no -d 3
```
and post the output.

GizmoLibre, 2021-04-06:
Try to Install firmware version 0.14, when i upgraded to 0.15 on my r30gb Yi Dome Guard, RTSP would not work at all. High res on 0.14 is still unstable, but at least low res is working perfectly. Onvif on 0.14 gives no image also, just not to add confusion.

roleoroleo, 2021-04-06:
@GizmoLibre 
Could you help me to make it more stable?

GizmoLibre, 2021-04-06:
@roleoroleo
Tried it, replaced the rRTSPServer in bin folder on 0.14 and 0.15 too. On 0.14 it is the same, high res crashes or freezes in VLC, low res works. On 0.15 i still can't get rtsp to load at all both on low, or high. So, i guess the issue isn't in rRTSPServer, it is some other part of firmware for r30gb that is conflicting with hardware of - in my case Yi Dome Guard. 

Still, dunno nothing about this stuff. In other thread you suggested "Run h264grabber instead of rRTSPServer and redirect to a file, then try to read the h265 stream with another app." i have no idea how to do that, i can try if you are willing to explain it step by step.

gjai, 2021-04-06:
with the first link 

> root@ENFANTS:~/bin# ./rRTSPServer -m q321br_lsx -r high -a no -d 3
debug on, level 3
1617721860457: mapping file /dev/shm/fshare_frame_buf, size 524588, to b6d1a000
1617721860458: closing the file /dev/shm/fshare_frame_buf
1617721860458: starting capture main loop
1617721860509: frame detected - frame_len: 604 - frame_counter: 3433 - frame_counter_last_valid: -1 - resolution: 0
1617721860509: incorrect frame counter - frame_counter: 64566 - frame_counter_last_valid: -1
1617721860509: frame detected - frame_len: 953 - frame_counter: 64566 - frame_counter_last_valid: -1 - resolution: 360
1617721860510: incorrect frame counter - frame_counter: 65170 - frame_counter_last_valid: -1
1617721860510: frame detected - frame_len: 1453 - frame_counter: 65170 - frame_counter_last_valid: -1 - resolution: 1080
1617721860510: SPS   detected - frame_len: 19 - frame_counter: 3434 - frame_counter_last_valid: -1 - resolution: 0
1617721860510: frame detected - frame_len: 8 - frame_counter: 3435 - frame_counter_last_valid: -1 - resolution: 0
1617721860510: frame detected - frame_len: 16804 - frame_counter: 3436 - frame_counter_last_valid: -1 - resolution: 0
1617721860510: incorrect frame counter - frame_counter: 64567 - frame_counter_last_valid: -1
1617721860510: frame detected - frame_len: 1054 - frame_counter: 64567 - frame_counter_last_valid: -1 - resolution: 360
1617721860510: incorrect frame counter - frame_counter: 65171 - frame_counter_last_valid: -1
1617721860510: frame detected - frame_len: 1473 - frame_counter: 65171 - frame_counter_last_valid: -1 - resolution: 1080
1617721860535: incorrect frame counter - frame_counter: 64568 - frame_counter_last_valid: -1
1617721860535: frame detected - frame_len: 1059 - frame_counter: 64568 - frame_counter_last_valid: -1 - resolution: 360
1617721860535: incorrect frame counter - frame_counter: 65172 - frame_counter_last_valid: -1
1617721860535: frame detected - frame_len: 1424 - frame_counter: 65172 - frame_counter_last_valid: -1 - resolution: 1080
1617721860535: incorrect frame counter - frame_counter: 64569 - frame_counter_last_valid: -1
1617721860535: frame detected - frame_len: 1016 - frame_counter: 64569 - frame_counter_last_valid: -1 - resolution: 360
1617721860536: incorrect frame counter - frame_counter: 65173 - frame_counter_last_valid: -1
1617721860536: frame detected - frame_len: 1452 - frame_counter: 65173 - frame_counter_last_valid: -1 - resolution: 1080
1617721860561: incorrect frame counter - frame_counter: 64570 - frame_counter_last_valid: -1
1617721860561: frame detected - frame_len: 1043 - frame_counter: 64570 - frame_counter_last_valid: -1 - resolution: 360
1617721860561: incorrect frame counter - frame_counter: 65174 - frame_counter_last_valid: -1
1617721860561: frame detected - frame_len: 1467 - frame_counter: 65174 - frame_counter_last_valid: -1 - resolution: 1080
1617721860561: incorrect frame counter - frame_counter: 64571 - frame_counter_last_valid: -1
1617721860561: frame detected - frame_len: 1044 - frame_counter: 64571 - frame_counter_last_valid: -1 - resolution: 360
1617721860561: incorrect frame counter - frame_counter: 65175 - frame_counter_last_valid: -1
1617721860561: frame detected - frame_len: 1481 - frame_counter: 65175 - frame_counter_last_valid: -1 - resolution: 1080
1617721860561: incorrect frame counter - frame_counter: 64572 - frame_counter_last_valid: -1
1617721860561: frame detected - frame_len: 1075 - frame_counter: 64572 - frame_counter_last_valid: -1 - resolution: 360

>

gjai, 2021-04-06:
> root@ENFANTS:~/bin# ./rRTSPServer -m q321br_lsx -r **low -**a no -d 3
debug on, level 3
1617721976496: mapping file /dev/shm/fshare_frame_buf, size 524588, to b6d15000
1617721976497: closing the file /dev/shm/fshare_frame_buf
1617721976497: starting capture main loop
1617721976749: incorrect frame counter - frame_counter: 2081 - frame_counter_last_valid: -1
1617721976749: frame detected - frame_len: 1682 - frame_counter: 2081 - frame_counter_last_valid: -1 - resolution: 1080
1617721976774: frame detected - frame_len: 794 - frame_counter: 4078 - frame_counter_last_valid: -1 - resolution: 0
1617721976774: incorrect frame counter - frame_counter: 1458 - frame_counter_last_valid: -1
1617721976774: SPS   detected - frame_len: 19 - frame_counter: 1458 - frame_counter_last_valid: -1 - resolution: 360
1617721976774: incorrect frame counter - frame_counter: 1459 - frame_counter_last_valid: -1
1617721976775: frame detected - frame_len: 8 - frame_counter: 1459 - frame_counter_last_valid: -1 - resolution: 360
1617721976800: incorrect frame counter - frame_counter: 1460 - frame_counter_last_valid: -1
1617721976800: frame detected - frame_len: 16913 - frame_counter: 1460 - frame_counter_last_valid: -1 - resolution: 360
1617721976825: incorrect frame counter - frame_counter: 2082 - frame_counter_last_valid: -1
1617721976826: frame detected - frame_len: 1677 - frame_counter: 2082 - frame_counter_last_valid: -1 - resolution: 1080
1617721976851: incorrect frame counter - frame_counter: 1461 - frame_counter_last_valid: -1
1617721976851: frame detected - frame_len: 1299 - frame_counter: 1461 - frame_counter_last_valid: -1 - resolution: 360
1617721976876: incorrect frame counter - frame_counter: 2083 - frame_counter_last_valid: -1
1617721976876: frame detected - frame_len: 1654 - frame_counter: 2083 - frame_counter_last_valid: -1 - resolution: 1080
1617721976901: incorrect frame counter - frame_counter: 1462 - frame_counter_last_valid: -1
1617721976901: frame detected - frame_len: 1167 - frame_counter: 1462 - frame_counter_last_valid: -1 - resolution: 360
1617721976927: incorrect frame counter - frame_counter: 2084 - frame_counter_last_valid: -1
1617721976927: frame detected - frame_len: 1687 - frame_counter: 2084 - frame_counter_last_valid: -1 - resolution: 1080
1617721976952: incorrect frame counter - frame_counter: 1463 - frame_counter_last_valid: -1
1617721976952: frame detected - frame_len: 1098 - frame_counter: 1463 - frame_counter_last_valid: -1 - resolution: 360
1617721976952: incorrect frame counter - frame_counter: 2085 - frame_counter_last_valid: -1
1617721976952: frame detected - frame_len: 1651 - frame_counter: 2085 - frame_counter_last_valid: -1 - resolution: 1080
1617721976978: frame detected - frame_len: 794 - frame_counter: 4079 - frame_counter_last_valid: -1 - resolution: 0
1617721977003: incorrect frame counter - frame_counter: 1464 - frame_counter_last_valid: -1
1617721977003: frame detected - frame_len: 1082 - frame_counter: 1464 - frame_counter_last_valid: -1 - resolution: 360

roleoroleo, 2021-04-06:
The log is good.
Try to add another option: "-s".

roleoroleo, 2021-04-06:
I'm sorry...

gjai, 2021-04-06:
that's the game ;)

closed 2021-04-10 by roleoroleo

apascual, 2021-06-08:
That camera looks like the Mibao D700: https://www.amazon.de/-/en/gp/product/B08FCDJDZ9

@roleoroleo I have a one and I can try more tests if you want to try supporting it.