roleoroleo/yi-hack-Allwinner-v2 #129: Issue on r30gb high res

roleoroleo, 2021-04-06:
Moved from this issue.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/124#issuecomment-814193656

rRTSPServer doesn't rely on other programs to work.
The only difference is about the command line options.

@GizmoLibre
Please copy the new binary https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6265336/rRTSPServer.gz in the cam (/tmp/sd) and try to run it manually in this manner.
Source environment, kill watchdog and process:
```
. /tmp/sd/yi-hack/script/env.sh
killall wd_rtsp.sh
killall rRTSPServer
```
Run RTSP server for high res:
```
RRTSP_RES=high RRTSP_AUDIO=no RRTSP_MODEL=r30gb RRTSP_DEBUG=3 /tmp/sd/rRTSPServer
```
or for low res:
```
RRTSP_RES=low RRTSP_AUDIO=no RRTSP_MODEL=r30gb RRTSP_DEBUG=3 /tmp/sd/rRTSPServer
```
and post the output.

GizmoLibre, 2021-04-06:
[putty.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6266123/putty.log)

RadaKk, 2021-04-10:
Hi, 

I think I got the same kind of issue with a yi dome guard (r30gb - 0.1.5), I connect the cam to Synology  DS cam.

Although my RTSP stream in high or low res is stable, when I use ONVIF connection (in high res) it works for a few seconds and then the cam keeps rebooting. ONVIF connection with low res is stable.

roleoroleo, 2021-04-10:
Onvif is only a protocol descriptor.
When the stream starts it uses always rtsp.
If you noticed a reboot using onvif, probably it's an "out of memory" problem.
Try to enable swap file.

RadaKk, 2021-04-10:
Yes, that was it... Thank you

closed 2022-06-11 by roleoroleo