roleoroleo/yi-hack-Allwinner-v2 #132: ONVIF pairing error on Windows 10

liunett, 2021-04-13:
Hi @roleoroleo

I've tried to pair my **y21ga** with firmware _0.1.5_ and _0.1.6_ with Windows 10 device pairing (Device enumeration and pairing Windows store app) to connect to the ONVIF service, but I can't manage to.

The **NotPaired** error pop-ups altho I can see the device on the ONVIF Device Manager.

[image]

**CONFIG:**
[image]

roleoroleo, 2021-04-21:
I don't know how to fix the problem.
The cam replies to the discovery request but Windows doesn't understand the response.
I have to investigate.

liunett, 2021-05-22:
> I don't know how to fix the problem.
> The cam replies to the discovery request but Windows doesn't understand the response.
> I have to investigate.

Is there any update with the issue?

roleoroleo, 2021-05-22:
I tried to change the response from the cam but no luck.
I don't know how to fix it.

closed 2023-01-08 by stale[bot]