roleoroleo/yi-hack-Allwinner-v2 #135: Static IP

rgcouto, 2021-04-16:
Hi!

I've just hacked my first Yii cam! It worked like a charm, you definitely did a great job! Congrats!

I only have one issue with it, wouldn't it be possible to allow users to specify a static IP? When I integrate it on Home Assistant I need to specify the IP address of the cam. If this isn't static it may happen that sometime the router decides to change the IP of the cam and I'll lost the connection to it from Home Assistant. 

I know that I can set a static lease directly to the router, but I'd lose this configuration if, for instance, I'd change routers. If this could be done directly on the cam I think it would be more beneficial. 

Thanks for the great work!

roleoroleo, 2021-04-16:
Read this issue:
https://github.com/roleoroleo/yi-hack-MStar/issues/208

rgcouto, 2021-04-16:
My bad. I've only searched on open issues. Sorry!

Thanks for pointing me!

closed 2021-04-16 by rgcouto

roleoroleo, 2021-04-16:
No problem.
It's a MStar issue but it's valid for this model too.