roleoroleo/yi-hack-Allwinner-v2 #140: How to connect 2k YI Camera with Chinese Yi Home Application ?

InterStellar223, 2021-04-21:
Hello RoleoRoleo, i have a question for you.
Recently i bought a camera from Taobao, but i couldn't connect it to my phone via app.
I tried Chinese version of app also..."This camera can only be used within China"
Do you have any ideas how to connect it to my phone?
Model is 2K (Resolution 1296P) Yi Ants Y4
So i decided to make a backup and attach here:
What should i do to prevent random bricking camera?
Can you explain step by step my actions ?
https://drive.google.com/file/d/1c2_b8AmrBuimVEPA7-ONjzOTqqEi_Mpp/view?usp=sharing

InterStellar223, 2021-04-21:
Y29GA_MB_V2.1
2020/04/07
QG2101A Chip
I attached more info in my backup winrar archive.

roleoroleo, 2021-04-21:
Your cam is a y29ga.
At the moment, this hack doesn't support it.
Checking your backup it seems that you tried to apply y21ga hack but probably it doesn't work.

Normally, to use a locked cam outside China you need to install a proxy like proxychain.
But it's not implemented on this model (it's compiled but not working at the moment).

InterStellar223, 2021-04-21:
> Ваша камера - y29ga.
> На данный момент этот хак его не поддерживает.
> При проверке резервной копии кажется, что вы пытались применить хак y21ga, но, вероятно, это не сработало.
> 
> Обычно, чтобы использовать заблокированную камеру за пределами Китая, вам необходимо установить прокси, например proxychain.
> Но на данной модели это не реализовано (скомпилировано, но пока не работает).

Thank you for fast reply, tell me please what can i do to make this process goes faster ? I can send you this camera if it helps .

vin-w, 2021-04-30:
I am interested to use Y4 Ant Camera too.
But I think it is h.265 by default (not sure if the low res stream is h.264)

https://www.xiaoyi.com/homeCamera/homeCamera_y4/index.html

> Normally, to use a locked cam outside China you need to install a proxy like proxychain.
> But it's not implemented on this model (it's compiled but not working at the moment).

@roleoroleo 
Do you mean all Allwinner-v2 cannot use proxychain, or just Y4 Ant?
I could successful use proxychain with Allwinner hack that you created.

roleoroleo, 2021-04-30:
@vin-w 
Do you know the suffix?

roleoroleo, 2021-04-30:
@InterStellar223 
Please send me 2 or 3 copies of the file /dev/shm/fshare_frame_buf
Make a copy, wait 5 secs and make another copy.
I hope your cam is not in the bathroom :)

roleoroleo, 2021-04-30:
@InterStellar223 
Try this hack:
[y29ga_0.1.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6407309/y29ga_0.1.6.tar.gz)

vin-w, 2021-05-02:
> @InterStellar223
> Try this hack:
> [y29ga_0.1.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6407309/y29ga_0.1.6.tar.gz)

does this hack comes with proxychain so that i could edit /home/yi-hack/etc/proxychains.conf?

roleoroleo, 2021-05-02:
proxychain is compiled and installed but there is no switch to enable it in the web page.
Try to run it manually.

InterStellar223, 2021-05-02:
> @InterStellar223
> Please send me 2 or 3 copies of the file /dev/shm/fshare_frame_buf
> Make a copy, wait 5 secs and make another copy.
> I hope your cam is not in the bathroom :)
> 
> EDIT
> 
> Is it a home or dome?
> ethernet or wifi only?

(/dev/shm/fshare_frame_buf) I will try to send you these files tomorrow. Because first I need to get to the place of work with a Linux computer. Then try to figure out how to do it.
It is a home and it has wifi only and it demands china app only.
I tried hack you sent, here the files after trying to connect: https://drive.google.com/file/d/1wBx94iw7xsyPqa813m2OwjO7mTgB15PL/view?usp=sharing
I hope your cam is not in the bathroom :) - Haha, no :)

P.S. THANK YOU FOR HELPING ME :)

roleoroleo, 2021-05-11:
News about fshare_frame_buf?

closed 2023-01-08 by stale[bot]