roleoroleo/yi-hack-Allwinner-v2 #143: IFUSY65 - y211ga - infinite boot loop

Dlinny, 2021-04-22:
Camera: IFUSY65
FW: 9.0.36.00_202011101611
Hack: y211ga_0.1.6

Infinite boot loop: orange led -> shutter sound -> blue led blinking (about 12 times) -> restart.
Have 2 cameras, one with FW updated to 9.0.36.01_202103231716 - same behaviour.
Any suggestions on what can I try to do to successfully hack them?

Logs from SD card attached.
[df.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6358809/df.txt)
[file.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6358810/file.txt)
[lsmod.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6358812/lsmod.txt)
[ps1.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6358813/ps1.txt)

roleoroleo, 2021-04-22:
Probably the same problem described here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/119

Dlinny, 2021-04-22:
Yes, seems similar to @emnatsak issue.

Whilst waiting for solution, as its suits my requirements, I did the following:
- Removed Factory.done from SD.
- Removed lower_half_init.sh from SD.
- Put debug.sh which just do chmod & sh of system.sh from yi-hack.
This seems to work so I can see web ui and rtsp server is running (I do not need to disable cloud, as can do filter on the network level).

I do not have access to serial, but telnet/ssh is working so can provide any details available via telnet/ssh if anything can be useful for resolving boot loop.

roleoroleo, 2021-04-22:
I think the problem is rmm.
The scripts starts rmm too late and the watchdog is triggered.

dzianiwonsacze, 2021-04-24:
@Dlinny 
> * Put debug.sh which just do chmod & sh of system.sh from yi-hack.

Could you please post the contents of the file for me to check if this will do the job in my case? I'm in the same boat and will also do with rtsp for now...

roleoroleo, 2021-04-24:
Could you please try this chages?
Comment dispatch in /tmp/sd/lower_half_init.sh
```
mount --bind /backup/tools/wifidhcp.sh /home/app/script/wifidhcp.sh

#./dispatch &
#sleep 2
#./rmm &
```
And add it before rmm in /tmp/sd/yi-hack/script/system.sh
Pay attention: rmm is called twice.
```
        if [ $(get_config SPEAKER_AUDIO) != "no" ]; then
            touch /tmp/audio_in_fifo.requested
        fi
        cd /home/app
        ./dispatch &
        LD_LIBRARY_PATH="/tmp/sd/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/tmp/sd:/tmp/sd/gdb" ./rmm &
        sleep 6
```

dzianiwonsacze, 2021-04-24:
I don't understand the "rmm is called twice"part, but I did comment out dispatch and added it in the other place, and now camera is showing yellow light for about 27 seconds, then the light goes off for split second and yellow light turn on again, in an infinite loop. The camera does not work in neither yi app nor web interface.

dzianiwonsacze, 2021-04-24:
FYI, after removing the card camera goes back to non-hacked operation smoothly.

roleoroleo, 2021-04-25:
rmm is called twice because there is an "if-else" statement.
So it's called for the if and for the else.

cretep, 2021-04-25:
Reboot cycle also persists for my new _IFUS y211ga/9.0.36.01_ camera after suggested move of `./dispatch`.

In case it helps, I added logging to the executables in question (ie. `./dispatch 2>&1 >> /tmp/sd/debug/dispatch.log &`) and redacted credentials:
 * [rmm.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6371821/rmm.log) (last message: `ptz_isruning ==1 so return`)
 * [dispatch.log](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6371822/dispatch.log) (last message: `no wifi dev`)

Moving `./dispatch` has no effect on the log output.

Main script continues with last output during `./yi-hack/script/mqtt_advertise/check_conf.sh`, a non-suspect script so likely just coincidental timing with system reset.

cretep, 2021-04-25:
New wifi device driver:

```
root@(none):~# ls -l /sys/class/net/wlan0/device/driver/module
lrwxrwxrwx    1 root     root             0 Apr 25 13:06 /sys/class/net/wlan0/device/driver/module -> ../../../../module/atbm603x_wifi_usb
```

roleoroleo, 2021-04-25:
You are right.
I copied as a template the wrong lower_half_init.sh
The file is both in /backup and in /home/app
And the 2 copies are different.
I will release a new beta in 1 hour.

Thanks.

roleoroleo, 2021-04-25:
Here it is:
[y211ga_0.1.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6372315/y211ga_0.1.6.tar.gz)

Please, let me know if it works.

dzianiwonsacze, 2021-04-25:
First pass success, thanks!

roleoroleo, 2021-04-25:
Please, check if rtsp and snapshot are working.
High and low res.

a20691, 2021-04-25:
Yes it's all wotking.

tulkas84, 2021-04-27:
I had the same problem, and is fixed with the beta :)

Some info, just in case it's ussefull:
-The wifi Strength from the main page shows 0%
-Dont show de Serial Number
-rtsp audio not working

rtsp for video, ftp upload, mqtt and snapshoots, working well :)

roleoroleo, 2021-04-27:
> -The wifi Strength from the main page shows 0%

Please post the output of the command `cat /proc/net/wireless`

> -Dont show de Serial Number

`dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null`

> -rtsp audio not working

Check if it works with aac codec.

tulkas84, 2021-04-27:
> cat /proc/net/wireless

Inter-| sta-|   Quality        |   Discarded packets               | Missed | WE
 face | tus | link level noise |  nwid  crypt   frag  retry   misc | beacon | 22
 wlan0: 0000   70.  -22.  -256        0      0      0      0     35        0
  p2p0: 0000    0     0     0        0      0      0      0      0        0

> dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null

Not fixed. Rebooted, and still no serial number on the main webpage

> Check if it works with aac codec.
Works for 2 seconds, and then crash

roleoroleo, 2021-04-27:
Post the output of the command
`dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null`

roleoroleo, 2021-04-27:
About signal strength, please try to change this line in /tmp/sd/yi-hack/www/cgi-bin/status.json
`WLAN_STRENGTH=$(cat /proc/net/wireless | grep wlan0 | awk '{ print $3 }' | sed 's/\.$//')`

tulkas84, 2021-04-27:
there is no output for dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null
Without the redirection: 
20+0 records in
20+0 records out
20 bytes (20B) copied, 0.007397 seconds, 2.6KB/s

> WLAN_STRENGTH=$(cat /proc/net/wireless | grep wlan0 | awk '{ print $3 }' | sed 's/\.$//')

Works!

roleoroleo, 2021-04-27:
Try this 
`dd bs=1 count=20 skip=656 if=/tmp/mmap.info of=/tmp/out.bin 2>/dev/null`
and check the content of the file /tmp/out.bin

tulkas84, 2021-04-27:
root@yi-cloe:~# dd bs=1 count=20 skip=656 if=/tmp/mmap.info of=/tmp/out.bin 2>/dev/null
root@yi-cloe:~# cat /tmp/out.bin
root@yi-cloe:~#

Nothing

roleoroleo, 2021-04-28:
Could you send me a copy ot the mmap.info?
Pay attention: the file contains personal info.
You can send it to the email address connected to the paypal address (see the readme).

roleoroleo, 2021-04-28:
Please change 656 with 784.
It should work.
Il will fix in the next release.

BobMiles, 2021-10-30:
Hi,

I have a:

Firmware Version | 0.1.9
Base Version | 9.0.36.00_202011101611
Model Suffix | y211ga
Serial Number | IFUSY65GAUJWFP

And my problem is similar. The camera comes on, I can access Yi-Hack webinterface just fine and configure the camera. But after some minutes it reboots - I see the Uptime timer restart.

I already tried different SD Cards, but to no avail.

Is there any debugging I can do?

Thank you so much!

roleoroleo, 2021-11-01:
Enable swap space.

closed 2023-01-08 by stale[bot]