roleoroleo/yi-hack-Allwinner-v2 #147: RTSP audio not working on y21ga

ivanfor, 2021-04-27:
Hi.
RTSP audio is not working on y21ga. However, recordings do have audio. I can reproduce it in 4 identical cams, so it doesn't seem hw related.
```
ivan@sinchan:~$ ffmpeg -hide_banner -i  rtsp://ctest/ch0_0.h264
Input #0, rtsp, from 'rtsp://ctest/ch0_0.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
  Duration: N/A, start: -0.046489, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 39 tbr, 90k tbn, 40 tbc
At least one output file must be specified
```
```
ivan@sinchan:~$ ffmpeg -hide_banner -i  rtsp://ctest/ch0_2.h264
[rtsp @ 0x5633d605a600] method DESCRIBE failed: 404 File Not Found, Or In Incorrect Format
rtsp://ctest/ch0_2.h264: Server returned 404 Not Found
```

Thinking it could be [this](https://github.com/roleoroleo/yi-hack-Allwinner/issues/237), I removed user and password but nothing changed.
Interestingly, AAC codec does work:
```
ivan@sinchan:~$ ffmpeg -hide_banner -i  rtsp://ctest/ch0_0.h264
Input #0, rtsp, from 'rtsp://ctest/ch0_0.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
  Duration: N/A, start: 0.000250, bitrate: N/A
    Stream #0:0: Video: h264 (Main), yuvj420p(pc, bt709, progressive), 1920x1080, 20 fps, 20 tbr, 90k tbn, 40 tbc
    Stream #0:1: Audio: aac (LC), 8000 Hz, mono, fltp
At least one output file must be specified
ivan@sinchan:~$ ffmpeg -hide_banner -i  rtsp://ctest/ch0_2.h264
Input #0, rtsp, from 'rtsp://ctest/ch0_2.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_2.h264
  Duration: N/A, start: 0.000000, bitrate: N/A
    Stream #0:0: Audio: aac (LC), 8000 Hz, mono, fltp
At least one output file must be specified
```

Is this a known bug?
Is there anything I can do to debug?
Thanks

roleoroleo, 2021-04-27:
There is a know bug, probably it depende on the old g++ used.
Sometimes when I build the solution, audio doesn't work.
Changing  optiomization options it comes back.
Please check if it works using aac codec.

ivanfor, 2021-04-27:
> There is a know bug, probably it depend on the old g++ used.

Yep. Aac works.
Should I recompile?

roleoroleo, 2021-04-27:
I will recompile and test it asap.
Are you using authentication with rtsp?

ivanfor, 2021-04-27:
I've tried both with and without auth. Same result.

roleoroleo, 2021-04-27:
Please overwrite this binary and let me know if it works:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6386560/rRTSPServer.gz)

ivanfor, 2021-04-27:
It's working perfectly, with and without authentication.
Thanks for this!

4ndinger, 2021-04-29:
Hi,
on my CAM IFUS with y21ga I have the same issue. Currently I'm not able to compile it for myself.
When I change to aac the audio is working but the video stream is very slow (I think aac is to cpu itense)
Will there come an update to have audio (not aac) available?

Best regards!
Andi

roleoroleo, 2021-04-29:
Use the binary above.

roleoroleo, 2021-05-01:
Please check if this fw works.
If it's ok, I will create a new release.

[y21ga_0.1.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6409361/y21ga_0.1.6.tar.gz)

ivanfor, 2021-05-01:
> Please check if this fw works.
> If it's ok, I will create a new release.
> 
> [y21ga_0.1.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6409361/y21ga_0.1.6.tar.gz)

I've extracted and overwritten /tmp/sd contents in my test cam, and everything seems to be working well.

roleoroleo, 2021-05-01:
> I've extracted and overwritten /tmp/sd contents in my test cam, and everything seems to be working well.

rtsp too?

ivanfor, 2021-05-01:
Yes. Rtsp and audio (only tested ulaw, though)

RedKage, 2021-05-01:
Works on my end as well, sound is back!

closed 2021-05-07 by roleoroleo