roleoroleo/yi-hack-Allwinner-v2 #148: Only get 1 entity in Home Assistant when adding as ONVIF

willsy555, 2021-04-29:
Thanks for all your hard work roleoroleo! :-)

I've just bought a Yi home y21ga from Amazon UK. Installed the hack fine and can see the UI and stream. 

Only problem is when I add as ONVIF onto Home Assistant integrations, I can only see 1 entity (the camera).

I thought there should be others. i.e. Tamper etc...

Tried the latest v6. Also tried v3 as others were having issues with ONVIF. Still only 1 entity.

Never used ONVIF before so might be doing something wrong?

Cheers

Will

roleoroleo, 2021-04-29:
I tried to implement the sensors in onvif but I had to stop because of memory leak problems.
If you use home assistant, try this:
https://github.com/roleoroleo/yi-hack_ha_integration

willsy555, 2021-04-29:
Ah thank you @roleoroleo. I should have started without preconceptions! I'll look at those steps.. Looking forward to trying out all the functions. This is awesome stuff!! :-)

closed 2021-04-29 by willsy555