roleoroleo/yi-hack-Allwinner-v2 #151: MQTT : motion_detection stay at 0

cvalentin-dkt, 2021-05-03:
Hello,

I have hack my yi home with your hack and enable MQTT but motion_detection  stay at 0 when motion detect.
Moreover when I arrive on the home page of the camera, all information is empty.

Firmware : 0.1.7

pabgn, 2021-05-04:
Same on r30gb. motion_detecion never publishes events when motion is detected.

roleoroleo, 2021-05-04:
Please post your mqtt config.

pabgn, 2021-05-05:
**General**
Enabled: True
Server IP: 192.168.1.5
Server Port: 1883
Client Id: yi-cam

**Authentication**
Username: xxxx
Password: xxxx

**Topics**
Topics Prefix: yicam
Motion Topic Suffix: motion_detection
Motion Start Message: motion_start

**Advanced**
Keepalive: 120
QoS: 1
Retain for birth and will messages: 1
Retain for motion messages: 1
Retain for motion image messages: 1
Retain for video list messages: 1

--

On topic yicam/motion_detection there's only one message: **motion_stop.** 

<br class="Apple-interchange-newline" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none;">

<br class="Apple-interchange-newline" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; text-decoration: none;">

roleoroleo, 2021-05-05:
Are you using human detection?

pabgn, 2021-05-05:
No, it's turned off.

roleoroleo, 2021-05-05:
If the web page is not available probably the hack is not working properly.
Please checkt the sd card and install again the hack.

If you are using human detection, read this issue: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/75

closed 2023-01-08 by stale[bot]