roleoroleo/yi-hack-Allwinner-v2 #152: MQTT connect loop.

foxel, 2021-05-07:
BFUS, 9.0.19*, y21ga flashed

MQTT flood with connected message.

I've tried running the mqttv4 on the camera via ssh:
```console
Starting mqttv4 v0.0.3
key: SOUND_DETECTION | value: sound
Unrecognized config.
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m1, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 1)
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m2, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 2)
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m3, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 3)
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m4, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 4)
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m5, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 5)
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m6, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 6)
Trying to connect... Client yi-cam-3 sending CONNECT
Client yi-cam-3 received CONNACK (0)
Client yi-cam-3 sending PUBLISH (d0, q1, r1, m7, 'foxhome/yi-cam-3/status', ... (6 bytes))
. connected!
Client yi-cam-3 received PUBACK (Mid: 7)
^C
```

Can you point me where to dig further?

roleoroleo, 2021-05-07:
Try the following steps:
1 - disable mqtt-config setting "Topics Prefix for remote configuration" to blank
2 - check if you are using the same client id with 2 or more devices

foxel, 2021-05-07:
Crap... the second case was exectly mine... I forgot I have assigned yi-cam-3 already %)

closed 2021-05-07 by foxel