roleoroleo/yi-hack-Allwinner-v2 #153: Suddenly no more WIFI Connection and permanent reboots

misterboo, 2021-05-07:
I have an BFUSY44 Camera

It was working perfecly until yesterday. Suddenly the camera reboots permanantly. Seems it cannot connect anymore to wifi.

There is an log_no_wifi_dev.tar.gz file. in the lsmod.txt the wlan driver 8189fs is missing compared to the log_wifi_connected.tar.gz from the time, the camera was working.

Any Ideas?

roleoroleo, 2021-05-07:
Is the cam hacked or not?

misterboo, 2021-05-07:
It was hacked with y21ga_0.1.4. As i said:  until yesterday the was perfectly working

PKilgannon, 2021-05-14:
Bingo, I have exactly the same problem!

roleoroleo, 2021-05-14:
If you remove the sd, does the cam works?

PKilgannon, 2021-05-14:
I paired the wifi using the YI Home app. Updated the firmware and applied the newest version of the YI hack and it's working again.

Thanks .

closed 2023-01-08 by stale[bot]