roleoroleo/yi-hack-Allwinner-v2 #154: Donation rtsp Yi cam outdoor

pjmakey2, 2021-05-09:
Hi @roleoroleo

Here is the thing.

I was able to use the firmware for a Yi-outdoor camera, but I realize that I need to make a donation to access a hash for active the RSTP service. 

I try to do that, but the mechanism to make the donation is blocked in my Country, then I try to do that from a USA account, and the donation don't go further. 

Is there another way to reach you and make the donation for your work,  so you can give me the hash file for active the RSTP service ?

I'll be waiting for your contact.

Thanks in advance.

pjmakey2, 2021-05-09:
Hi.

The "yi-hack-allwinner" work with "Yi outdoor IFUS" ?.

If I made a donation through the donation link on this page, can I send you the hash of my camera, so you can give me the file to enable the RSTP service ?

Thank in advanced.

roleoroleo, 2021-05-10:
If you are using yi-hack-v4 you need to send a donation to the owner of that project: https://github.com/TheCrypt0/yi-hack-v4
Not to me.
If you are using yi-hack-MStar, yi-hack-Allwinner or yi-hack-Allwinner-v2 you don't have to pay.

roleoroleo, 2021-05-10:
Or try to use this hack:
https://github.com/alienatedsec/yi-hack-v5/
if your cam is compatible.

pjmakey2, 2021-05-19:
Sorry, for the delay to respond. Ok I will try that. Thanks for your time. Certainly I will donate to your project.

closed 2021-05-19 by pjmakey2