roleoroleo/yi-hack-Allwinner-v2 #156: Sound detection publish to MQTT only once per boot

r-science, 2021-05-12:
Hi, 
I have problem with sound detection. It's publish sound detecion message to MQTT only once per boot. There is not another messages on this topic till I reboot my cam (yi Home 1080p). I'm tried changing sensitivity and retain flag (MQTT) but with no luck.

Regards

roleoroleo, 2021-05-12:
Is retain enabled?

roleoroleo, 2021-05-12:
How sound detection works is not so clear to me (yi side).
But this type of problem is new.

Try to kill mqttv4 and run manually and check the output.

r-science, 2021-05-12:
`Starting mqttv4 v0.0.3
key: MQTT_PREFIX_CONFIG | value: yicam-config
Unrecognized config.
key: SOUND_DETECTION | value: sound
Unrecognized config.
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
Client yi-cam sending PUBLISH (d0, q1, r1, m1, 'yicam/status', ... (6 bytes))
. connected!
Client yi-cam received PUBACK (Mid: 1)
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m2, 'yicam/motion_detection', ... (11 bytes))
Client yi-cam received PUBACK (Mid: 2)
CALLBACK MOTION START
Client yi-cam sending PUBLISH (d0, q1, r1, m3, 'yicam/motion_detection', ... (12 bytes))
Starting program
Mapping file /dev/shm/fshare_frame_buf, size 1786224, to b6d67000`

Is this normal?:
key: SOUND_DETECTION | value: sound
Unrecognized config.

roleoroleo, 2021-05-13:
Check the file mqttv4.conf: check if SOUND_DETECTION_MSG exists.
If it doesn't exist, rename SOUND_DETECTION to SOUND_DETECTION_MSG.
If exists remove SOUND_DETECTION line completely.

closed 2023-01-08 by stale[bot]