roleoroleo/yi-hack-Allwinner-v2 #158: Problem with decoding video in Samsung Galaxy S20plus

playersct, 2021-05-14:
Hello

I have problem with decoding video stream in Samsung Galaxy S20+. No matter where I run it - be it in vlc or at home assistant official app i have something like that:

[image]

In HA I tried your's HA integration and webrtc. it is the same in both cases. trying to run rstp stream in vlc has the same effect.
In my windows laptop is ok, in my wife's Xiaomi Note 8 Pro also is ok.

Any help?

jacek903, 2021-08-29:
I have exactly the same problem. In VLC on s10 or s21, only when hardware acceleration in settings is off, rtsp stream is OK. Others, ie HA with webrtc or logged via chrome, the rtsp stream is very disrupted. The yi application on the phone works well.

When I check the same on huawei or old Android phones everything is fine.

Firmware version: 0.1.8
Basic version: 9.0.26.07_202104231123
Model: h30ga

I also use the y21g with the same firmware. Everything works very well on windows, android and ios.

jacek903, 2021-09-18:
It seems that I found a temporary solution. I experimentally replaced rRTSPServer file from the y21ga_0.1.3 version and now everything works very well. There is no problem with displaying rtsp on any of my mobile devices now.

@roleoroleo I know this is only a temporary solution, but this problem here only applies to rRTSPServer and its later changes. Would it be possible to check it and fix this in new releases?

roleoroleo, 2021-09-19:
> I experimentally replaced rRTSPServer file from the y21ga_0.1.3 version and now everything works very well. There is no problem with displaying rtsp on any of my mobile devices now.

Wich model? h30ga?

jacek903, 2021-09-20:
Yes, h30ga, YI Outdoor. I also checked the later versions of rRTSPServer, so from 0.1.4 also works well. The issue starts with the 0.1.5 and it does not work at all, version 0.1.6 and later work but with the same effect as above, means with image distortion on some mobile devices.

Anyway, great project, your beers on the way :-)

roleoroleo, 2021-09-20:
This is the list of the canges between 0.1.4 and 0.1.5 (about RTSP server):

1. Changed optimization flag from -O1 to -O0
2. Added aac audio with fdk-aac library
3. Changed from byte stream to framed source
4. Upgraded live library from live.2020.05.15.tar.gz to live.2021.01.29.tar.gz
5. Changed presentation time

I don't know which change could cause this issue.
I can try to roll back line 1, line 4 and line 5.
About 2 and 3 I can't go back.

jacek903, 2021-09-20:
If possible to roll back line 1, 4, 5, I can test it. However, I am afraid that the problem may be related to the point 3.

roleoroleo, 2021-09-20:
Here it is:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7195976/rRTSPServer.gz)

jacek903, 2021-09-20:
Thanks @roleoroleo Unfortunately, does not work :-(

[image]

0-Al, 2021-09-20:
Not sure if it's worth anything but I got results like that picture when setting a higher resolution than the camera was outputting, i.e the camera was streaming 1080p while I set 3K for the RTSP. I notice there's an AUTO/SD/HD setting in the app, idk what auto does or if it would cause issues.

jacek903, 2021-09-20:
This screenshot is from HA, When I check the same in VLC for Android it is OK only when hardware acceleration in the app settings is off.
I do not know how to set 3K for the rtsp as long as we are talking about HA and WebRTC Camera integration or yi app. SD/HD/Auto settings in yi app have no effect on what happens with the rtsp imo.

roleoroleo, 2021-09-20:
There is no difference between y21ga abd y211ga about RTSP and stream grabbing.
I can't understand why your y21ga works and y211ga doesn't.

jacek903, 2021-09-20:
Yes, I am currently using h21ga, with the latest release 0.1.9 - no change and h30ga with rRTSPServer file from y21ga/y211ga version 0.1.4 and the rest from release 0.1.9 . It works well in all devices. 
The streaming problem only affects some mobile devices (in my case iPad, s10 and s21) and the h30ga camera. I can live with that because all the functionality of yi-hack works well. I hope that the manual file swap at the next release will be possible to get the correct RTSP.

jacek903, 2021-09-21:
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7200605/fshare_frame_buf.zip)

roleoroleo, 2021-09-21:
Probably I found the problem.
The yi firmware of your cam is newer than what I have tested.
And the SPS of the h264 stream is different.
Try this new binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7202123/rRTSPServer.gz)

jacek903, 2021-09-21:
Great! Yes, everything works very well. Many thanks @roleoroleo

closed 2023-01-08 by stale[bot]