roleoroleo/yi-hack-Allwinner-v2 #160: r30gb - No onvif stream video available

Camaca71, 2021-05-17:
Hi, 
I own a IFUS camera (dome guard) hacked r30gb v0.1.7.
It works fine by using rtsp but it seems not stable on onvif. I have tried different apps but only Onvif Device Manger is able to detect it. Nevertheless no stream video is available (I see "no signal" on black screen) even if the PT features seems working.
Thank you in advance for any help or suggestion.

Camaca71, 2021-05-24:
Fixed by using the low resolution stream.

roleoroleo, 2021-05-24:
I never completed the final release for this model because it's the only model usgin h265.
Could you please give me a feedback about rtsp?
Is high res rtps working properly?
Is low res rtps working properly?

Camaca71, 2021-05-24:
Hello.
It works fine only by using low res rtps. With high res rtps no stream video is availabe.
Thank you

roleoroleo, 2021-05-24:
Could you please send me 2 or 3 copies of the file /dev/shm/fshare_frame_buf?

Copy the file, wait 5 seconds, copy again...

Is the file that contains the video buffer.

Camaca71, 2021-05-24:
Of course. 
How can I send you? 
Can I attach here?
Rgs

roleoroleo, 2021-05-24:
Yes, I will remove it.

WalkingDot, 2021-05-25:
Is this only an ONVIF related issue?
Because the high resolution stream from my r30gb is working.

I'm using ffplay:
```Input #0, rtsp, from 'rtsp://usr:pw@ip/ch0_0.h264':
  Metadata:
    title           : Session streamed by "rRTSPServer"
    comment         : ch0_0.h264
  Duration: N/A, start: 0.000633, bitrate: N/A
    Stream #0:0: Video: hevc (Main), yuvj420p(pc, bt709), 1920x1080, 20 fps, 19.98 tbr, 90k tbn, 19.98 tbc
[hevc @ 0xffff80069980] Could not find ref with POC 7    0B f=0/0   
[swscaler @ 0xffff7c693cf0] deprecated pixel format used, make sure you did set range correctly```

Camaca71, 2021-05-25:
It's only related with Device Manager software. No matter by using home assitant in onvif mode.

roleoroleo, 2021-05-26:
> It's only related with Device Manager software. No matter by using home assitant in onvif mode.

Are you using authentication with rtsp?

Camaca71, 2021-05-26:
I am not sure about it. Let me better explain my scenario.
First of all, I am using the Onvif Device Manager on a Windows workstation. 
When I launch the software, it automatically finds the camera on my local network. If I click on "live stream" menu, I can't see anything (black screen with "no signal"). 
If I preliminary change the camera resolution on the YI-hack web server (disabling high res rtsp and keeping low res) and I repeat the step above, I can see the video stream. 
In other words, Onvif Device Manager recognizes the camera but allows the stream video only when high res rtsp is disabled.
Thank you for your help.

roleoroleo, 2021-05-26:
There are 2 different options in the Configuration web page:
RTSP Stream: select the stream you want to generate (low, high or both).
ONVIF Profile: select the profile you want to publish with onvif (low, high or both).
Check if you have set them correctly.

| ONVIF | Stream | Correct |
| Low | Low | Yes |
| Low | High | No |
| Low | Both | Yes |
| High | Low | No |
| High | High | Yes |
| High | Both | Yes |
| Both | Low | No |
| Both | High | No |
| Both | Both | Yes |

Camaca71, 2021-05-30:
Hi, 
I've checked the configuration. Actually it runs as the table above.
Thank you very much.

closed 2023-01-08 by stale[bot]