roleoroleo/yi-hack-Allwinner-v2 #161: Support for QFUS - yi 1080p PTZ outdoor Camera

ABrauser, 2021-05-17:
Hello Roleoroleo,

is it possible to customize the firmware to support the yi 1080p PTZ outdoor Camera or is it all ready compatible? This model is quite new I guess. and I bought it via Ali Express. it is fully integrated with the Yi Home  App, but I was not able to find a custom firmware to use the rtsp streams. Regarding some Discord channels some of the developer mentioned that your firmware is the closest one that is compatible with the camera. Is the true or does it require some development time to get it working? Firmware Version is 9.0.45.00_202101251914 and the ID Beginns with "QFUSXXXXXXX"

bfboarder, 2021-05-31:
Same camera here, seems to do something r30gb (does boot into yi hack and can get on the yi hack webinterface)
can take a snapshot, but the image seems encrypted, after taking a snapshot the cam reboots.

Would be nice to have support for it, if testing is needed i'll be happy to do some

roleoroleo, 2021-06-02:
Are you able to dump a backup?
Check the wiki.

bfboarder, 2021-06-03:
> Are you able to dump a backup?
> Check the wiki.
> 
> About reboot, enable swap file.

Yes, backup seems to have worked

[QFUS_PTZ_backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6589094/QFUS_PTZ_backup.zip)

roleoroleo, 2021-06-03:
Your cam is a r40ga.
I will try to add it.

roleoroleo, 2021-06-04:
Here it is a first alpha version:
[r40ga_0.1.7.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6597833/r40ga_0.1.7.tar.gz)

Probably rtsp and snapshot don't work.
Install it and check if ssh and web work.

bfboarder, 2021-06-04:
Man your fast :-) thanks for all the work your putting into this.
Below my test findings on the alpha version:

Configuration:
Hostname - works
date time server and timezone - works
ftp - works
ssh - works
telnet - works
Mqtt seems to work, but when enabled and rebooting the cam using the webinterface it will never let you get back into the webinterface without powercycling the camera bij unplugging it from the mains.
Allso making changes to other settings and rebooting via maintanance locks you out of the webinterface untill you power cycle.
Disabeling yi cloud does work, but no files are stored local
Motion events does seem to create a folder, but does not record
On ssh I get "mount: mounting none on /sys/kernel/debug failed: Resource busy"
Not sure if this is related to the not recording and the freezing after sooft reboot, but it might be that it can not mount the sd correctly?!?
snapshot - not working
rtsp - not working
onviv -  seems to find the camera but shows a connection refused
ptz - not working (stating it is not availale on this cam)

roleoroleo, 2021-06-06:
Ok, send me 3 copies of your /dev/shm/fshare_frame_buf
It's the video buffer.
Make a copy, wait 5 seconds, make another copy, wait 5 seconds and make the last copy.
Compress it and send it to me.
I will try to fix rtsp and snapshot.

bfboarder, 2021-06-06:
Not sure how exact the 5 seconds in between have to be......but i tried to stay close to 5 seconds :) 

[dev_shm_fshare.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6604700/dev_shm_fshare.zip)

roleoroleo, 2021-06-07:
The video is very different.
I need a lot of time to make this cam compatible.

bfboarder, 2021-06-07:
> The video is very different.
> I need a lot of time to make this cam compatible.

No problem, the cam is not going anywhere so take all the time you need.
I wish I could help, but my linux and camera knowledge is not so good.
If I can do anything, help testing or sending files/logs let me know, i'll do what i can.

ABrauser, 2021-06-07:
i will donate you some beers to keep the stress level down during the process! cheers :)

roleoroleo, 2021-06-07:
Thank you, I'll let you know as soon as possible.

roleoroleo, 2021-06-08:
Overwrite the file in /tmp/sd/yi-hack/bin with the content of this archive:
[r40ga.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6614144/r40ga.tar.gz)

bfboarder, 2021-06-08:
> Overwrite the files in /tmp/sd/yi-hack/bin with the content of this archive:
> [r40ga.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6614144/r40ga.tar.gz)

Test with the new files.
no luck
- snapshot paes stays empty
- no video is saved in the motion tab (only empty date time folders are created
- VLC won't connect to the stream (vlc log below)
- onvif does find cam but gives connection error as well

-- logger module started --
main: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
main: playlist is empty
live555 error: Failed to connect with rtsp://192.168.1.43:554
main error: connection failed: Connection refused by peer
satip error: Failed to connect to RTSP server 192.168.1.43:554
main error: connection failed: Connection refused by peer
access_realrtsp error: cannot connect to 192.168.1.43:554
live555 error: Failed to connect with rtsp://192.168.1.43:554/ch0_0.h264
main error: connection failed: Connection refused by peer
satip error: Failed to connect to RTSP server 192.168.1.43:554
main error: connection failed: Connection refused by peer
access_realrtsp error: cannot connect to 192.168.1.43:554
-- logger module stopped --

roleoroleo, 2021-06-08:
>     * snapshot page stays empty

Try to run the webservices:
http://192.168.1.200:8080/cgi-bin/snapshot.sh
http://192.168.1.244:8080/cgi-bin/snapshot.sh?res=low

>     * VLC won't connect to the stream (vlc log below)

Did you try both high and low res?
Log in with ssh and check if programs are running (command ps).

>     * onvif does find cam but gives connection error as well

This is normal because rtsp doesn't work.

bfboarder, 2021-06-08:
Both web requests return a blank page (empty even in the dev tools there are no tags or anything) only a 200 OK

Request URL: http://192.168.1.43:8080/cgi-bin/snapshot.sh?res=low
Request Method: GET
Status Code: 200 OK

Below the list of running programs:
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  207 root         0 SW   [oom_reaper]
  208 root         0 SW<  [writeback]
  210 root         0 SW   [kcompactd0]
  211 root         0 SW<  [crypto]
  212 root         0 SW<  [bioset]
  214 root         0 SW<  [kblockd]
  255 root         0 SW   [sys_user]
  263 root         0 SW<  [cfg80211]
  269 root         0 SW<  [watchdogd]
  289 root         0 SW<  [spi0]
  299 root         0 SW   [kswapd0]
  383 root         0 SW<  [bioset]
  388 root         0 SW<  [bioset]
  393 root         0 SW<  [bioset]
  398 root         0 SW<  [bioset]
  403 root         0 SW<  [bioset]
  408 root         0 SW<  [bioset]
  413 root         0 SW<  [bioset]
  418 root         0 SW<  [bioset]
  423 root         0 SW<  [bioset]
  451 root         0 SW   [irq/302-sunxi-m]
  453 root         0 SW   [irq/166-sdc0 cd]
  509 root         0 SW<  [bioset]
  510 root         0 SW   [mmcqd/0]
  535 root         0 SW<  [kworker/0:1H]
  551 root         0 SWN  [jffs2_gcd_mtd4]
  557 root       980 S    -/bin/sh
  588 root       980 S    /usr/sbin/telnetd
  615 root      1172 S    ./dispatch
 1114 root     52624 S    ./rmm
 1176 root         0 SW   [RTW_CMD_THREAD]
 1195 root      1424 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
 1433 root       980 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -
 1586 root      2896 S    ./mp4record
 1601 root       860 S    httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.
 1619 root       844 S    pure-ftpd (SERVER)
 1634 root       876 S    dropbear -R -B
 1636 root       668 S    ipc_multiplexer
 1835 root       860 S    ntpd -p pool.ntp.org
 1911 root       996 S    {wd_rtsp.sh} /bin/sh /tmp/sd/yi-hack/script/wd_rtsp.
 1967 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --
 1968 root      2420 S    framefinder r40ga
 1993 root       860 S    /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/cr
13609 root         0 SW   [kworker/u2:1]
22850 root         0 SW   [kworker/u2:2]
23347 root         0 SW   [kworker/0:1]
26439 root         0 SW   [kworker/0:0]
26777 root         0 SW   [kworker/u2:0]
28729 root       900 R    dropbear -R -B
28784 root       980 S    -ash
28868 root       980 S    sleep 10
28890 root       980 R    ps

roleoroleo, 2021-06-08:
rtsp is not running, I don't know why.
Try to run it manually and check the log:
```
killall wd_rtsp.sh
. /tmp/sd/yi-hack/script/env.sh
RRTSP_MODEL=r40gb RRTSP_RES=both RRTSP_AUDIO=no RRTSP_DEBUG=1 rRTSPServer
```

bfboarder, 2021-06-08:
Do the 
. /tmp/sd/yi-hack/script/env.sh 
and 
RRTSP_MODEL=r40gb RRTSP_RES=both RRTSP_AUDIO=no RRTSP_DEBUG=1 rRTSPServer 
need to be on 1 line?
if i do it on 2 lines i get the error below

on the same line i get no error, but it results in the server still not running i think

root@TestYi:~# killall wd_rtsp.sh
root@TestYi:~# . /tmp/sd/yi-hack/script/env.sh
root@TestYi:~# RRTSP_MODEL=r40gb RRTSP_RES=both RRTSP_AUDIO=no RRTSP_DEBUG=1 rRT
**SPServer
Segmentation fault**
root@TestYi:~# killall wd_rtsp.sh
killall: wd_rtsp.sh: no process killed
root@TestYi:~# . /tmp/sd/yi-hack/script/env.sh RRTSP_MODEL=r40gb RRTSP_RES=both
RRTSP_AUDIO=no RRTSP_DEBUG=1 rRTSPServer
root@TestYi:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  207 root         0 SW   [oom_reaper]
  208 root         0 SW<  [writeback]
  210 root         0 SW   [kcompactd0]
  211 root         0 SW<  [crypto]
  212 root         0 SW<  [bioset]
  214 root         0 SW<  [kblockd]
  255 root         0 SW   [sys_user]
  263 root         0 SW<  [cfg80211]
  269 root         0 SW<  [watchdogd]
  289 root         0 SW<  [spi0]
  299 root         0 SW   [kswapd0]
  383 root         0 SW<  [bioset]
  388 root         0 SW<  [bioset]
  393 root         0 SW<  [bioset]
  398 root         0 SW<  [bioset]
  403 root         0 SW<  [bioset]
  408 root         0 SW<  [bioset]
  413 root         0 SW<  [bioset]
  418 root         0 SW<  [bioset]
  423 root         0 SW<  [bioset]
  451 root         0 SW   [irq/302-sunxi-m]
  453 root         0 SW   [irq/166-sdc0 cd]
  509 root         0 SW<  [bioset]
  510 root         0 SW   [mmcqd/0]
  535 root         0 SW<  [kworker/0:1H]
  551 root         0 SWN  [jffs2_gcd_mtd4]
  556 root       980 S    -/bin/sh
  588 root       980 S    /usr/sbin/telnetd
  615 root      1172 S    ./dispatch
 1114 root     52692 S    ./rmm
 1143 root         0 SW   [kworker/u2:2]
 1177 root         0 SW   [RTW_CMD_THREAD]
 1195 root      1424 S    /backup/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
 1252 root       980 S    udhcpc -i wlan0 -b -s /backup/tools/default.script -
 1469 root      2932 S    ./mp4record
 1485 root       860 S    httpd -p 8080 -h /tmp/sd/yi-hack/www/ -c /tmp/httpd.
 1496 root       844 S    pure-ftpd (SERVER)
 1510 root       876 S    dropbear -R -B
 1513 root       668 S    ipc_multiplexer
 1633 root       860 S    ntpd -p pool.ntp.org
 1734 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --
 1735 root      2420 S    framefinder r40ga
 1752 root       860 S    /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/cr
 8436 root         0 SW   [kworker/0:0]
11064 root         0 SW   [kworker/0:2]
11748 root         0 SW   [kworker/u2:1]
15051 root         0 SW   [kworker/0:1]
16334 root       900 S    dropbear -R -B
16347 root       984 S    -ash
17053 root       980 R    ps

bfboarder, 2021-06-08:
Some more testing makes me think "RRTSP_MODEL=r40gb" is generating the Segmentation fault error, if i change it to any other model it does not give a fault but starts spitting out:
1623176041877: warning - unexpected NALU header
1623176041902: warning - unexpected NALU header
1623176041927: warning - unexpected NALU header
1623176041953: warning - unexpected NALU header
1623176041978: warning - unexpected NALU header
1623176042003: warning - unexpected NALU header
1623176042028: warning - unexpected NALU header
1623176042053: warning - unexpected NALU header
1623176042079: warning - unexpected NALU header

roleoroleo, 2021-06-08:
Sorry, model is r40ga and not r40gb as I wrote above.

bfboarder, 2021-06-08:
> Sorry, model is r40ga and not r40gb as I wrote above.

lol...i tested all the options and did not notice the type error :)

anyway, the same error when using r40ga

root@TestYi:~# RRTSP_MODEL=r40ga RRTSP_RES=both RRTSP_AUDIO=no RRTSP_DEBUG=1 rRT
SPServer
Segmentation fault

roleoroleo, 2021-06-08:
Thank you for your donation.
This segfault is strange.
Looking at the code I don't understand.
I tried to start it on my y21ga and I have no errors.

Please try to run it with --help:
`rRTSPServer --help`

bfboarder, 2021-06-08:
dev/shm/fshare_frame_buf is 1786156

in the help the model is missing, not sure if that has something to do with it?

Usage: rRTSPServer [-r RES] [-p PORT] [-d]

        -m MODEL, --model MODEL
                set model: y21ga, y211ga, h30ga, r30gb, h52ga, h51ga or q321br_l                       sx (default y21ga)
        -r RES,   --resolution RES
                set resolution: low, high or both (default high)
        -a AUDIO, --audio AUDIO
                set audio: yes, no, alaw, ulaw, pcm or aac (default yes)
        -p PORT,  --port PORT
                set TCP port (default 554)
        -s,       --sti
                don't overwrite SPS timing info (default overwrite)
        -d DEBUG, --debug DEBUG
                0 none, 1 grabber, 2 rtsp library or 3 both
        -h,       --help

roleoroleo, 2021-06-09:
Thanks, I will check the code again.

roleoroleo, 2021-06-09:
If  the help doesn't show r40ga model, you aren't running the new binary.
Please check it.

bfboarder, 2021-06-09:
Snapshot + higres +lowres rtsp is working ! thanks man!

> If the help doesn't show r40ga model, you aren't running the new binary.
> Please check it.

You are right, it was still running the old files when i checked on the SD card.

> Overwrite the files in /tmp/sd/yi-hack/bin with the content of this archive:
> [r40ga.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6614144/r40ga.tar.gz)

It looks like the overwriting of the files did not work.
I removed the SD card and changed the files directly on it now with more succes.

roleoroleo, 2021-06-09:
Good news!
Now, please check other services: onvif, ptz, etc...

roleoroleo, 2021-06-09:
Please try this new archive.
I made some code improvement.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/179#issuecomment-857515796

bfboarder, 2021-06-09:
> Please try this new archive.
> I made some code improvement.
> [#179 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/179#issuecomment-857515796)

Tests done with new archive:
- RTSP High/low - OK
- Snapshot - OK
- SSH - OK
- Telnet - OK
- FTP - OK
- Legacy FTP - NOT TESTED
- Swap File - NOT TESTED
- MQTT - OK
- NTPD - OK
- PTZ - Pan/Tilt working, no Zoom option on the webinterface
- Onvif - detected on network but can't connect
- Motion Events - motion is detected but files are not saved (date/time folder is created but no recordings in it)

bfboarder, 2021-06-09:
> * Motion Events - motion is detected but files are not saved (date/time folder is created but no recordings in it)

Some extra info on this.......
- recordings are made and stored in /tmp/sd/record/ (can find them using ftp)
- recordings do not show on web interface or on sd card

the above and the fact tat the files where not updated when doing the " Overwrite the files in /tmp/sd/yi-hack/bin with the content of this archive:" makes me think that the cam does store things in it's memory but is not retaining it on the SD 
But i might be completly off...........

bfboarder, 2021-06-09:
> * PTZ - Pan/Tilt working, no Zoom option on the webinterface (not even sure if the cam has it, it's not showing in the yi app, so guess not)

Some more info on this.....
- It was not working untill i changed the all.js file to include the r40ga

roleoroleo, 2021-06-09:
Please update with this file:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/179#issuecomment-857771325

About recording, I will check later.
Questions:
1. What network interface are you using?
2. Does your cam have zoom?

bfboarder, 2021-06-09:
> 1. What network interface are you using?
WLAN0 
if needed for testing I can switch to eth0 and use a cable

> 2\. Does your cam have zoom?
I can't find any zoom option in the yi original app, so i think not.........
(the name most have tricked me... who calls a camera PTZ outdoor if it only has PT, must be marketing trick)

bfboarder, 2021-06-09:
On the webinterface not showing files....
Could it have to do with the filename format it is storing it in??
The naming convention seems different if i compare **y21ga** with **r40ga** files, so there is a good chance the js or cgi can't convert it to date time and keeps data == 0, showing the blank page "No events in this folder" 

**y21ga - filename **
21M51S09.mp4

**r40ga - filename**
E222M12S48.mp4

y21ga

roleoroleo, 2021-06-09:
I will check.it.

bfboarder, 2021-06-09:
The "problem" is in this file : https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/www/httpd/cgi-bin/eventsfile.sh
line 21 for loop:
for f in `ls -r /tmp/sd/record/$DIR | grep mp4`; do
    if [ ${#f} == **14** ]; then

it counts the characters and expects 12 but the new names are 14
changing it to 14 works for showing the file, but it makes the date time wrong du to where the code is expecting minutes and seconds

needs some thinkering to make it work on both old and new filenames (12 and 14)

roleoroleo, 2021-06-10:
Your changes are ok, I moved to my source.
Thanks.

bfboarder, 2021-06-11:
Onvif is not starting on this cam, starting it manual like advised here https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/179#issuecomment-858400828 does work

bfboarder, 2021-06-11:
I think there is a small typo in the new system.sh 

line 265
if [[ $MODEL_SUFFIX == "r30gb" ]] || [[ $MODEL_SUFFIX == "r40gb" ]] || [[ $MODEL_SUFFIX == "h5.................................................

Think this shoul be **r40ga** not **r40gb**

bfboarder, 2021-06-11:
With the updated system.sh from https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/179#issuecomment-859405630
this camera seems to have everything working now i think

roleoroleo, 2021-06-11:
You are right.

ABrauser, 2021-06-17:
Thank you for the effort! @bfboarder can you please give me a short description how to get the camera on the same state like yours for some testing? @roleoroleo i also noticed that the "Release Page" Link was removed from the read.me. (I just found it by looking at yi-hack-Allwinner (non-V2) and just change the url. if its to complicated i will just wait for the next release.

roleoroleo, 2021-06-17:
The release page is linked in the right side of the main page.
But a release for this model is not yet available.
Maybe in a few days.

edmundasa, 2022-02-22:
Hi @roleoroleo sorry to post in old thread, but any news on this release?
I bought this camera and old firmware files in this thread do not seem to work for me.
Thanks for your time and energy.

roleoroleo, 2022-02-22:
If your cam is the same model, it's released: r40ga_0.2.0.tgz is your file.

edmundasa, 2022-02-28:
Hi @roleoroleo thanks, I am a moron :D I didn't expand release page so I only saw 4 release versions 🤦 
Thanks for you help.

bogoss95, 2022-08-14:
Hi i need your help, i have the same camera but QFCN (chinese version) imposible use this out of China, web ip:port don't work. Thank for your work and your help

roleoroleo, 2022-08-18:
You need to configure proxychains-ng.

bogoss95, 2022-08-18:
Could give more details ou link to do this with the camera please ?

roleoroleo, 2022-08-18:
This wiki is for a different model but it should work.
https://github.com/roleoroleo/yi-hack-MStar/wiki/Use-proxychains-ng-to-connect-your-app-(thanks-to-@x571)

closed 2023-01-07 by stale[bot]