roleoroleo/yi-hack-Allwinner-v2 #162: Motion detection with Yi Outdoor 1080p IFUS

marioedani, 2021-05-17:
Hi, first of all I would like to say thank you for your hard work.
I just "hacked" a new Outdoor Camera (2021 version) using this hack but I noticed some strage things:

1.
If, in the Configurations page, I leave "Timezone" parameter empty (I live in Italy), the time printed on the video streaming is correct, but MP4 files have wrong date and name (they are 2 hours back).
If, on the contrary, I write the correct Timezone, corresponding to my country ("CET-1CEST,M3.5.0,M10.5.0/3"), the date and the name of the files are correct but the time printed on video streaming is 2 hours forward, and plays recordings with the app doesn't work.

2.
When a motion is detected I receive the MQTT message but withou the names of the MP4 files, even if the files exist, and both with or without the timezone parameter set.

3.
FTP Upload of recordings works only once and only for the first file created ever. I must delete all MP4 files from SD card to make it work again.

4.
When I open the "Motion Events" page and I want to browse the files, I cannot see any file, only folders. When I click on the folder name i receive a "No events in this folder" message. But files are there.

5.
Is it possible to activate/deactivate motion detection via MQTT/HTML? It would be great to stop motion detection when I arrive home...
 

Thank you
Regards

roleoroleo, 2021-05-17:
> If, in the Configurations page, I leave "Timezone" parameter empty (I live in Italy), the time printed on the video streaming is correct, but MP4 files have wrong date and name (they are 2 hours [...]

This is normal because if you don't set timezone, the cam works with GMT hour (2 hours back).
I didn't find a solution to use the right time in both programs.

> When a motion is detected I receive the MQTT message but withou the names of the MP4 files, even if the files exist, and both with or without the timezone parameter set.
 
Please send me an example.

> FTP Upload of recordings works only once and only for the first file created ever. I must delete all MP4 files from SD card to make it work again.

Please post your ftp upload settings.

> When I open the "Motion Events" page and I want to browse the files, I cannot see any file, only folders. When I click on the folder name i receive a "No events in this folder" message. But files are there.

With or without timezone settings?

> Is it possible to activate/deactivate motion detection via MQTT/HTML? It would be great to stop motion detection when I arrive home...

There is no api to deactivate motion detection. You can disable the message sent from the cam.
But It's not simple because you need to remove the topic (leaving it blank) and restart mqtt daemon.
The first configuration can be done through mqtt, the second one using web services (see the wiki).
I could change this method and make it simpler.

marioedani, 2021-05-18:
> > When a motion is detected I receive the MQTT message but withou the names of the MP4 files, even if the files exist, and both with or without the timezone parameter set.
> 
> Please send me an example.
> 
[https://drive.google.com/file/d/1KbDS-W0DmYYB7o_1QsLL_g_gBZRcmPFw/view?usp=sharing](url)
[https://drive.google.com/file/d/1ahawGpEumbclDqcSQJaos3NvFqp6OF57/view?usp=sharing](url)

> > FTP Upload of recordings works only once and only for the first file created ever. I must delete all MP4 files from SD card to make it work again.
> 
> Please post your ftp upload settings.
> 
[https://drive.google.com/file/d/1wwNITfFg4tMpeaNjCWt5ot-swInILBT_/view?usp=sharing](url)
I tried to change the "FTP Delete" setting with no success

> > When I open the "Motion Events" page and I want to browse the files, I cannot see any file, only folders. When I click on the folder name i receive a "No events in this folder" message. But files are there.
> 
> With or without timezone settings?
> 
In both cases

closed 2023-01-08 by stale[bot]