roleoroleo/yi-hack-Allwinner-v2 #163: Wifi configuration over web page on a virgin cam?

WalkingDot, 2021-05-17:
I' m very happy about the newly implemented feature in 0.1.5: "Add wifi configuration to web page".
Because I can't install the app. Thank you so much!!!

But how can I connect to the camera over wifi if no access data for my router has been stored before?

Will the camera then start in access point mode?
Or can I place the "wpa_supplicant" somewhere in the hack folder?

roleoroleo, 2021-05-17:
You should use the official yi app.

WalkingDot, 2021-05-17:
I have only an unsupported windows phone.

roleoroleo, 2021-05-17:
Try this:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/289

WalkingDot, 2021-05-17:
wow... that' s so awesome! Thank you roleoroleo and Arkady23.
I'll try it as soon as the camera arrives.

WalkingDot, 2021-05-21:
It worked great. Thanks again :)

WalkingDot, 2021-05-23:
I reopened this issue because after that the yi home app is unable to pair with the camera.
(to test the sound detection from this issue: https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/168#issuecomment-846449030)

The web interface option "Disable Cloud" is disabled.
And the app tells me to press the reset button.

- is there maybe a way to set the camera into pairing mode over ssh or something else?
- Can I press the button to reset the wifi or is there a risk of a brick?
- Should the sd card be removed before the reset?

roleoroleo, 2021-05-28:
> * is there maybe a way to set the camera into pairing mode over ssh or something else?

No, there isn't.

> * Can I press the button to reset the wifi or is there a risk of a brick?

You can reset the cam without risk of a brick.

> * Should the sd card be removed before the reset?

It's reccomended.

WalkingDot, 2021-06-02:
Thank you, after the reset button the pairing worked fine.