roleoroleo/yi-hack-Allwinner-v2 #168: No Sound Detection

WalkingDot, 2021-05-22:
Model Suffix: r30gb
Firmware Version: 0.1.7
Base Version: 9.0.05.33_202009251856

Sound detection does not work on my camera.
I tested all sensitivity values and for each one I have restarted the cam.

Motion detection is working great, with sound in video.

roleoroleo, 2021-05-22:
Do you receive it in the original app?

WalkingDot, 2021-05-22:
I have setup the camera without the app, because I have no compatible smartphone.
I could ask a friend if it is very important to use the app.

roleoroleo, 2021-05-24:
AFAIK the sound detection feature doesn't work as expected.
Sometimes it detects sound smoothly, other times it doesn't work regardless of the sound volume.
Maybe a hysteresis?

WalkingDot, 2021-05-24:
In my case, it never worked.
Even if I screamed directly in front of the camera.

roleoroleo, 2021-05-24:
You should check if it works with the app.
I need to know if the problem is related to the original app or to my hack.

WalkingDot, 2021-05-24:
Could you help me with this app/cam pairing problem, please?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/163#issuecomment-846583898

Then I can test the app sound detection.
Sorry to waste your time with this.

roleoroleo, 2021-05-28:
Check my post.

WalkingDot, 2021-06-02:
I was now able to test the app.

I enabled in the app sound detection and 50db.
After a short time there were several videos in the alarm menu with the note sound detected.

If I disable the app and restart the camera with sd-card (hack), the sound detection does not work anymore. Even if I clap with my hands or do similar loud things.

I also noticed another strange thing.
After the camera was connected with the app, the cam follows now automatically my movements (PTZ).

roleoroleo, 2021-06-02:
When you enabled sound detection in the app, do you receive the mqtt message?

WalkingDot, 2021-06-02:
I don't have mqtt enabled and unfortunately no server for it
But I received an app message/notification.

roleoroleo, 2021-06-03:
> Edit: Or have I misunderstood something? I think I have never used mqtt.

It depends...
If you don't use mqtt and app, where do you want to receive the detection output?

> Motion detection is working great, with sound in video.

When you write "Motion detection is working great", where do you receive the event?

This is the detection flow:
1. the yi detection process detects the event
2. the yi detection process sends the event to the yi communication process
3. the yi communication process sends the event to the yi app
4. mqttv4 process, added with the hack, captures the event (between processes in the 2nd step)
5. mqttv4 process sends the event to the mqtt server

WalkingDot, 2021-06-03:
Now I realize that i have misinterpreted these two functions "Save video when a motion is detected" and "Sound Detection".

I thought when a recording is started (tmp file), a successful motion or sound detection has been recognized. This was my simple way to test the functions ;-)
And I'm so sorry that I have wasted your time and nerves with this misunderstanding.

The app has recordings for both motion and sound detection.
Could you maybe add the function "Save video when a sound is detected" to your hack, please?

roleoroleo, 2021-06-03:
Don't apologize, you're not completely wrong. 
The settings you find in the page "Camera Settings" are a clone of the settings you find in the app.
When you check/unchek a setting in this page, the hack sends to the cam a message like the original app would.
Try to toggle a check and update the app, you will see the configuration change.

So the feature "Save video when a motion is detected" is 
[image]
(I don't know how it's in your language).

The feature "Sound Detection" is 
[image]

There isn't "Motion Detection" because I haven't been able to detect a message when you check this box.

If the check "Save video when a motion is detected" is off video will be always saved but if it's on video will be saved only after a detection. So you can test the detection function checking the tmp file only if this option is on.

Now, try to disable this option and check if a video is created after a motion or a sound detection.

WalkingDot, 2021-06-03:
Great, I will test it as soon as possible (2-3 days).

During my last test, 6 second videos were saved on sound detections in the yi-cloud.
The same with motion detections.

If I'm repeating this, the hack has to be on the SD-card inside the cam, right?
Because I need the ftp server to check the tmp file.

WalkingDot, 2021-06-05:
> Now, try to enable this option and check if a video is created after a motion or a sound detection.

yay... the app recording is working:
[image]

roleoroleo, 2021-06-07:
> Edit: But if I start the cam with the hack, only motion detections are recorded. Unfortunately, sound detections are not recorded.

I will check it.

WalkingDot, 2021-06-11:
Is it normal that there are so many mtdblock7 files in the main directory?
```
drwxr-xr-x 1 root root     0 Jun  8 11:56 .
drwxrwxrwx 1 root root     0 Jun 11  2021 ..
drwxr-xr-x 1 root root     0 Mai 22 00:36 Factory
drwxr-xr-x 1 root root     0 Jun  8 11:40 log
-rwxr-xr-x 1 root root  1747 Apr 13 14:24 lower_half_init.sh
-rwxr-xr-x 1 root root 65536 Jan 19  2016 mtdblock7_20160119064751.bin
-rwxr-xr-x 1 root root 65536 Apr 17  2020 mtdblock7_20200417223044.bin
-rwxr-xr-x 1 root root 65536 Apr 17  2020 mtdblock7_20200417225444.bin
-rwxr-xr-x 1 root root 65536 Mai 22 23:16 mtdblock7_20210522221647.bin
-rwxr-xr-x 1 root root 65536 Mai 23 00:10 mtdblock7_20210522231052.bin
-rwxr-xr-x 1 root root 65536 Mai 23 01:48 mtdblock7_20210523004832.bin
-rwxr-xr-x 1 root root 65536 Mai 23 02:06 mtdblock7_20210523010605.bin
-rwxr-xr-x 1 root root 65536 Mai 27 19:29 mtdblock7_20210527182907.bin
-rwxr-xr-x 1 root root 65536 Jun  4 22:01 mtdblock7_20210604210059.bin
-rwxr-xr-x 1 root root 65536 Jun  4 22:20 mtdblock7_20210604212008.bin
-rwxr-xr-x 1 root root 65536 Jun  4 22:23 mtdblock7_20210604212343.bin
-rwxr-xr-x 1 root root 65536 Jun  4 23:02 mtdblock7_20210604220230.bin
-rwxr-xr-x 1 root root 65536 Jun  5 19:14 mtdblock7_20210605181458.bin
-rwxr-xr-x 1 root root 65536 Jun  5 19:42 mtdblock7_20210605184208.bin
-rwxr-xr-x 1 root root 65536 Jun  5 19:44 mtdblock7_20210605184409.bin
-rwxr-xr-x 1 root root 65536 Jun  5 23:36 mtdblock7_20210605223610.bin
drwxr-xr-x 1 root root     0 Apr 13 14:24 newbackup
drwxr-xr-x 1 root root     0 Jun 11  2021 record
drwxr-xr-x 1 root root     0 Mai  2 23:35 yi-hack
```

roleoroleo, 2021-06-11:
> Is it normal that there are so many mtdblock7 files in the main directory?

It's normal if you used the procedure to change the wifi credentials.
It's a backup of the partition.

> And if I insert a microsd-card without the hack, motion and sound detections are recorded on the microsd-card.

This is strange, but I didn't find the reason.

WalkingDot, 2021-06-11:
Ah.. ok, every time I used the app, I also changed the wifi. Now it's clear... thanks ;-)

Could it help if I remove the hack completely and set up everything again?

WalkingDot, 2021-06-12:
I have an idea for a workaround.

You have explained the detection flow to me:
> 1. the yi detection process detects the event
>
>     2. the yi detection process sends the event to the yi communication process
>
>     3. the yi communication process sends the event to the yi app [...]

Is it at step 4. possible to execute a command on sound detection?
e.g.:
`ipc_cmd -S 60 &`

roleoroleo, 2021-06-12:
Yes, but before this change I need to understand why sound events are not recorded.

WalkingDot, 2021-06-15:
I also have a new R30GB with 64 MB memory (https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/175#issuecomment-860237090).

```
cat /proc/meminfo
MemTotal:          60908 kB
MemFree:             824 kB
MemAvailable:      13588 kB
```

Could this be the cause of the problem?

WalkingDot, 2021-06-16:
I'm trying to find a workaround. So I have tried to start mqttv4 over ssh, to get stdout messages.
But i get a bunch of errors and don't understand them:

```
Error loading shared library libmosquitto.so.1: No such file or directory (needed by ./mqttv4)
Error relocating ./mqttv4: mosquitto_connack_string: symbol not found
Error relocating ./mqttv4: mosquitto_publish: symbol not found
Error relocating ./mqttv4: mosquitto_publish_callback_set: symbol not found
Error relocating ./mqttv4: mosquitto_connect_callback_set: symbol not found
Error relocating ./mqttv4: mosquitto_log_callback_set: symbol not found
Error relocating ./mqttv4: mosquitto_username_pw_set: symbol not found
Error relocating ./mqttv4: mosquitto_lib_cleanup: symbol not found
Error relocating ./mqttv4: mosquitto_new: symbol not found
Error relocating ./mqttv4: mosquitto_connect: symbol not found
Error relocating ./mqttv4: mosquitto_strerror: symbol not found
Error relocating ./mqttv4: mosquitto_disconnect_callback_set: symbol not found
Error relocating ./mqttv4: mosquitto_destroy: symbol not found
Error relocating ./mqttv4: mosquitto_loop: symbol not found
Error relocating ./mqttv4: mosquitto_will_set: symbol not found
Error relocating ./mqttv4: mosquitto_lib_init: symbol not found
```

Is it generally possible to receive a locale output (stdout or log) of mqttv4, to check the events with a script?
So I could check the output maybe with regexp and start manual recording.

roleoroleo, 2021-06-16:
You have to source the environment before starting mqttv4.
`. /tmp/sd/yi-hack/script/env.sh`

WalkingDot, 2021-06-16:
I have entered:
`/tmp/sd/yi-hack/script/env.sh`

and after that:
`/tmp/sd/yi-hack/bin/mqttv4`

But I'm receiving the same errors.

WalkingDot, 2021-06-16:
Ah... ok, thanks! I got it.
Is there maybe a start paramter/option to disable the connection attempts? I left the ip "0.0.0.0".
```
Trying to connect... Unable to connect (Connection refused).
Unable to connect (Connection refused).
```

roleoroleo, 2021-06-16:
Try to run ipc_sniff
[ipc_sniff.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6665498/ipc_sniff.gz)

This binary detects the ipc messages between yi processes.
You should see a numeric code when a event is detected.

roleoroleo, 2021-06-16:
> Is there maybe a start paramter/option to disable the connection attempts? I left the ip "0.0.0.0".

The ip address of your mqtt server.

WalkingDot, 2021-06-17:
Thank you so much! Your sniffer is awesome.
Sound is detectable:
`04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00`

Now i have all necessary for a workaround, tthanks ;-)

roleoroleo, 2021-06-17:
This code is already detected inside ipc_multiplexer.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/ipc_cmd/ipc_cmd/ipc_multiplexer.h
It should work.

WalkingDot, 2021-06-17:
Output from ipc_multiplexer:
```
IPC message. Len: 16. Status: No error information!
Parsing message.
04 00 00 00 02 00 00 00 04 60 04 60 00 00 00 00 
Sending SOUND_DETECTION to queue #1
Sending SOUND_DETECTION to queue #2
Sending SOUND_DETECTION to queue #3
Sending SOUND_DETECTION to queue #4
Sending SOUND_DETECTION to queue #5
Sending SOUND_DETECTION to queue #6
Sending SOUND_DETECTION to queue #7
Sending SOUND_DETECTION to queue #8
Sending SOUND_DETECTION to queue #9
IPC message. Len: 16. Status: Resource temporarily unavailable!
Parsing message.
```

roleoroleo, 2021-06-17:
It's ok.
Now mqttv4 should read it from the queue and send to the mqtt broker.

WalkingDot, 2021-06-23:
Sorry... one last thing before I start with the workaround.
Can you confirm these codes to enable/disable sd-card recording?
```
02 00 00 00 08 00 00 00 79 00 01 00 00 00 00 00 
02 00 00 00 08 00 00 00 78 00 01 00 00 00 00 00 
```

WalkingDot, 2021-07-01:
The workaround is running stable since a few days. So I thought to publish it, in case someone should have the same problem.

[Sound Event Recording Workaround for YI Hack Allwinner-v2](https://github.com/WalkingDot/yi-hack-awv2-recsoundevent-workaround)

Just copy the files into the following locations and restart the cam:
`/tmp/sd/yi-hack/startup.sh`
`/tmp/sd/yi-hack/bin/recsoundevent`

Optional startup arguments can be added:
**-t**  : Recording duration in minutes (1-60, default: 10). 
**-d**  : dB sensitivity to trigger recording (30-90, default: Cam dB setting).

roleoroleo, 2021-07-03:
Thank you for your code.
I'm still trying to to understand why sound events are not recorded, but no luck at the moment.

closed 2023-01-08 by stale[bot]