roleoroleo/yi-hack-Allwinner-v2 #179: Request: Yi Dome U Pro Support

seventhplug, 2021-06-01:
Hi I bought a Yi Dome U Pro (BLACK) Camera i thought it would be supported. I hope you can please add this camera to your supported device.. Thank you.  It has a Device ID: LFUS

roleoroleo, 2021-06-02:
I need to know the firmware version.
You can check it in the app.

seventhplug, 2021-06-03:
Hi its. 9.0.27.19_202105131533

roleoroleo, 2021-06-03:
Actually it's not supported.
If you want to help me to add it to the supported devices, please make a backup copy and share it with me.
Check the wiki.

seventhplug, 2021-06-03:
Ok [https://www.mediafire.com/file/mike0vhzi2jmh7v/backup.rar/file](url)
Thank you ^_^

roleoroleo, 2021-06-03:
Your cam is h60ga.
At the moment it's not supported, I will check it.

seventhplug, 2021-06-04:
ok, thank you ^_^ appreciate it.

roleoroleo, 2021-06-04:
Here it is a first alpha version:
[h60ga_0.1.7.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6597827/h60ga_0.1.7.tar.gz)

Probably rtsp and snapshot don't work.
Install it and check if ssh and web work.

seventhplug, 2021-06-06:
Wow that was fast.. anyway before trying this.. is there a way to undo the hack or restore the backup i created?

roleoroleo, 2021-06-06:
If you brick "softly" the cam you can restore the previous state with unhack procedure (I will create a new unhack procedure for your cam).
If the brick is "hard" you need a serial connection and the backup files to restore the cam (check the wiki).

seventhplug, 2021-06-07:
I already ordered my second Yi Dome U Pro. I'll wait for you unhack procedure before trying.. tnx

roleoroleo, 2021-06-07:
Here it is:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Remove-the-hack

seventhplug, 2021-06-08:
Hi It worked. I will test it further. Thank you for your hardwork
[image]

roleoroleo, 2021-06-08:
Ok, now I need 3 copies of your video buffer.
Copy the file /dev/shm/fshare_frame_buf, wait 5 seconds, copy it again, wait 5 seconds and copy it again.
Zip the files and share the archive with me.
I will try to enable rtsp and snapshot.

seventhplug, 2021-06-08:
Hello.. Here is the file <a href="http://www.filehost.pt/1qr96~i?9b0520bf3f4000655b203418f611ad71" target="_blank" title="Download From File Host PT">download fshare_frame_buf.rar from File Host PT</a> 

i dont know why my for loop wont work Lol.. so i manually used this command

cp /dev/shm/fshare_frame_buf fshare_frame_buf0
sleep 5
cp /dev/shm/fshare_frame_buf fshare_frame_buf1
sleep 5
cp /dev/shm/fshare_frame_buf fshare_frame_buf2
sleep 5
cp /dev/shm/fshare_frame_buf fshare_frame_buf3
sleep 5
cp /dev/shm/fshare_frame_buf fshare_frame_buf4

anyway may i request another enhancement? like DARK Mode Web interface and the ability to add other wifi 
networks in wpa_supplicant via web interface. Because i have a multiple AP so if one disconnects it will automatically 
connect to the next AP available. Thank you for your hardwork.

roleoroleo, 2021-06-08:
> anyway may i request another enhancement? like DARK Mode Web interface

Yes, I can add it.

> and the ability to add other wifi
> networks in wpa_supplicant via web interface. Because i have a multiple AP so if one disconnects it will automatically
> connect to the next AP available. Thank you for your hardwork.

This is not possible.
Wifi credentials management is not in my hack. Yi is responsible for it.

seventhplug, 2021-06-09:
Sorry here's the new link https://www.mediafire.com/file/n5f1mxejik9us18/fshare_frame_buf.rar/file

roleoroleo, 2021-06-09:
Done.

roleoroleo, 2021-06-09:
Overwrite the files in /tmp/sd/yi-hack/bin with the content of this archive:
[new.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6622423/new.tar.gz)
and check if rtsp and snapshot work.

seventhplug, 2021-06-09:
Thank you again!  rtsp and snapshot are now working. 

FTP - Working
Legacy FTP - Not Tested
Telnet - Working
SSH - Working
Web Interface - Working
PTZ Not Working -- website interface says "PTZ functions are not available for this camera model"
RTSP -Working High and Low Stream (a bit choppy on High Stream)  i guess its hardware limitations
ONVIF -- Not tested
Snapshot - Working High/Low Res
MQTT -- Not tested

Swap File Enabled

seventhplug, 2021-06-09:
[image]

I tried ONVIF using ONVIF Device Manager from sourceforge (https://sourceforge.net/projects/onvifdm/) i set my camera with no user/password. Its not connecting..

seventhplug, 2021-06-09:
Hi.. Test Results are
PTZ - PTZ functions are not available for this camera model.
Onvif - wsdd is running but not onvif_srvd

[image]

roleoroleo, 2021-06-09:
Try to update the cache of your browser. PTZ should apper.

About onvif_srvd, try to run it manually:
`onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num BFUSY44R6RH41Y200807 --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi-bin/snapshot.sh?res=low&watermark=yes --type H264`

seventhplug, 2021-06-09:
clearing cache did the trick.. PTZ now working

here is the output on onvif

```
root@Gavin:~# onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 -
-hardware_id TEST --serial_num XXX --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --
width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi-bin/snapshot.sh?res=low&watermark=yes
 --type H264
-ash: --type: not found
root@Gavin:~# -ash: onvif_srvd: not found

```

seventhplug, 2021-06-09:
Hi clearing cache did the trick.. PTZ now working

but the onvif there is the output

`login as: root

BusyBox v1.27.2 () built-in shell (ash)

mount: mounting none on /sys/kernel/debug failed: Resource busy

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@Gavin:~# onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num XXX --ifs wlan0 -
-port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi-bin/snapshot.sh?r
es=low&watermark=yes --type H264
-ash: --type: not found
root@Gavin:~# -ash: onvif_srvd: not found
root@Gavin:~# cd /tmp/sd/yi-hack/bin
[1]+  Done(127)                  onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num XXX --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi-bin/snapshot.sh?res=low
root@Gavin:/tmp/sd/yi-hack/bin# ls
base64           h264grabber_h    mosquitto_pub    rRTSPServer
dbclient         h264grabber_l    mosquitto_sub    rmdir
dropbearconvert  hostname         mqtt-config      scp
dropbearkey      imggrabber       mqttv4           speaker
dropbearmulti    ipc_cmd          netstat          ssh
framefinder      ipc_multiplexer  onvif_srvd       tar
gzip             ipc_read         proccgi          wsdd
h264grabber      jq               proxychains4
root@Gavin:/tmp/sd/yi-hack/bin# onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num
 XXX --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi
-bin/snapshot.sh?res=low&watermark=yes --type H264
-ash: --type: not found
root@Gavin:/tmp/sd/yi-hack/bin# -ash: onvif_srvd: not found
root@Gavin:/tmp/sd/yi-hack/bin# ./onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_n
um XXX --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/c
gi-bin/snapshot.sh?res=low&watermark=yes --type H264
-ash: --type: not found
[1]-  Done(127)                  onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num XXX --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi-bin/snapshot.sh?res=low
root@Gavin:/tmp/sd/yi-hack/bin# onvif_srvd: Error: not set no one profile more details see --help

[2]+  Done                       ./onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num XXX --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 640 --height 360 --url rtsp://%s/ch0_1.h264 --snapurl http://%s:8080/cgi-bin/snapshot.sh?res=low
root@Gavin:/tmp/sd/yi-hack/bin#
`

roleoroleo, 2021-06-10:
Try again with this:
`/tmp/sd/yi-hack/bin/onvif_srvd --pid_file /var/run/onvif_srvd.pid --model "Yi Hack" --manufacturer Yi --firmware_ver 0.1.7 --hardware_id TEST --serial_num XXX --ifs wlan0 --port 80 --scope "onvif://www.onvif.org/Profile/S" --name Profile_0 --width 640 --height 360 --url "rtsp://%s/ch0_1.h264" --snapurl "http://%s:8080/cgi-bin/snapshot.sh?res=low&watermark=yes" --type H264`

kill the previous daemin before running it:
`killall onvif_srvd`

seventhplug, 2021-06-10:
It worked.. onvif is now working on the Yi Dome U Pro.. 

but i tried it on my Yi Home 1080p y211ga onvif did not work.. should i file a new issue for this model?

roleoroleo, 2021-06-11:
This is strange.
The command I wrote above is the same that system.sh runs during the boot.

seventhplug, 2021-06-11:
When i issued "killall onvif_srvd" it says that onvif_srvd is not running.  I guess it doesnt automatically starts even if its turned on the  web interface.  I got the same error with my y211ga before this fix

roleoroleo, 2021-06-11:
Try these commands and post the output:
```
dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null | cut -c1-20
dd bs=1 count=4 skip=656 if=/tmp/mmap.info 2>/dev/null | cut -c1-4
```

And try if this works:
```
./onvif_srvd --pid_file /var/run/onvif_srvd.pid --model "Yi Hack" --manufacturer "Yi" --firmware_ver "0.1.7" --hardware_id XXX --serial_num SN --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 1920 --height 1080 --url "rtsp://%s/ch0_0.h264" --snapurl "http://%s:8080/cgi-bin/snapshot.sh?res=high&watermark=yes" --type H264 --name Profile_1 --width 640 --height 360 --url "rtsp://%s/ch0_1.h264" --snapurl "http://%s:8080/cgi-bin/snapshot.sh?res=low&watermark=yes" --type H264 --ptz --move_left "/tmp/sd/yi-hack/bin/ipc_cmd -M left" --move_right "/tmp/sd/yi-hack/bin/ipc_cmd -M right" --move_up "/tmp/sd/yi-hack/bin/ipc_cmd -M up" --move_down "/tmp/sd/yi-hack/bin/ipc_cmd -M down" --move_stop "/tmp/sd/yi-hack/bin/ipc_cmd -M stop" --move_preset "/tmp/sd/yi-hack/bin/ipc_cmd -p %t"
```

seventhplug, 2021-06-11:
Ok I'll try this when I get home. After 7hrs i'll post the output. Thanks

bfboarder, 2021-06-11:
Not sure if this helps, but i did try this on r40ga

> And try if this works:
> 
> ```
> ./onvif_srvd --pid_file /var/run/onvif_srvd.pid --model "Yi Hack" --manufacturer "Yi" --firm
> ```

For me (on a different cam (r40ga) this gives an error on the -- type H at the end (Can't set type for Profile: type dont support)
removing --type H and it works

roleoroleo, 2021-06-11:
> Not sure if this helps, but i did try this on r40ga

Yes, it's the same.

It should be --type H264 and not --type H
Please try this new system.sh:
[system.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6636803/system.sh.gz)

Copy it in /tmp/sd/yi-hack/script

roleoroleo, 2021-06-11:
Ok, now it's clear.
I reached the max size of the command line.
I need to change the method to pass parameters.

bfboarder, 2021-06-11:
> Ok, now it's clear.
> I reached the max size of the command line.
> I need to change the method to pass parameters.

Ahh.... I tried to change it to H264 in the command line, now i know why it would not type any more characters :)

bfboarder, 2021-06-11:
This should fix it (at least it's working for me now)

[system.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6637149/system.zip)

fix typo for r40ga
fix command line to long

seventhplug, 2021-06-11:
> Try these commands and post the output:
>
> ```
> dd bs=1 count=20 skip=656 if=/tmp/mmap.info 2>/dev/null | cut -c1-20
> dd bs=1 count=4 skip=656 if=/tmp/mmap.info 2>/dev/null | cut -c1-4
> ```
>
> And try if [...]

Here is the output 
[image]

Yep the Code Works
[image]<br /><a target='_blank' href='https://geojsonlint.com/'>mapbox add geojson</a><br />

seventhplug, 2021-06-11:
> > Not sure if this helps, but i did try this on r40ga
> 
> Yes, it's the same.
> 
> It should be --type H264 and not --type H
> Please try this new system.sh:
> [system.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6636803/system.sh.gz)
> 
> Copy it in /tmp/sd/yi-hack/script

onvif works upon reboot now

roleoroleo, 2021-06-11:
Please send me a copy of you mmap.info.

And try these new files: onvif_srvd now reads configuration from a file.
[upd_1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6639039/upd_1.tar.gz)

seventhplug, 2021-06-11:
Onvid not working again upon booting

Please rename
[mmap.info.txt](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6639140/mmap.info.txt)

seventhplug, 2021-06-11:
> Please send me a copy of you mmap.info.
> 
> And try these new files: onvif_srvd now reads configuration from a file.
> [upd_1.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6639047/upd_1.tar.gz)
> 
> With r40gb typo fixed.

i tried this and onvid did not work again.. 

> > Not sure if this helps, but i did try this on r40ga
> 
> Yes, it's the same.
> 
> It should be --type H264 and not --type H
> Please try this new system.sh:
> [system.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6636803/system.sh.gz)
> 
> Copy it in /tmp/sd/yi-hack/script
i revert to this system.sh and it worked again upon bootup

roleoroleo, 2021-06-11:
Are you using authentication?

seventhplug, 2021-06-11:
No authentication

SmartM-ui, 2021-06-12:
hi, 
which sensors can the integration for the Dome U Pro camera import into the home assistant?  
Does the motion sensor, face detection and the ability to put it in privacy mode appear?
Thanks

roleoroleo, 2021-06-12:
I compared again upd_1.tar.gz and system.sh.gz.
upd_1.tar.gz must work.
Did you copy both files: onvif_srvd and system.sh?

seventhplug, 2021-06-13:
Hi. Yes i copied both files

here is the output. I set rtsp (High) stream and onvif (High) Profile
```
login as: root

BusyBox v1.27.2 () built-in shell (ash)

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@Gavin:~# /tmp/sd/yi-hack/bin/onvif_srvd --conf_file=/tmp/onvif_srvd.conf
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::at: __n (which is 0) >= this->size() (which is 0)
Aborted
root@Gavin:~# cat /tmp/onvif_srvd.conf
pid_file=/var/run/onvif_srvd.pid
model=Yi Hack
manufacturer=Yi
firmware_ver=0.1.7
hardware_id=
serial_num=
ifs=wlan0
port=80
scope=onvif://www.onvif.org/Profile/S

#Profile 0
name=Profile_0
width=1920
height=1080
url=rtsp://%s/ch0_0.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=high
type=H264

#PTZ
ptz=1
move_left=/tmp/sd/yi-hack/bin/ipc_cmd -M left
move_right=/tmp/sd/yi-hack/bin/ipc_cmd -M right
move_up=/tmp/sd/yi-hack/bin/ipc_cmd -M up
move_down=/tmp/sd/yi-hack/bin/ipc_cmd -M down
move_stop=/tmp/sd/yi-hack/bin/ipc_cmd -M stop
move_preset=/tmp/sd/yi-hack/bin/ipc_cmd -p %t
root@Gavin:~#
```

and here is the output i set both stream on rtsp and both profile on onvif 

```

login as: root

BusyBox v1.27.2 () built-in shell (ash)

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@Gavin:~# /tmp/sd/yi-hack/bin/onvif_srvd --conf_file=/tmp/onvif_srvd.conf
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::at: __n (which is 0) >= this->size() (which is 0)
Aborted
root@Gavin:~# cat /tmp/onvif_srvd.conf
pid_file=/var/run/onvif_srvd.pid
model=Yi Hack
manufacturer=Yi
firmware_ver=0.1.7
hardware_id=
serial_num=
ifs=wlan0
port=80
scope=onvif://www.onvif.org/Profile/S

#Profile 0
name=Profile_0
width=1920
height=1080
url=rtsp://%s/ch0_0.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=high
type=H264

#Profile 1
name=Profile_1
width=640
height=360
url=rtsp://%s/ch0_1.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=low
type=H264

#PTZ
ptz=1
move_left=/tmp/sd/yi-hack/bin/ipc_cmd -M left
move_right=/tmp/sd/yi-hack/bin/ipc_cmd -M right
move_up=/tmp/sd/yi-hack/bin/ipc_cmd -M up
move_down=/tmp/sd/yi-hack/bin/ipc_cmd -M down
move_stop=/tmp/sd/yi-hack/bin/ipc_cmd -M stop
move_preset=/tmp/sd/yi-hack/bin/ipc_cmd -p %t
root@Gavin:~#

```

roleoroleo, 2021-06-13:
Maybe empty vars are the problem.
Try to fill these vars:
```
hardware_id=
serial_num=
```
and try this command:
```
dd bs=1 count=20 skip=784 if=/tmp/mmap.info 2>/dev/null | cut -c1-20
```

roleoroleo, 2021-06-13:
Ok, try this:
[onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6644521/onvif_srvd.gz)

seventhplug, 2021-06-14:
Hello, I did not try this.. 
> Maybe empty vars are the problem.
> Try to fill these vars:
> 
> ```
> hardware_id=
> serial_num=
> ```
> 
> and try this command:
> 
> ```
> dd bs=1 count=20 skip=784 if=/tmp/mmap.info 2>/dev/null | cut -c1-20
> ```

I skipped ahead and tried this instead.. now onvif works.. 

> Ok, try this:
> [onvif_srvd.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6644521/onvif_srvd.gz)

and it do its boot sequence  (rotating and blinking) 2-3 times before actually starting up after that its working

seventhplug, 2021-06-14:
Thank you again.. here are the test results

FTP - Working
Legacy FTP - Not Tested
Telnet - Working
SSH - Working
Web Interface - Working
PTZ - WORKING
RTSP -Working High and Low Stream (a bit choppy on High Stream) i guess its hardware limitations
ONVIF - WORKING
Snapshot - Working High/Low Res
MQTT -- Not tested

Swap File Enabled

I dont have any IoT knowledge at the moment so i dont know how to test MQTT sorry.
Native resolution is 2304 x 1296 but downscaled to 1920x1080p i like it better because it looks smoother.

roleoroleo, 2021-06-14:
Is ptz working in the right direction?
Is it inverted?

mqtt is very useful if you want to integrate your cam in your domotics system.
But not everyone uses it.

Thank you for your support.

seventhplug, 2021-06-14:
On PTZ Left and Right is good. Up and Down is inverted

fields are still empty  but onvif is working 

```
BusyBox v1.27.2 () built-in shell (ash)

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@Gavin:~# /tmp/sd/yi-hack/bin/onvif_srvd --conf_file=/tmp/onvif_srvd.conf
root@Gavin:~# Error 28 fault detected[no subcode]
"Address in use"
Detail: bind failed in soap_bind()
root@Gavin:~# cat /tmp/onvif_srvd.conf
pid_file=/var/run/onvif_srvd.pid
model=Yi Hack
manufacturer=Yi
firmware_ver=0.1.7
hardware_id=
serial_num=
ifs=wlan0
port=80
scope=onvif://www.onvif.org/Profile/S

#Profile 0
name=Profile_0
width=1920
height=1080
url=rtsp://%s/ch0_0.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=high
type=H264

#PTZ
ptz=1
move_left=/tmp/sd/yi-hack/bin/ipc_cmd -M left
move_right=/tmp/sd/yi-hack/bin/ipc_cmd -M right
move_up=/tmp/sd/yi-hack/bin/ipc_cmd -M up
move_down=/tmp/sd/yi-hack/bin/ipc_cmd -M down
move_stop=/tmp/sd/yi-hack/bin/ipc_cmd -M stop
move_preset=/tmp/sd/yi-hack/bin/ipc_cmd -p %t
root@Gavin:~#
```

Thank you for supporting this device ^_^ i'll make time to study mqtt for my next pi iot project

helmbrandt, 2021-06-14:
I have the same Cam for testing here, LFUS on the sticker, can test things too

seventhplug, 2021-06-14:
> I have the same Cam for testing here, LFUS on the sticker, can test things too

Yes please help us test... More tester the better 😁

roleoroleo, 2021-06-14:
> On PTZ Left and Right is good. Up and Down is inverted

Ok, I will fix it

> fields are still empty but onvif is working

Run this command and post the output:
`dd bs=1 count=20 skip=784 if=/tmp/mmap.info 2>/dev/null | cut -c1-20`

seventhplug, 2021-06-14:
Hi, here's the output ^_^ Don't forget the Dark Mode on your final release. Thank you again and again 

```
login as: root

BusyBox v1.27.2 () built-in shell (ash)

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@Gavin:~# dd bs=1 count=20 skip=784 if=/tmp/mmap.info 2>/dev/null | cut -c1-
20
LFUSY60P6QLD9P210126
root@Gavin:~#
```

SmartM-ui, 2021-06-14:
Hi @roleoroleo ,
could you tell me if these functions are already tested and working for the Dome U Pro or if they still need to be tested using the MQTT protocol:
Camera on / off (privacy mode)
AI human detection
Motion detection

If I want to buy the same model (so as to do it also as a beta tester), how can I know in advance the firmware version and the prefix file if I buy through the online site?

Thank you

roleoroleo, 2021-06-14:
> Hi @roleoroleo ,
> could you tell me if these functions are already tested and working for the Dome U Pro or if they still need to be tested using the MQTT protocol:
> Camera on / off (privacy mode)
> AI human detection
> Motion detection

Actually not tested but probably working.
When I add a new supported model, normally it's not necessary to customize these features.

> If I want to buy the same model (so as to do it also as a beta tester), how can I know in advance the firmware version and the prefix file if I buy through the online site?

It's not possible.

roleoroleo, 2021-06-14:
The offset 784 is ok.
The new system.sh should work.
See the last commit.

Dark mode is already commited.

seventhplug, 2021-06-14:
> @seventhplug
> The offset 784 is ok.
> The new system.sh should work.
> See the last commit.
> 
> Dark mode is already commited.

Thank you very much

helmbrandt, 2021-06-14:
will this be compiled and added to the releases?

[54 more replies]

closed 2023-01-08 by stale[bot]