roleoroleo/yi-hack-Allwinner-v2 #180: no S/N and forgotten wifi after unplug power

MATN92, 2021-06-02:
i've got problem with my camera. I bought it from a seller who said that after updating a  firmware camera 'bricked' [according him - only speaks Chinesse and cant add in Yi Home app or Chinesse version]. after install hack it works again, but it cant be visible S/N and also I noticed that after disconnection with power forgets wifi ssid and password [after power reset i need scan QR code from app]. Is there any possibility to add wpa_suplicant with this hack or update firmware for working YI cloud - i preferred wpa_suplicant :)
[image]

DID: BFUSY446G8QKMM200807
S/N: BY21GAEUW09027706
processor:QG2101A

[image]

roleoroleo, 2021-06-02:
I will check it.

roleoroleo, 2021-06-02:
Could you please make a copy of your /backup folder?

MATN92, 2021-06-02:
here is backup before hack
[Archive.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6587277/Archive.zip)

roleoroleo, 2021-06-03:
I checked your files and all is ok.
Send me a copy of your /tmp/mmap.info
Pay attention to the content, hide your personal data but don't change the binary structure.

MATN92, 2021-06-03:
where i can find nmap.info?

roleoroleo, 2021-06-03:
In /tmp

MATN92, 2021-06-03:
i cant find folder tmp or i havent got it... can i acess via ssh?
[image]

roleoroleo, 2021-06-03:
Go up.
`cd /tmp`

MATN92, 2021-06-06:
I might be doing something wrong, but here what i got:
^P^@^@^@^P^@^@^@^@^@^@^@https://api.eu.xiaoyi.com^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^
^@^@^@^@^@^@^@.^G^@^@}^B^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
^@^@^@^A^@^@^@^@^@^@^@.Q^A^@^@^@^@^@^@^@^@^@^@^@^@^BP^@^@^@^P'^@^@^@^@^@^@02544d53413136475548a9d3d80149b1
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^A^@^@^@^@^@^@^@^B^@^@^@^@^A^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@
~
~
~
~
~
~

MATN92, 2021-06-06:
i checked wpa_supplicant.conf and seems good, but why forgoted every time when power reset?
[image]

roleoroleo, 2021-06-06:
wpa_supplicant is a temporary file created when system boots up.
You can't edit it.
Please, send me mmap.info zipped as a binary file.

MATN92, 2021-06-06:
[mmap.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6604672/mmap.zip)
he it is...

roleoroleo, 2021-06-07:
There is no serial number in your mmap.info
I don't know why...

MATN92, 2021-06-07:
for now i use https://github.com/roleoroleo/yi-hack-Allwinner/issues/289 and its working, but i think is bricked sw - i cant find newer firmware from producent for y21ga [this model]

roleoroleo, 2021-06-07:
rootfs, home and backup partitions are ok.
Did you try to reset the cam with the button?

MATN92, 2021-06-07:
many time... every time after reset talked chinese and cant connected via Yi Home app Global or Chinesse [connected WiFi, but not YI cloud etc]... i think missing S/N is the first problem on this device...

roleoroleo, 2021-06-07:
We could try to overwrite the partition where the cam stores sn.

MATN92, 2021-06-07:
ok thanks

roleoroleo, 2021-06-07:
You could try to overwrite mtd7 with this file:
[mtdblock7.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/6611767/mtdblock7.zip)

But only if you can unbrick the cam with a serial connection because it's a risky operation.

uklatt, 2022-03-04:
I had the same problem.  Look here: https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/383
I took mtdblock6 and mtdblock7 from an international camera and wrote it to the chinese camera.
Both models are y21ga!
Now the camera speaks english and works with yi-hack without problems ;-)

Uwe

closed 2023-01-08 by stale[bot]