roleoroleo/yi-hack-Allwinner-v2 #181: RTSP Audio

tunerooster, 2021-06-03:
I am trying to get audio to work with https://github.com/AlexxIT/WebRTC.
`AAC` audio does not work with `WebRTC` (only PCM) and `MSE protocol` (with AAC) is so choppy it is unusable.

I see there is an PCM option in the Configurations page for RTSP audio, but it does not seem to do anything (or am I missing something)?

_I would like to know if it is possible to get PCM audio in the RTSP stream, so that it might be a solution for using WebRTC with audio via yi-hack RTSP._

I am using a `Yi Outdoor 1080p (h30ga)` release.

Thanks!

roleoroleo, 2021-06-03:
alaw and ulaw are inside the PCM family.
They are different from linear PCM but work with WebRTC.
You can use them.

tunerooster, 2021-06-04:
Outstanding!  They works great!  Can't really tell any difference between alaw and ulaw.  No need to reply, but if you, or anyone, have any wisdom about which to use, I would be happy to hear it.  In the mean time, I am using ulaw...

Thanks!!!

closed 2021-06-04 by tunerooster

roleoroleo, 2021-06-04:
Both algorithms are implemented in telephony systems
Ulaw is currently being used by companies in North America and in Japan while alaw is being used in Europe.
Hard to say which is the best.

tunerooster, 2021-06-05:
Thanks for your reply!