roleoroleo/yi-hack-Allwinner-v2 #187: any way to off/on camera from home assitant?

wild-run, 2021-06-08:
any way to off/on camera from home assitant?

roleoroleo, 2021-06-08:
Yes it's work in progress.
But it doesn't really turn on/off the cam.
Only enable/disable services...
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/ac91d7debe5c74b9f79377edad37b3f3e3f8c2ca
But it's enough to add a switch to the ha custom integration.

shoaei, 2021-06-10:
In the meantime, you can use MQTT integration for turning on/off the cam (services) and many other settings. Here is how I did it:
1- Install a MQTT Broker / Server: You can install Mosquitto broker in HA add-on store or use your own server
2- Configure HA MQTT Integration (HA > Configuration > Add Integration > MQTT) and link it to the MQTT Broker (you need to reboot HA afterwards)
2- Enable YI-Hack MQTT and point it to your MQTT Broker IP. I used the default settings and left the Authentication blank for simplicity but you can set it for yourself and everywhere else needed
3- Configure YI-Hack MQTT Discovery (the link is on top of the YI-Hack MQTT page):
    3.1 - Enable the MQTT Camera Settings Advertise & Start on Boot
    3.2 - Set both 'Retain for Camera Settings' and 'QoS' to 0
    3.3 - Enable the Home Assistant & Start on Boot (keep the default settings)
4- Reboot your cam
Now you should see new entities starting with Yi in HA Entities including a new switch 'Yi Camera Status' that you can use turning on/off the cam. 
Hope it helps.

closed 2021-09-02 by roleoroleo