roleoroleo/yi-hack-Allwinner-v2 #188: Camera stuck on yellow light, no wifi, no reset possible -- Yi 1080p Home IFUS 	9.0.19* 	y21ga

Pulsar981, 2021-06-09:
Yi 1080p Home IFUS 	9.0.19* 	y21ga

Hi, after the allwinnerV2 hack the cam worked fine (offline, without yi-app). I've done some changes on the webinterface, which was fine so far.
The next day turning it on again, the yellow light immediately goes on, but nothing else happens. It doesn't connect to wifi, nor can I reset it - with and without SD card. Formatting SD to apply the hack again doesn't work, nothing happens. SD card is fine. Also tried with yi-app and internet connection, no success. Tried a manual firmware update like described on yi-support-page. But no, the cam stays in an unaccessable state with yellow light permanently on. Any ideas anybody?

roleoroleo, 2021-06-09:
Try the unhack procedure.

Pulsar981, 2021-06-09:
I've tried to unhack, did not work unfortunately. Same behaviour as described above.
(https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Remove-the-hack)

Do you have any other ideas except for hard unbrick?

roleoroleo, 2021-06-09:
Unbrick in the wiki.

roleoroleo, 2021-06-09:
If you try backup procedure, does it works?

Pulsar981, 2021-06-09:
I've tried, it did not work. Cam hangs, no matter what I do or what I put into SD.
I'm going to talk to the seller, maybe this is a hardware failure and I'll get a new one.
Thanks so far.

bfboarder, 2021-06-09:
> I've tried, it did not work. Cam hangs, no matter what I do or what I put into SD.
> I'm going to talk to the seller, maybe this is a hardware failure and I'll get a new one.
> Thanks so far.

I have one camera exactly doing the same (y21ga)
It did work fine for a few days and then is stoped working and only give a yellow light.

Up to now I did not try to fix it, but i might give it a shot in the next few days.
I'll let you know if i manage to make it work again.

roleoroleo, 2021-06-09:
Are you able to connect a serial cable?

bfboarder, 2021-06-09:
> Are you able to connect a serial cable?

Yeah i'll try that, but I can't do it today. Hope to be able to do it next weekend

mcanteri, 2021-06-23:
Have you tried to change the cable and use its own power supply ? 
This camera is very sensible to power, at least mine and happen to me to have a fixed yellow light and then swapped to proprietary cable and power plug and all went back working. Worth a try.

davidedg, 2021-06-23:
> 
> 
> Have you tried to change the cable and use its own power supply ?
> This camera is very sensible to power, at least mine and happen to me to have a fixed yellow light and then swapped to proprietary cable and power plug and all went back working. Worth a try.

I noticed the same thing.
I tried like 6 usb cables - none of them worked, only its proprietary one.
I think I was stucked, since I needed to place it far away from a power source, but I tried an usb extender and it worked flawlessly (I used an usb supercharger ~2.4A then the extender and in the end the proprietary cable).

closed 2023-01-08 by stale[bot]